UZ-SLAMLab/ORB_SLAM3 #340: Mono inertial scale too small

shigemori0625, 2021-06-22:
> ORB-SLAM3 Copyright (C) 2017-2020 Carlos Campos, Richard Elvira, Juan J. Gómez, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
ORB-SLAM2 Copyright (C) 2014-2016 Raúl Mur-Artal, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
This program comes with ABSOLUTELY NO WARRANTY;
This is free software, and you are welcome to redistribute it
under certain conditions. See LICENSE.txt.
Input sensor was set to: Monocular-Inertial
Loading ORB Vocabulary. This could take a while...
Vocabulary loaded!
Creation of new map with id: 0
Creation of new map with last KF id: 0
Seq. Name: 
Camera Parameters: 
Camera: Pinhole
fx: 699.3
fy: 699.3
cx: 640
cy: 360
k1: 0
k2: 0
p1: 0
p2: 0
k3: 0
fps: 28
color order: BGR (ignored if grayscale)
ORB Extractor Parameters: 
Number of Features: 1000
Scale Levels: 8
Scale Factor: 1.2
Initial Fast Threshold: 20
Minimum Fast Threshold: 7
Left camera to Imu Transform (Tbc): 
[1, 0, 0, 0;
 0, 1, 0, 0;
 0, 0, 1, 0;
 0, 0, 0, 1]
IMU frequency: 100 Hz
IMU gyro noise: 1.3e-08 rad/s/sqrt(Hz)
IMU gyro walk: 1.9393e-05 rad/s^2/sqrt(Hz)
IMU accelerometer noise: 2e-08 m/s^2/sqrt(Hz)
IMU accelerometer walk: 0.003 m/s^3/sqrt(Hz)
First KF:0; Map init KF:0
New Map created with 330 points
Point distribution in KeyFrame: left-> 330 --- right-> 0
build optimization graph
start optimization
end optimization
scale too small
build optimization graph
start optimization
end optimization
scale too small
build optimization graph
start optimization
end optimization
scale too small

When I ran Mono_Inertial, the System said "scale too small" .
Why did the System say so ??? Please give me some information. 
Many thanks in advance for any suggestion!

xinruozhishui201314, 2021-06-24:
我也出现过这个问题，好像IMU和相机的时间戳没对齐

MagdaZal, 2022-08-18:
Hi, have you solved this problem?

zoldaten, 2022-09-21:
try TUM-VI_far.yaml in Mono_Inertial