UZ-SLAMLab/ORB_SLAM3 #342: Error while building

charalambos02, 2021-06-27:
In file included from /home/charalambos/ORB_SLAM3/include/KeyFrame.h:24,
from /home/charalambos/ORB_SLAM3/include/MapPoint.h:23,
from /home/charalambos/ORB_SLAM3/include/Map.h:23,
from /home/charalambos/ORB_SLAM3/include/Atlas.h:22,
from /home/charalambos/ORB_SLAM3/include/MapDrawer.h:23,
from /home/charalambos/ORB_SLAM3/src/MapDrawer.cc:20:
/home/charalambos/ORB_SLAM3/Thirdparty/DBoW2/DBoW2/BowVector.h:17:10: fatal error:
boost/serialization/serialization.hpp: No such file or directory
17 | #include <boost/serialization/serialization.hpp>
| ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/ORB_SLAM3.dir/build.make:219:
CMakeFiles/ORB_SLAM3.dir/src/MapDrawer.cc.o] Error 1
In file included from /home/charalambos/ORB_SLAM3/include/KeyFrame.h:24,
from /home/charalambos/ORB_SLAM3/include/MapPoint.h:23,
from /home/charalambos/ORB_SLAM3/include/ORBmatcher.h:27,
from /home/charalambos/ORB_SLAM3/src/ORBmatcher.cc:20:
/home/charalambos/ORB_SLAM3/Thirdparty/DBoW2/DBoW2/BowVector.h:17:10: fatal error:
boost/serialization/serialization.hpp: No such file or directory
17 | #include <boost/serialization/serialization.hpp>
| ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/ORB_SLAM3.dir/build.make:128:
CMakeFiles/ORB_SLAM3.dir/src/ORBmatcher.cc.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:390: CMakeFiles/ORB_SLAM3.dir/all] Error 2
make: *** [Makefile:84: all] Error 2

Any idea what is going on? I have install all the requirements and libraries and I get those 2 errors which I have no idea what they are.

NickPerezCarletonUniversity, 2021-06-28:
i'm getting the same error. it looks like an issue with the repository, but i'm not sure. have you solved this issue yet?

charalambos02, 2021-06-30:
Haven't managed to find a solution. I've tried cloning the whole project and libraries from the beginning but I still get the same issue.

NickPerezCarletonUniversity, 2021-06-30:
I did some digging just now... the error is coming from a file that is included from this third party repo: https://github.com/dorian3d/DBoW2/blob/master/include/DBoW2/BowVector.h

As you can see, in this commit: https://github.com/UZ-SLAMLab/ORB_SLAM3/commit/df16ecc040d35ecbd9f81477e32e80b49460276c#diff-23604be25b2cb4a894a5b827e2092beac958e24a16a00baf9f1d6d1ae8f138ec
They added '#include <boost/serialization/serialization.hpp>' to the file. That code is not present in the original repo they took the code from. not sure why they added it there. Doing a search for 'serialization' on the last link I included, shows that ORB SLAM 3 has a mention of 'serialization' in the CMakesList.txt file.

NickPerezCarletonUniversity, 2021-06-30:
Orb Slam 2 does not have the added code:
https://github.com/raulmur/ORB_SLAM2/blob/master/Thirdparty/DBoW2/DBoW2/BowVector.h

NickPerezCarletonUniversity, 2021-06-30:
in orb slam 3's CMakeLists.txt there is this code: 

target_link_libraries(${PROJECT_NAME}
${OpenCV_LIBS}
${EIGEN3_LIBS}
${Pangolin_LIBRARIES}
${PROJECT_SOURCE_DIR}/Thirdparty/DBoW2/lib/libDBoW2.so
${PROJECT_SOURCE_DIR}/Thirdparty/g2o/lib/libg2o.so
-lboost_serialization
-lcrypto
)

doing a google search for -lboost_serialization brings up nothing. I imagine it is a library we have to install to get Orb Slam 3 to compile properly.

NickPerezCarletonUniversity, 2021-06-30:
...and the readme in the DBoW2 directory is saying that this code is 'the same version from ORB SLAM 2' which is not correct.

https://github.com/UZ-SLAMLab/ORB_SLAM3/blob/df16ecc040d35ecbd9f81477e32e80b49460276c/Thirdparty/DBoW2/README.txt

NickPerezCarletonUniversity, 2021-06-30:
this thread might have a solution to our problem: https://github.com/UZ-SLAMLab/ORB_SLAM3/issues/328

NickPerezCarletonUniversity, 2021-07-12:
@charalambos02 ... you need to install boost.... https://www.boost.org/doc/libs/1_76_0/more/getting_started/unix-variants.html#get-boost

charalambos02, 2021-07-13:
@NickPerezCarletonUniversity 
I was doing some research and working on some other stuff, didn't have much time to look into it. Does boost solve the issue?

NickPerezCarletonUniversity, 2021-07-13:
I am certain installing boost will solve the issue. I just haven't gotten around to installing it. Boost is usually included in Linux releases. If you are using a stripped down Linux release (e.g. Ubuntu 'base') it won't come with Boost installed, and you'll have to do it manually.

harshalkataria, 2021-09-14:
> @des1gnerw1tch no, I think you need to have the actual C files for boost on your computer. i tried solving the issue by installing boost the same way on Ubuntu (using the command 'sudo apt-get install libboost-all-dev'), and it didn't solve my issue. can you let me know if/how you get it to work?

sudo apt install libboost-dev solved the problem for me while building on Ubuntu 20.04

prasadvagdargi, 2022-01-07:
Install Boost libraries and include the header files in the folder as include directories during compilation.

li195111, 2022-08-25:
Hi, I build the boost 
```
wget https://boostorg.jfrog.io/artifactory/main/release/1.80.0/source/boost_1_80_0.tar.bz2
tar --bzip2 -xf boost_1_80_0.tar.bz2
rm -rf boost_1_80_0.tar.bz2
cd boost_1_80_0
sudo ./bootstrap.sh
sudo ./b2
sudo ./b2 install
```
it will install the boost into `/usr/local/lib` and `/usr/local/include`
then add the code `include_directories("/usr/local/include")` into `Thirdparty/DBoW2/CMakeLists.txt` under the `set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -Wall  -O3 -march=native")` line 10 maybe
and solve the problem

zoldaten, 2022-08-26:
sudo apt-get install libboost-dev

MuneebHoda, 2023-11-02:
I tried everything mentioned here but still no results. I am getting the same error. BowVector.h not found.

kyuhyong, 2023-11-14:
I figured this out.
You will need to run `build.sh` rather than `cmake` yourself.
That way DBoW will build prior to other projects.