UZ-SLAMLab/ORB_SLAM3 #343: master version building error

jhx2002, 2021-07-02:
```
/home/huaxing/oproj/orbslam3/ORB_SLAM3-master/src/LocalMapping.cc:1488:12: warning: unused variable 't_inertial_only' [-Wunused-variable]
     double t_inertial_only = std::chrono::duration_cast<std::chrono::duration<double> >(t1 - t0).count();
            ^~~~~~~~~~~~~~~
CMakeFiles/ORB_SLAM3.dir/build.make:110: recipe for target 'CMakeFiles/ORB_SLAM3.dir/src/LocalMapping.cc.o' failed
make[2]: *** [CMakeFiles/ORB_SLAM3.dir/src/LocalMapping.cc.o] Error 1
CMakeFiles/Makefile2:289: recipe for target 'CMakeFiles/ORB_SLAM3.dir/all' failed
make[1]: *** [CMakeFiles/ORB_SLAM3.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2
```

Ador2, 2021-07-05:
add this line `-lboost_serialization` to **CMakeLists.txt**
```
target_link_libraries(${PROJECT_NAME}
${OpenCV_LIBS}
${EIGEN3_LIBS}
${Pangolin_LIBRARIES}
${PROJECT_SOURCE_DIR}/Thirdparty/DBoW2/lib/libDBoW2.so
${PROJECT_SOURCE_DIR}/Thirdparty/g2o/lib/libg2o.so
-lboost_serialization
-lcrypto
)
```

AliBabaei0, 2021-07-06:
> add this line `-lboost_serialization` to **CMakeLists.txt**
> 
> ```
> target_link_libraries(${PROJECT_NAME}
> ${OpenCV_LIBS}
> ${EIGEN3_LIBS}
> ${Pangolin_LIBRARIES}
> ${PROJECT_SOURCE_DIR}/Thirdparty/DBoW2/lib/libDBoW2.so
> ${PROJECT_SOURCE_DIR}/Thirdparty/g2o/lib/libg2o.so
> -lboost_serialization
> -lcrypto
> )
> ```
> 
> and check/reinstall Openssl

It didn't solve the issue. Still the same problem.

Ador2, 2021-07-06:
This one `warning: unused variable 't_inertial_only'` is a warning, I assumed the error was boost :p
can you provide a log of what is printing out?

AliBabaei0, 2021-07-06:
I looked into the logs. Amongst all warnings, there was a single error :
`error: no match for ‘operator/’ (operand types are ‘cv::Matx<float, 3, 1>’ and ‘float’)
 x3D = x3D_h.get_minor<3,1>(0,0) / x3D_h(3);`
 
 to solve that, I used this solution [here](https://www.gitmemory.com/issue/UZ-SLAMLab/ORB_SLAM3/300/825588236) and it worked!

Ador2, 2021-07-07:
exactly!