UZ-SLAMLab/ORB_SLAM3 #345: What does the vocabulary mean?

oliver0922, 2021-07-05:
Hi I'm using stereo-inertial Camera with IMU.

I want use my own datasets. Therefore I record the dataset using rosbag.

I change the EuRoC.yaml file according to my Camera and IMU specifications. And then,run the code below.

**rosrun ORB_SLAM3 Stereo_Inertial Vocabulary/ORBvoc.txt Examples/Stereo-Inertial/EuRoC.yaml true**

Do I have to need my **own** ORBvoc.txt file for customizing? What does the vocaulary mean?

Ador2, 2021-07-05:
The provided Vocabulary file is trained from a huge dataset, it is somehow generic and should work very well with your setup.
so no need to make your own Vocabulary file.

NickPerezCarletonUniversity, 2021-07-09:
the vocabulary is used for DBoW2. it's what allows the algorithm to better recognize places it has already seen, from what I understand. the details are in the paper's they wrote for Orb Slam.

NickPerezCarletonUniversity, 2021-07-09:
DBoW2 also has a paper with details.

jdtardos, 2021-12-23:
Dorian Gálvez-López and Juan D. Tardós. Bags of Binary Words for Fast Place Recognition in Image Sequences. IEEE Transactions on Robotics, vol. 28, no. 5, pp. 1188-1197, 2012.

closed 2021-12-23 by jdtardos