UZ-SLAMLab/ORB_SLAM3 #346: Mono_Inertial mode does not work properly with RealSense D455 for image and IMU data acquisition

liangchen97, 2021-07-05:
Mono_Inertial mode does not work properly with RealSense D455 for image and IMU data acquisition

I made sure that I had ORB-SLAM3 installed correctly and that the Euroc dataset was running properly.Now I need to use RealSense D455 to run ORB-SLAM3's MONO mode and MONO_INERTIAL mode
We calibrate the D455 camera parameters, modify the topic in the code, and edit the YAML file

%YAML:1.0

#--------------------------------------------------------------------------------------------
# Camera Parameters. Adjust them!
#--------------------------------------------------------------------------------------------
Camera.type: "PinHole"

# Camera calibration and distortion parameters (OpenCV) 
Camera.fx: 638.1018216271331
Camera.fy: 638.3300728549654
Camera.cx: 629.4728680757129
Camera.cy: 375.5517788016263

Camera.k1: -0.05529716280253562
Camera.k2: 0.04869113994652115
Camera.p1: -0.0012081319997317495
Camera.p2: 0.000948771411746312

Camera.width: 1280
Camera.height: 720 

# Camera frames per second 
Camera.fps: 20.0

# Color order of the images (0: BGR, 1: RGB. It is ignored if images are grayscale)
Camera.RGB: 1

#--------------------------------------------------------------------------------------------
# ORB Parameters
#--------------------------------------------------------------------------------------------

# ORB Extractor: Number of features per image
ORBextractor.nFeatures: 1000

# ORB Extractor: Scale factor between levels in the scale pyramid 	
ORBextractor.scaleFactor: 1.2

# ORB Extractor: Number of levels in the scale pyramid	
ORBextractor.nLevels: 8

# ORB Extractor: Fast threshold
# Image is divided in a grid. At each cell FAST are extracted imposing a minimum response.
# Firstly we impose iniThFAST. If no corners are detected we impose a lower value minThFAST
# You can lower these values if your images have low contrast			
ORBextractor.iniThFAST: 20
ORBextractor.minThFAST: 7

#--------------------------------------------------------------------------------------------
# Viewer Parameters
#---------------------------------------------------------------------------------------------
Viewer.KeyFrameSize: 0.05
Viewer.KeyFrameLineWidth: 1
Viewer.GraphLineWidth: 0.9
Viewer.PointSize: 2
Viewer.CameraSize: 0.08
Viewer.CameraLineWidth: 3
Viewer.ViewpointX: 0
Viewer.ViewpointY: -0.7
Viewer.ViewpointZ: -1.8
Viewer.ViewpointF: 500

I can run Mono mode as normal.So I also calibrated the IMU foreign parameters using kalibr and edited the YAML file as follows:

%YAML:1.0

[... 181 lines cut ...]
LM: End reseting Local Mapping...
LM: Reset free the mutex
LM: Active map reset, Done!!!
mnFirstFrameId = 222
mnInitialFrameId = 199
67 Frames set to lost
First KF:35; Map init KF:24
New Map created with 77 points
Point distribution in KeyFrame: left-> 77 --- right-> 0
Fail to track local map!
IMU is not or recently initialized. Reseting active map...
SYSTEM-> Reseting active map in monocular case
LM: Active map reset recieved
LM: Active map reset, waiting...
LM: Reseting current map in Local Mapping...
LM: End reseting Local Mapping...
LM: Reset free the mutex
LM: Active map reset, Done!!!
mnFirstFrameId = 250
mnInitialFrameId = 245
69 Frames set to lost

It seems that ORB-SLAM3 has been unable to detect feature points and occasionally initializes soon after detecting and calculating the path, thus losing the result

[image]

Ador2, 2021-07-05:
notice `IMU is not or recently initialized. Reseting active map...`
you may need to excite the IMU more while keeping detected features.

hellovuong, 2021-07-06:
It could be due to Tbc matrix is not correct, in my case, I use this matrix provide by librealsense and it works:
`Tbc: !!opencv-matrix`
`   rows: 4`
`   cols: 4`
`   dt: f`
`   data: [ 0.999999,-0.00128063,0.000919103,-0.0289604812860489, `
`            0.00127786,0.999995,0.0030171,-0.00753192370757461,`
`            -0.000922961,-0.00301592,0.999995,-0.0156808998435736,`
`            0.0, 0.0, 0.0, 1.0]`
Moreover, your IMU noises parameters should be increased. 
Also please, change text properties in your question, so that it is easier to look at.

liangchen97, 2021-07-07:
Thank you for your answer. I will try to use librealsense to get the Tbc matrix. Besides, do I need to increase the noise of imu? These parameters of mine are calculated with kalibr. Can I refer to your yaml file?

liangchen97, 2021-07-07:
Thank you for your answer. I will try to use librealsense to get the Tbc matrix. Besides, do I need to increase the noise of imu? These parameters of mine are calculated with kalibr. Can I refer to your yaml  file?
@hellovuong

hellovuong, 2021-07-07:
hi @liangchen97, about IMU noise, I think yes you need to increase it by 10 times. Here is my config file for your reference [D455.txt](https://github.com/UZ-SLAMLab/ORB_SLAM3/files/6775572/D455.txt). It is Stereo-Intertial config, so just change it a little to work with Mono-Inertial case.

liangchen97, 2021-07-07:
hi @hellovuong 
Thank you again for providing the file. I have made the following modifications to my YAML file.
1. Modify the external parameters of IMU according to rs-enumerate-devices -c
2. Increase IMU internal parameters by 10 times
Things seem to be getting a little better, but they're still not good.Now feature points are easily lost and the predicted trajectory is erratic
Maybe my understanding is wrong.Can I ask you a few questions?
1. Why does "Transformation from camera to body-frame (IMU)" reflect Extrinsic from "Gyro" to "Color"
2. How to get the IMU noise data? The IMU noise obtained by using rs-enumerate-devices -c is all 0
3. I noticed that IMU.Frequency is 400, default is 200.Do I need to change it here
4. Have you tried Mono mode? My result is always that Mono-inertial is not as stable as Mono

hellovuong, 2021-07-07:
Good to know that. Sorry in the file I sent I used Infrared camera not Color so the Tbc in the file is Extrinsic from "Gyro"	  To	  "Infrared1". I forgot to change that line.

1. This would surprise you that either  Extrinsic from "Gyro" to "Infrared 1" or Extrinsic from "Infrared 1" to "Gyro" works in my experiments (Stereo +IMU). I could also confirm that Extrinsic from "Gyro" to "Color" or Extrinsic from "Color" to "Gyro" works in my experiments (Monocular using Color camera +IMU).  It is bothering me too, but I have no time to investigate it yet.
2. Exactly, they did not provide IMU noise. I used this [repo](https://github.com/gaowenliang/imu_utils) to compute it instead. Follow the instruction, you will get IMU noise then increase it by 10 or 20 times or follow [TUM-VI ](https://vision.in.tum.de/data/datasets/visual-inertial-dataset) on how they increased their IMU noise to inflating value.
3. Yes, I used IMU fps at 400. It would be no different if you use 200 (T265 used 200Hz) but I still prefer 400 and of course, you need to config it in your camera also.
4. I tested all cased: Mono only, Stereo Only, Mono/Stereo + IMU with both D435i, D455 and T265. Camera + IMU is as you said is "not stable", it is obvious due to the nature of IMU, but when you config it correctly it work very well especially in RECENTLY_LOST case.

I want to inform that how I recorded my datasets. Realsense provides accel and gyro separately, so you need to enable Online linear interpolation to merge that into 1 topic. That may caused problem, once I tested with my rosbag format datasets with IMU online linear interpolation and the result is not good as I remmeber correctly, it is been while ago . So recently, I recorded dataset in bag file with seperated accel and gyro topic. Then, I parsed that rosbag file into a dataset in TUM-VI or EuRoC format with images and IMU data in txt files. I wrote my own interpolation code to merge them in to IMU data base on gyro timestamps. I used that interpolated IMU data and the config file that I sent above and It works every well. I am not quite fimilar with ROS so that all my testing datasets are in EuRoC format, it is easier to debug. So, I may recommend you to do same proceduces on your dataset to see how the result.
Moreover at first 30s of your dataset, you should provide enough motion for your camera include moving along x-y-z and roll-pitch-yaw with NO constant velocity, so that the initialized process have good estiamtion of IMU bias and velocity.
I also found a [modified librealsense repo](https://github.com/ethz-asl/vi_realsense) that changed some stuffs related to IMU interpolation to make it works with maplab (V-I system). I have not time to test it yet but you may want to take a look at.
My config file for D455_Color [D455_Color.txt](https://github.com/UZ-SLAMLab/ORB_SLAM3/files/6778089/D455_Color.txt) for your reference 

Hope this can help you.

liangchen97, 2021-07-08:
@hellovuong  
Thank you for your willingness to take the time to answer my question. Based on the answers above, I have conducted further tests. I found that ORB-SLAM3 does not have strict requirements for external and internal parameters. Even if the imu noise of euroc.yaml is all set to 0, it can run with better results. So it is likely to be a problem with enable Online linear interpolation.
Since I am new to this aspect, could you please explain in detail how to use the recorded rosbag containing images, accelerometer and angular velocity to generate a EuroC format rosbag that can be used？

hellovuong, 2021-07-09:
Hi,
Please refer to this [Associate_datasets](https://github.com/hellovuong/Associate_datasets)

liangchen97, 2021-07-12:
Thank you, I will try this project

liangchen97, 2021-07-26:
@hellovuong 
Hello, I ran it according to the above method, but the result is still not getting better. I think it may be the problem of orbslam3, it may be that the configuration file of realsense d455 is not configured properly. As far as I am concerned, I only configure acc and gyr to be turned on, and merge them, the others are the default formats. Is this configuration sufficient? Or some insights in other directions above?

Pallav1299, 2021-08-19:
@hellovuong, thanks for your insights. 
I am working on an application which requires Monocular+Inertial SLAM. So, I came across ORB-SLAM3 and started testing it. It worked well with Euroc dataset. Then I tried using a Realsense D435i, but didn't achieve good results with the same even after doing external calibration using **kalibr** and **imu_utils**.  I am witnessing the following behaviour:
```
Input sensor was set to: Monocular-Inertial

Loading ORB Vocabulary. This could take a while...
Vocabulary loaded!

Creation of new map with id: 0
Creation of new map with last KF id: 0
Seq. Name: 

Camera Parameters: 
- Camera: Pinhole
- fx: 628.724
- fy: 626.18
- cx: 271.739
- cy: 218.344
- k1: 0.180974
- k2: -0.268076
- p1: -0.0105464
- p2: -0.0283861
- fps: 30
- color order: RGB (ignored if grayscale)

ORB Extractor Parameters: 
- Number of Features: 2000
- Scale Levels: 8
- Scale Factor: 1.2
- Initial Fast Threshold: 20
- Minimum Fast Threshold: 7

Left camera to Imu Transform (Tbc): 
[0.99755955, 0.013658711, 0.068472087, 0.030867778;
 -0.016439233, 0.99905598, 0.040210493, 0.0017487752;
 -0.067858227, -0.041237988, 0.99684238, -0.0033510376;
 0, 0, 0, 1]

IMU frequency: 400 Hz
IMU gyro noise: 0.00212992 rad/s/sqrt(Hz)
IMU gyro walk: 1.67862e-05 rad/s^2/sqrt(Hz)
IMU accelerometer noise: 0.0244312 m/s^2/sqrt(Hz)
IMU accelerometer walk: 0.000506753 m/s^3/sqrt(Hz)
First KF:0; Map init KF:0
New Map created with 95 points
Point distribution in KeyFrame: left-> 95 --- right-> 0
build optimization graph
start optimization
end optimization
IMU in Map 0 is initialized
Fail to track local map!
IMU is not or recently initialized. Reseting active map...
SYSTEM-> Reseting active map in monocular case
LM: Active map reset recieved
LM: Active map reset, waiting...
LM: Reseting current map in Local Mapping...
LM: End reseting Local Mapping...
LM: Reset free the mutex
LM: Active map reset, Done!!!
mnFirstFrameId = 117
mnInitialFrameId = 0
1270 Frames set to lost
First KF:107; Map init KF:0
New Map created with 161 points
Point distribution in KeyFrame: left-> 161 --- right-> 0
build optimization graph
start optimization
end optimization
IMU in Map 0 is initialized
Fail to track local map!
[... 211 lines cut ...]
LM: Active map reset recieved
LM: Active map reset, waiting...
LM: Reseting current map in Local Mapping...
LM: End reseting Local Mapping...
LM: Reset free the mutex
LM: Active map reset, Done!!!
mnFirstFrameId = 3456
mnInitialFrameId = 3301
2903 Frames set to lost
```

> I want to inform that how I recorded my datasets. Realsense provides accel and gyro separately, so you need to enable Online linear interpolation to merge that into 1 topic. That may cause the problem, once I tested with my rosbag format datasets with IMU online linear interpolation and the result is not good as I remember correctly, it is been while ago. So recently, I recorded the dataset in rosbag format with separated accel and gyro topics. Then, I parsed that rosbag file into a dataset in TUM-VI or EuRoC format with images and gyro/accel data in txt files. I wrote my own interpolation code to merge them into the IMU data based on gyro timestamps. I used that interpolated IMU data and the config file that I sent above and It works very well on ORB_SLAM3. I am not quite familiar with ROS so that all my testing datasets are in EuRoC format, it is easier to debug. So, I may recommend you to do the same procedures on your dataset to see how the result. If It works, the problem will be at linear interpolation in librealsence as I assumed. Otherwise, we need to investigate further.

After going through your discussion it seemed to be an IMU related issue in D435i. Following which I tried converting my collected Rosbag to Euroc format with different time interpolation using [https://github.com/hellovuong/Associate_datasets](Associate_datasets) but still didn't achieve good results.

> I also found a [modified librealsense repo](https://github.com/ethz-asl/vi_realsense) that changed some stuff related to IMU interpolation to make it works with maplab (V-I system). I have not time to test it yet but you may want to take a look at it.

I even tried using the [modified librealsense repo](https://github.com/ethz-asl/vi_realsense). The results were still the same. 

Few questions that I wanted to ask:

- What is the primary odometry source in ORB-SLAM3 (visual or inertial)?
- Will it work in featureless environments by just using an IMU? Any constraints?

I also tried using D435i with Vins Fusion (Mono+Inertial) where I could achieve stable tracking as compared to ORB-SLAM3(Mono+Inertial). Tracking using IMU was also working but degrades after sometime in featureless environments where Visual odometry is not available.

hellovuong, 2021-08-19:
@liangchen97 hi sorry, I didn't see your reply. Actually, it is wired that the preprocessed dataset is not working. 
I did not set to merge acc and gyro at the time I recorded the dataset, it is two separate topics only, merge them by the repo that I shared

hellovuong, 2021-08-19:
@Pallav1299 hi, have you try the [OpenLoris](https://lifelong-robotic-vision.github.io/dataset/scene.html) dataset on ORB_SLAM3, they are using D435i also. Along with the data, they provided config files including Tbc. May you want to try that dataset first? then if it works well, you can try their configuration with your camera, pay attention on Tbc matrix.

Pallav1299, 2021-08-23:
Thanks for your suggestion @hellovuong. 
I tried ORB-SLAM3 with the cafe dataset by converting the rosbag data to Euroc dataset format using [https://github.com/UZ-SLAMLab/ORB_SLAM3/issues/Associate_datasets](https://github.com/UZ-SLAMLab/ORB_SLAM3/issues/Associate_datasets) for IMU interpolation. 

I was able to achieve satisfactory results with the dataset. But I am still not able to achieve similar results on my setup although my extrinsic calibration was almost the same as the on in the dataset.

Can you help me in understanding the suspects for such behaviour? I suspect my calibration for the same. What do you think? If so, can you help me out with the exact procedure for calibrating Realsense D435i? 

I believe the following things to be causing trouble:
- noisy IMU
- large reprojection error in Kalbr results
- improper IMU intrinsic calibration.

Just for clarification, I am using the RGB color camera(rolling shutter) and IMU data from D435i.

HuangTY96, 2021-08-24:
I am using D435i to do mono-inertial slam. When I use IMU (200Hz) and color (20Hz), it will show same problem, but now I change them to IMU(400Hz) and color(30Hz), and the result is nice.

Pallav1299, 2021-08-24:
I was able to get it working after proper calibration. Thanks.

yaelbenoren, 2021-09-09:
could you please explain how you required the proper calibration? I have calibrated using Kalibr and am getting very bad results

HuangTY96, 2021-09-09:
> could you please explain how you required the proper calibration? I have calibrated using Kalibr and am getting very bad results

I just use Kalibr, but now I use IMU(400Hz) and color(30Hz), you can also try this parameter.

yaelbenoren, 2021-09-09:
This won't help in my case since I am using IMU in 416Hz and camera in 10Hz (and can not change this).  
@Pallav1299 used Kalibr too but modified something in the calibration to get good results.

Pallav1299, 2021-09-09:
@yaelbenoren, ORB-SLAM3 mono-inertial won't work well for the RGB sensor(rolling shutter) + IMU on D435i. Also mentioned on this [ISSUE](https://github.com/UZ-SLAMLab/ORB_SLAM3/issues/94). I used Kalibr for extrinsic calibration and I was able to achieve decent results. The odometry degrades badly in some cases and is not able to recover in such scenarios. You can use the Camera-IMU extrinsics from [OpneLoris](https://lifelong-robotic-vision.github.io/dataset/scene.html) as is. What remains is the camera intrinsics and IMU intrinsics. Please mind the camera model that you choose for its calibration. I uses the pinhole-radtan model while using Kalibr. For IMU intrinsics I used [imu_utils](https://github.com/gaowenliang/imu_utils). 

Now I am using the left IR camera(global shutter on D435i) as input to ORB-SLAM3 in place of the RGB sensor. You can use the following modified [Realsense ros-wrapper](https://github.com/ZJU-FAST-Lab/ego-planner#improved-ros-realsense-driver) for removing the projector dots from the IR cameras.

mateosss, 2021-10-01:
I was having similar logs to the ones presented in this issue with the D455, the most notorious symptom was the map resetting every 10 keyframes. In my case, I was able to solve it by checking different things. The bottom line is: double-check that all the timestamps that you are sending to `TrackMonocular` or `TrackStereo` make sense.

That includes:
- Making sure that all timestamps (frames, accelerometer, gyroscope) are in the same "timestamp domain" (a concept from the RealSense SDK). In my case, I was unable to install the DKMS kernel patch required by the RealSense API and this made the gyroscope stream work in a different timestamp domain which was about ~30ms off. (A quick solution for this is to use the accelerometer timestamps at 250hz and just copy the values from the last gyroscope sample at 200hz)
- Making sure you don't block execution in the RealSense threads when receiving a new sample. Make the receiving of RealSense samples lightweight, just push to a [threadsafe queue](https://github.com/cameron314/concurrentqueue) (or even better, a single-writer single-consumer queue like [this](https://github.com/cameron314/readerwriterqueue)) and process the samples in a separate thread.
- I didn't need to implement IMU interpolation, but be sure to save for later IMU samples that have a timestamp greater than the image frame you are sending, else ORB-SLAM3 discards them.

RoZhong, 2021-11-30:
> color

May I ask what your color is? I don't know what this 20Hz is

songxuekun, 2022-04-07:
> 

Have you solved the problem yet

wwendy233, 2022-08-12:
> @yaelbenoren, ORB-SLAM3 mono-inertial won't work well for the RGB sensor(rolling shutter) + IMU on D435i. Also mentioned on this [ISSUE](https://github.com/UZ-SLAMLab/ORB_SLAM3/issues/94). I used Kalibr for extrinsic calibration and I was able to achieve decent results. The odometry degrades badly in some cases and is not able to recover in such scenarios. You can use the Camera-IMU extrinsics from [OpenLoris](https://lifelong-robotic-vision.github.io/dataset/scene.html) as is. What remains is the camera intrinsics and IMU intrinsics. Please mind the camera model that you choose for its calibration. I uses the pinhole-radtan model while using Kalibr. For IMU intrinsics I used [imu_utils](https://github.com/gaowenliang/imu_utils).
> 
> Now I am using the left IR camera(global shutter on D435i) as input to ORB-SLAM3 in place of the RGB sensor. You can use the following modified [Realsense ros-wrapper](https://github.com/ZJU-FAST-Lab/ego-planner#improved-ros-realsense-driver) for removing the projector dots from the IR cameras.
> 
> I am still sceptical about the failure detection and managing on ORB-SLAM3. Open to suggestion for the same.

What is the reprojection error after you calibrate the D345i with kalibr? 
The reprojection error I calibrated is about 0.1. T_ic (cam0 to imu0) generated by kalibr is uesd to run monocular_inertia, but the map still resets every 10 keyframes. 
I wonder if it's because of the calibration accuracy is not enough.

FzzfSisyphus, 2025-02-17:
I change the line from USB 2.0 to USB 3.0  it works well . 
You can use  `lsusb -t` to check the data transport speed