UZ-SLAMLab/ORB_SLAM3 #347: T265 stereo_inertial, orb slam 3 keeps initialising

Shervina-lim, 2021-07-07:
Hi all,

I'm having troubles running **stereo_inertial using realsense T265**. I tried using 2 different camera models: kannalabrandt8 and pinhole, but none worked.
For **kannalabrandt8**, it keeps resetting the map every few seconds. For **pinhole**, it resets for around every minute, slightly less frequent than kannalabrandt8. The image below shows the "error" I got.

[image]

I have calibrated my camera using kalibr and these are my config files: [t265_kannala.txt](https://github.com/UZ-SLAMLab/ORB_SLAM3/files/6773936/t265_kannala.txt), [t265_pinhole.txt](https://github.com/UZ-SLAMLab/ORB_SLAM3/files/6773937/t265_pinhole.txt). Couldn't upload as yaml so, changed it to txt. By the way, I managed to run **mono_inertial** so, the imu part of the config files should be alright. I'm just not very sure about the stereo part. Also I set **stereo rectification to false** when running it.

Other than not able to run stereo_inertial using t265, I have some other questions.

1. What is the difference between using kannalabrandt8 and pinhole for t265 and when to use which? Some uses kanalabrandt8 ([issue](https://github.com/UZ-SLAMLab/ORB_SLAM3/issues/305)), while some uses pinhole ([issue](https://github.com/UZ-SLAMLab/ORB_SLAM3/issues/231)) .
2. It is necessary to do stereo rectification for t265?

Can anyone please help me with this! Thanks in advance!

Regards,
Shervina

hellovuong, 2021-07-07:
I used kanalabrandt8 model. Here is my config file  [T265.txt](https://github.com/UZ-SLAMLab/ORB_SLAM3/files/6774776/T265.txt). It works both for my dataset and [OpenLoris](https://lifelong-robotic-vision.github.io/dataset/scene.html) datasets.

NickPerezCarletonUniversity, 2021-07-09:
did you get this error when building: https://github.com/UZ-SLAMLab/ORB_SLAM3/issues/342

If so, how did you solve it?

Shervina-lim, 2021-07-12:
> did you get this error when building orb slam: #342
> 
> If so, how did you solve it?

Hi Nick,

I also got errors when building ORB SLAM 3 but can't remember if its the same as yours. Btw I'm using [ORB SLAM 3 fixed](https://github.com/shanpenghui/ORB_SLAM3_Fixed) instead.

Shervina-lim, 2021-07-12:
> I used kanalabrandt8 model. Here is my config file [T265.txt](https://github.com/UZ-SLAMLab/ORB_SLAM3/files/6774776/T265.txt). It works both for my dataset and [OpenLoris](https://lifelong-robotic-vision.github.io/dataset/scene.html) datasets.

Hi Hellovuong,

Thanks for the info! I'll give it a try.

iamchoking, 2021-07-12:
> I used kanalabrandt8 model. Here is my config file [T265.txt](https://github.com/UZ-SLAMLab/ORB_SLAM3/files/6774776/T265.txt). It works both for my dataset and [OpenLoris](https://lifelong-robotic-vision.github.io/dataset/scene.html) datasets.

I was looking for a solution to this, including the massive drift I get when using this, The imu noise and walk parameters really helped me alot! thanks @hellovuong  !

iamchoking, 2021-07-12:
@Shervina-lim  Also, If you are getting not enough acceleration, try moving the camera just a little to initialize, if that solves the problem, great (I think ORB_SLAM needs at least a little nudge to initialize the imu)! If it doesn't try tweaking the noise values!

Shervina-lim, 2021-07-13:
Hi!
I tried the factory calibration parameters but it didn't work too! 
Once the camera is not moving, it will print **not enough acceleration**.  When moving, it keeps re initializing the map. 

Regarding the imu noise parameters, how do I tweak them? I got mine from doing imu noise calibration with [imu_utils](https://github.com/gaowenliang/imu_utils)

sumitsarkar1, 2021-07-14:
> I used kanalabrandt8 model. Here is my config file [T265.txt](https://github.com/UZ-SLAMLab/ORB_SLAM3/files/6774776/T265.txt). It works both for my dataset and [OpenLoris](https://lifelong-robotic-vision.github.io/dataset/scene.html) datasets.

Thanx for the settings file for t265....Do you face map initialising problem ? Becasuse I am facing the map initialising problem...

jdtardos, 2021-12-23:
Version 1.0 now includes examples to run ORB-SLAM3 with Intel Realsense T265

closed 2021-12-23 by jdtardos

IAMHAADICOOL, 2023-12-16:
I am encountering the same problem with intel realsense D435i. What shoud I do? Any suggestions.