UZ-SLAMLab/ORB_SLAM3 #348: ORB-SLAM3 RGB-D SENSOR

thomCastillo, 2021-07-14:
I am running the SLAM algorithm using RGB-D sensor. What should I get the Depth Threshold(ThDepth) and baseline(bf) values that we get from the YAML file while running the algorithm?

I couldn't find an article that gives information about Threshold depth(ThDepth) do you have any article you can recommend on this subject? We got the bf value as (color fx*baseline) is it correct? In addition, while we were running the SLAM algorithm with depth (RGB-D), we encountered an error of 10 cm at 8 meters, although the camera parameters we got from RealSense were correct. Why this error and is this error normal?

iamchoking, 2021-07-14:
@thomCastillo Hello! I can confirm that the bf value is correct. I have used the fx*baseline value and it worked great for me for various stereo sensors.

thomCastillo, 2021-07-14:
@iamchoking What do you get the depth threshold(ThDepth) value and do these values(bf and thDepth) affect the depth error?

thomCastillo, 2021-07-31:
Could we add depth sensor error as parameter(i.e noise parameter in IMU) to optimization in RGB-D SLAM? If so, how can we add it?