UZ-SLAMLab/ORB_SLAM3 #349: RGB-D + Inertial Sensor for ORB SLAM 3 ROS

iamchoking, 2021-07-14:
Hello! I am very new to ORB SLAM 3 and had alot of fun using it, and was surprised at how far just one week of tweaking can get me!

I was wondering if I could somehow use the IMU data for ORB SLAM with my RGBD sensor (i.e. D455) to improve the outputs?

I think the current release does not support this in ROS. Are there plans to add this feature?

If not, how would I go about doing this on my own?

Thanks in advance!

jdtardos, 2021-12-23:
Version v1.0 includes a RGB-D+IMU demo

closed 2021-12-23 by jdtardos

fbliman, 2023-05-25:
I guess the question was  for the ros pacakge. That mode is not available.
I am also trying to run orbslam in RGBD_inertial mode and I have my data in a .bag adn was wondering if I could do it with the orbslam3 ros pacake