UZ-SLAMLab/ORB_SLAM3 #355: Why do you need to use sort() in function DistributeOctTree?

dima2876, 2021-07-19:
Hi, 
I try to figure out why sort() is used in function DistributeOctTree... Can somebody explain please.

wangtsiao, 2021-08-19:
there sort( ) function seems unused...

jeremvv, 2021-10-07:
Hi,

In my understanding, the sort function is used to sort the vSizeAndNode vector before doing the last step when computing the octree.
This last step is started when we predict that we gonna have as many nodes as the number of features we set as limit or when all nodes already have 1 feature only.

So, when we start that step, we sort the vector to start dividing the nodes with the highest number of features first, to minimize the nodes with lots of features in the end.

The only problem I saw in that part, is that the sorting, for the elements with the same number of features (the first element of the pair vSizeAndNode) is done based on the second element of the pair. This second element is a pointer, so an address which is not reproductible.
So in my opinion this bring some random in the octree at every run.

dima2876, 2021-10-10:
jeremvv, you are completely right! The problem, you are talking about indeed exists. To solve it, I added  a comparison condition to the sort() function.

jeremvv, 2021-10-12:
I think this is the sorting condition by default no ?
It is still comparing the first element but doing nothing specific on the second one (which is a pointer value so can have completely random values).