UZ-SLAMLab/ORB_SLAM3 #360: Failed to load image at ORB_SLAM-3

ayushGHub, 2021-08-05:
I followed all the suggestions given to date, check PATH as well as naming in the .txt file and image name in the dataset folder. Everything seems to be correct.

Any IDEA what could be the error here?

```
abulary/ORBvoc.txt ./Examples/Monocular/EuRoC.yaml ~/Datasets/EuRoc/MH01 ./Examples/Monocular/EuRoC_TimeStamps/MH01.txt
num_seq = 1
Loading images for sequence 0...LOADED!

-------

ORB-SLAM3 Copyright (C) 2017-2020 Carlos Campos, Richard Elvira, Juan J. Gómez, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
ORB-SLAM2 Copyright (C) 2014-2016 Raúl Mur-Artal, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
This program comes with ABSOLUTELY NO WARRANTY;
This is free software, and you are welcome to redistribute it
under certain conditions. See LICENSE.txt.

Input sensor was set to: Monocular

Loading ORB Vocabulary. This could take a while...
Vocabulary loaded!

Creation of new map with id: 0
Creation of new map with last KF id: 0
Seq. Name: 

Camera Parameters: 
- Camera: Pinhole
- fx: 458.65399169921875
- fy: 457.29598999023438
- cx: 367.21499633789062
- cy: 248.375
- k1: -0.28340810537338257
- k2: 0.073959067463874817
- p1: 0.00019359000725671649
- p2: 1.7618711353861727e-05
- fps: 20
- color order: RGB (ignored if grayscale)

ORB Extractor Parameters: 
- Number of Features: 1000
- Scale Levels: 8
- Scale Factor: 1.2000000476837158
- Initial Fast Threshold: 20
- Minimum Fast Threshold: 7
**libpng warning: Application built with libpng-1.2.54 but running with 1.6.20**

**Failed to load image at: /home/Datasets/EuRoc/MH01/mav0/cam0/data/1403636579763555584.png
Segmentation fault (core dumped)**
```

closed 2021-08-10 by ayushGHub

iiiicce8010, 2021-09-05:
I met the same problem, could you share your experience plz

Mechazo11, 2022-03-30:
@ayushGHub I got the author's Version 1.0 to work now after solving the same problem that you were having.

Two things I did,
1. Checked the name of the "Dataset" directory. In my case, I named it as "DATASETS" but was passing "Datasets".
2. The first image "1403636579763555584.png" may be really corrupted even if we can open it manually. I deleted the "1403636579763555584.png" image and also carefully removed that timestep from the /Examples/Monocular/EuRoC_TimeStamps/MH01.txt file. 

As soon as I performed the above two operation, not only did ORB_SLAM3 started to work, but also the Pangolin visualization started to work too.

Hope this helps.

With best,
Mechazo11

iiiicce8010, 2022-03-31:
Thank you for your reply, got it!0.0

从 Windows 版邮件<https://go.microsoft.com/fwlink/?LinkId=550986>发送