UZ-SLAMLab/ORB_SLAM3 #365: Mono-Inertial fail when few feature points

HuangTY96, 2021-08-24:
I am using D435i to run MONO-INERTIAL slam. before I move camera towards wall without feature points, the result is nice, but after that, it shows:

Fail to track local map!
PIOLKF: NOT ENOUGH EDGES
Fail to track local map!
PIOLKF: NOT ENOUGH EDGES
Fail to track local map!
PIOLF: NOT ENOUGH EDGES
Fail to track local map!
PIOLKF: NOT ENOUGH EDGES
Fail to track local map!
PIOLKF: NOT ENOUGH EDGES
Fail to track local map!
PIOLKF: NOT ENOUGH EDGES
Fail to track local map!
PIOLKF: NOT ENOUGH EDGES
Fail to track local map!
PIOLKF: NOT ENOUGH EDGES
Fail to track local map!
PIOLKF: NOT ENOUGH EDGES
Fail to track local map!
PIOLKF: NOT ENOUGH EDGES
Fail to track local map!
PIOLKF: NOT ENOUGH EDGES
Fail to track local map!
PIOLKF: NOT ENOUGH EDGES
Fail to track local map!
Segmentation fault (core dumped)

I dont get it, because I think in MONO-INERTIAL case, slam will be robust even in less feature situation.

Pallav1299, 2021-08-24:
I am witnessing the similar results.

HuangTY96, 2021-08-30:
> hi，could you share video and imu data collected by your realsence sensor, i want to dig deeper into this issue.

yes, could you give me your private email.
ty_huang2021@163.com, this is mine.

BogdanSerban95, 2021-09-22:
Hello! Did you find the cause of this error? I am facing the same issue when running in stereo-inertial mode.