UZ-SLAMLab/ORB_SLAM3 #374: Build Error with Pangolin

bytemani, 2021-08-30:
Hi, I met many errors related with Pangolin (version 0.6) as below. Any one knows what is going on? Thank you.
My working environment is ubuntu 18.04.

[ 59%] Building CXX object CMakeFiles/ORB_SLAM3.dir/src/ORBmatcher.cc.o
[ 60%] Building CXX object CMakeFiles/ORB_SLAM3.dir/src/LoopClosing.cc.o
In file included from /home/guliang/slam/Pangolin/components/pango_core/include/pangolin/utils/signal_slot.h:3:0,
                 from /home/guliang/slam/Pangolin/components/pango_windowing/include/pangolin/windowing/window.h:34,
                 from /home/guliang/slam/Pangolin/components/pango_display/include/pangolin/display/display.h:34,
                 from /home/guliang/slam/Pangolin/components/pango_core/include/pangolin/pangolin.h:38,
                 from /home/guliang/ORB_SLAM3/include/Map.h:27,
                 from /home/guliang/ORB_SLAM3/include/KeyFrameDatabase.h:30,
                 from /home/guliang/ORB_SLAM3/include/KeyFrame.h:29,
                 from /home/guliang/ORB_SLAM3/include/MapPoint.h:23,
                 from /home/guliang/ORB_SLAM3/src/MapPoint.cc:19:
/home/slam/Pangolin/components/pango_core/include/signals/signals.hpp:257:17: error: ‘auto’ parameter not permitted in this context
   template<auto PMF, class C>
                 ^~~
/home/slam/Pangolin/components/pango_core/include/signals/signals.hpp:269:17: error: ‘auto’ parameter not permitted in this context
   template<auto func>
                 ^~~~
/home/slam/Pangolin/components/pango_core/include/signals/signals.hpp: In member function ‘fteng::connection_raw fteng::signal<void(A ...)>::connect(C*) const’:
/home/slam/Pangolin/components/pango_core/include/signals/signals.hpp:261:36: error: forming reference to void
    auto& call = calls.emplace_back();
                                    ^
/home/slam/Pangolin/components/pango_core/include/signals/signals.hpp: In lambda function:
/home/slam/Pangolin/components/pango_core/include/signals/signals.hpp:263:101: error: ‘PMF’ was not declared in this scope
    call.func = reinterpret_cast<void*>(+[](void* obj, A ... args) {((*reinterpret_cast<C**>(obj))->*PMF)(args...); });
                                                                                                     ^~~
/home/slam/Pangolin/components/pango_core/include/signals/signals.hpp: In member function ‘fteng::connection_raw fteng::signal<void(A ...)>::connect() const’:
/home/slam/Pangolin/components/pango_core/include/signals/signals.hpp:272:19: error: ‘func’ was not declared in this scope
    return connect(func);
                   ^~~~
/home/slam/Pangolin/components/pango_core/include/signals/signals.hpp:272:19: note: suggested alternative: ‘sync’
    return connect(func);
                   ^~~~
                   sync
/home/slam/Pangolin/components/pango_core/include/signals/signals.hpp: In member function ‘fteng::connection_raw fteng::signal<void(A ...)>::connect(void (*)(A ...)) const’:
/home/slam/Pangolin/components/pango_core/include/signals/signals.hpp:278:36: error: forming reference to void
    auto& call = calls.emplace_back();
                                    ^
/home/slam/Pangolin/components/pango_core/include/signals/signals.hpp: In member function ‘fteng::connection_raw fteng::signal<void(A ...)>::connect(F&&) const’:
/home/slam/Pangolin/components/pango_core/include/signals/signals.hpp:289:23: error: ‘is_convertible_v’ is not a member of ‘std’
    if constexpr (std::is_convertible_v<f_type, void(*)(A...)>)
                       ^~~~~~~~~~~~~~~~

tianxiaguixin002, 2021-08-30:
Me too, ubuntu18.04, I gauss orbslam3 use old version pangolin

marcusabate, 2021-08-30:
I also see this error in both 18.04 and 20.04, if the authors could provide a last working commit for Pangolin we could use that.

joshi-bharat, 2021-08-31:
> I also see this error in both 18.04 and 20.04, if the authors could provide a last working commit for Pangolin we could use that.

check out v0.6 for pangolin which is the latest release.

ashwinvk94, 2021-09-01:
When cloning pangolin run `git checkout v0.5`(before building) to use v0.5 and the build works for me.

Chris0919, 2021-09-07:
I also met this 
use pangolin-0.6 will fix this

UcefMountacer, 2021-09-13:
@Chris0919 

I built pangolin 0.6 from this repo https://github.com/stevenlovegrove/Pangolin/tree/v0.6

but it didn't solve the problem.

Can you link me to the source from where you built? thanks

baimao-v, 2021-11-01:
I use the pangolin 0.6 and solve the error sucessfully

liaozihang, 2021-11-11:
change your CMakeLists.txt
c++11 -> c++14

clydemcqueen, 2025-03-18:
This worked for me on Ubuntu 22.04:
* `git checkout v0.6`
* Change `set (CMAKE_CXX_STANDARD 14)` to `set (CMAKE_CXX_STANDARD 17)` in CMakeLists.txt

DanielChaseButterfield, 2026-09-04:
Pangolin v0.6 worked for me as well on the master branch