UZ-SLAMLab/ORB_SLAM3 #375: CMakeFiles/ORB_SLAM3.dir/build.make:107: recipe for target 'CMakeFiles/ORB_SLAM3.dir/src/LocalMapping.cc.o' failed

mobiiin, 2021-08-31:
`CMakeFiles/ORB_SLAM3.dir/build.make:110: recipe for target 'CMakeFiles/ORB_SLAM3.dir/src/LocalMapping.cc.o' failed
make[2]: *** [CMakeFiles/ORB_SLAM3.dir/src/LocalMapping.cc.o] Error 1
CMakeFiles/Makefile2:289: recipe for target 'CMakeFiles/ORB_SLAM3.dir/all' failed
make[1]: *** [CMakeFiles/ORB_SLAM3.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2`

faced this after following all the instructions
ubuntu 18

mobiiin, 2021-08-31:
turns out there is something wrong with the source file
found this solution:
add this to top of the file 
`ORB_SLAM3-master/include/CameraModels/KannalaBrandt8.h `

```
namespace cv 
{ 
  template<typename _Tp, int m, int n> static inline Matx<_Tp, m, n> operator / (const Matx<_Tp, m, n>& a, float alpha) 
    {
     return Matx<_Tp, m, n>(a, 1.f / alpha, Matx_ScaleOp()); 
    } 
}
```
compile with `make -j1`

closed 2021-08-31 by mobiiin

zhh2005757, 2022-01-10:
Great job! Your answer solved my same problem EXACTLY! Thanks a lot!

muqianjx, 2024-08-13:
```
In file included from /home/muqian/SLAM/ORB_SLAM_v3/include/System.h:31:0,
                 from /home/muqian/SLAM/ORB_SLAM_v3/src/System.cc:21:
/home/muqian/SLAM/ORB_SLAM_v3/include/Tracking.h: At global scope:
/home/muqian/SLAM/ORB_SLAM_v3/include/Tracking.h:47:70: error: redefinition of ‘template<class _Tp, int m, int n> cv::Matx<_Tp, m, n> cv::operator/(const cv::Matx<_Tp, m, n>&, float)’
   template<typename _Tp, int m, int n> static inline Matx<_Tp, m, n> operator / (const Matx<_Tp, m, n>& a, float alpha)
```
I had added it,but it is still wrong.
ubuntu18 too