UZ-SLAMLab/ORB_SLAM3 #384: Running RGB-D on freiburg1 desk dataset

UcefMountacer, 2021-09-16:
HI,

I installed a docker image provided by @jahaniam. It works well and tested it with the shell files provided in the project.

I tried a dataset called freiburg1 desk dataset which is obviously should work since they already included a txt files of image and depth frames.

I made a shell file to run these datasets, but the container does not show any sign of processing, although the laptop is fired up.

here is the shell file: 

```
#!/bin/bash
pathDatasetTUM_VI='/home/youssef/Readar/rgbd_dataset_freiburg1_desk' #Example, it is necesary to change it by the dataset path

echo "Launching freiburg1 desk dataset"

./RGB-D/rgbd_tum ../Vocabulary/ORBvoc.txt RGB-D/TUM1.yaml "$pathDatasetTUM_VI"/ /associations/fr1_desk_correct.txt

```

Is it because of the way the dataset are stored inside the directory ? I tried a bunch of stuff like putting them in the same folder. The rgbd_tum.cc file apparently wants this command format on the shell to work well:

`./rgbd_tum path_to_vocabulary path_to_settings path_to_sequence path_to_association
`

I hope someone will help troubleshoot it or better have done it ! 

Thanks :)

jahaniam, 2021-09-17:
@UcefMountacer Thanks for trying it out. 

You are correct. This problem is not related to orbslam. It's the mapping inside container and outside. The path 
`pathDatasetTUM_VI='/home/youssef/Readar/rgbd_dataset_freiburg1_desk' #Example, it is necesary to change it by the dataset path`
does not exist inside container and I'm guessing that's why orbslam does nothing and freezes like a while(1) without any sleep inside the loop.
To fix that there is a folder in the repo called `Dataset`, put your dataset files there. That folder is mapped to `/Datasets`

another option would be in build_container.sh add the mapping to your existing folder dataset. between line 22 and 23 add:
`-v /home/youssef/Readar/rgbd_dataset_freiburg1_desk:/home/youssef/Readar/rgbd_dataset_freiburg1_desk \`
and rerun the build_container.sh

UcefMountacer, 2021-09-18:
Hi @jahaniam, thanks for the idea. I tried it but seems the problem comes from path to association file (timestamp, rgb, depth) being wrong. The codes lacks a bit of error messages to explain problems.

abtabrizi, 2023-08-16:
hi @UcefMountacer were you able to resolve this issue?