UZ-SLAMLab/ORB_SLAM3 #385: Does orb slam 3 work on WSL (Windows Subsystem for Linux)

NickPerezCarletonUniversity, 2021-09-17:
I haven't tried this yet. Does anyone know if it works? I will update this question if/when I try using Orb Slam on WSL.

DavidPetkovsek, 2021-09-22:
There are a lot of dependencies that need to be installed through apt but you can indeed run the orb slam 3 examples and evaluations provided in the repository. However please not that if you intend to use it with your own webcam that WSL does not support USB pass through and so is not compatible.

NickPerezCarletonUniversity, 2021-09-23:
is there anything outside of the libraries listed on the Orb Slam 3 repository that I need to install? Or anything else 'special' I have to do to get Orb Slam to run on WSL? I heard there are issues with openGL and WSL, not sure if that has been resolved recently. Orb Slam 3 uses pangolin, which uses openGL. also, I believe WSL can't access your GPU, so you wouldn't be able to accelerate any performance using your GPU (again, unless this was changed recently, I'm just parroting what I have seen other people say in other forums).

NickPerezCarletonUniversity, 2021-09-23:
and when I say WSL, I really mean WSL 2.

DavidPetkovsek, 2021-09-24:
I just checked WSL and turns out I have WSL 1 not WSL 2.
However, that being said I did follow all information for WSL 2 (I thought I updated my laptop to 2 but turns out I only updated my desktop). I would imagine it wouldn't be too different for you.

There are a bunch of basic libraries you need to install not listed in the Orb Slam 3 repository.
The reason there are so many apt packages to install is because the ubuntu install has many basic elements normally packaged with a standard ubuntu install missing. Things like cmake, g++/gcc are not in there by default! I had no real trouble installing libraries and packages, sometimes it took me a bit to figure out which one is needed. The closest thing to a 'special' library is pangolin since I compiled that from source and NOTE that according to some of the issues you need to use v0.5 release for pangolin, not the current release. I use the normal eigen3 from apt (it produces a lot of warnings for depricated stuff but it compiles fine). I compiled opencv from source, (note you need version 3.x for the master repository, however, there are some pull requests and forks that are updated to opencv 4.x). Just be careful since some forks may use orbslam v0.3 or v0.2 instead of the current v0.4.

Unfortunately, I did not keep track of what I installed and did to get it all running and since I have already done other projects in WSL you might need to install more than I did.

I would start by building the pangolin and opencv from source. Make sure you have them installed with the options and libraries you want for other projects too! Deal with whatever libraries are missing for those libraries, then go down the remaining list for Orb Slam 3.

For displaying things in windows from WSL I am using x11 using vcxsrv.
I added the following to my .bashrc file to resolve some warnings while the program is running.
export LIBGL_ALWAYS_INDIRECT=1
export DISPLAY=127.0.0.1:0.0

As for performance I don't have any lagging with opengl despite it not using the gpu. (i7-8565U CPU 8GB ram)

Maximvdw, 2021-10-04:
I use ORB-slam3 in a docker container supported by wsl2 (under Windows 11 technical preview), it works fine.
I followed these instructions to make the nvidia gpu available:
https://docs.nvidia.com/cuda/wsl-user-guide/index.html