UZ-SLAMLab/ORB_SLAM3 #387: error: ‘slots_reference’ was not declared in this scope

RG2806, 2021-09-20:
/home/rohan/Pangolin/components/pango_core/include/sigslot/signal.hpp:1180:65: error: ‘slots_reference’ was not declared in this scope 
1180 |         cow_copy_type<list_type, Lockable> ref = slots_reference();
      |                                                  ~~~~~~~~~~~~~~~^~
make[2]: *** [CMakeFiles/ORB_SLAM3.dir/build.make:104: CMakeFiles/ORB_SLAM3.dir/src/LocalMapping.cc.o] Error 1
make[2]: *** [CMakeFiles/ORB_SLAM3.dir/build.make:272: CMakeFiles/ORB_SLAM3.dir/src/Frame.cc.o] Error 1
make[2]: *** [CMakeFiles/ORB_SLAM3.dir/build.make:118: CMakeFiles/ORB_SLAM3.dir/src/LoopClosing.cc.o] Error 1
make[2]: *** [CMakeFiles/ORB_SLAM3.dir/build.make:90: CMakeFiles/ORB_SLAM3.dir/src/Tracking.cc.o] Error 1
make[2]: *** [CMakeFiles/ORB_SLAM3.dir/build.make:356: CMakeFiles/ORB_SLAM3.dir/src/G2oTypes.cc.o] Error 1
make[2]: *** [CMakeFiles/ORB_SLAM3.dir/build.make:258: CMakeFiles/ORB_SLAM3.dir/src/Optimizer.cc.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:358: CMakeFiles/ORB_SLAM3.dir/all] Error 2
make: *** [Makefile:91: all] Error 2

owolabioromidayo, 2021-09-24:
were you able to fix it?

RG2806, 2021-09-24:
yes, you need to change the compiler version in cmakelist.txt from c++10 -> c++14

closed 2021-09-24 by RG2806

maghauke, 2021-09-28:
@RG2806 Do you mean c++11 - > c++14?

RG2806, 2021-09-28:
yes,

senliontec, 2021-10-15:
@RG2806 Can you give some more details? Thanks alot.

Matthew-Bradley, 2021-11-03:
> This worked for me:
> 
> ```
> sed -i 's/++11/++14/g' CMakeLists.txt
> ```
> 
> It's a bit hacky since the demos use `#ifdef COMPILEDWITHC11` but it works.

Do the demos still work?

rgreid, 2021-11-03:
> Do the demos still work? If not, is there a way to make them work?

They all build fine after `sed -i 's/++11/++14/g' CMakeLists.txt` and I can confirm the TUM examples execute fine.

cly20-cqu, 2021-12-01:
This didn't work for me

si-fahad-nadeem, 2022-01-03:
> This worked for me:
> 
> ```
> sed -i 's/++11/++14/g' CMakeLists.txt
> ```
> 
> It's a bit hacky since the demos use `#ifdef COMPILEDWITHC11` but it works.

worked for me

Mechazo11, 2022-03-11:
> sed -i 's/++11/++14/g' CMakeLists.txt

@rgreid This also worked for me. Thank you so much. But I was wondering, what is this command doing to the CMakeLists ?

burhanbs, 2022-03-11:
> > sed -i 's/++11/++14/g' CMakeLists.txt
> 
> @rgreid This also worked for me. Thank you so much. But I was wondering, what is this command doing to the CMakeLists ?

hello Mecazo,
could you please tell me where to add this command in cmakelist.txt. i am confusing..??

thanks alot

Matthew-Bradley, 2022-03-11:
> > > sed -i 's/++11/++14/g' CMakeLists.txt
> > 
> > 
> > @rgreid This also worked for me. Thank you so much. But I was wondering, what is this command doing to the CMakeLists ?
> 
> hello Mechazo, could you please tell me where to add this command in cmakelist.txt. i am confusing..??
> 
> thanks alot

This is a command to run in the terminal that will replace "++11" with "++14" in the CMakeLists.txt files. You do not put it int the files themselves.

jc211, 2022-03-21:
In CMakeLists.txt, remove
```  
# Check C++11 or C++0x support
include(CheckCXXCompilerFlag)
CHECK_CXX_COMPILER_FLAG("-std=c++11" COMPILER_SUPPORTS_CXX11)
CHECK_CXX_COMPILER_FLAG("-std=c++0x" COMPILER_SUPPORTS_CXX0X)
if(COMPILER_SUPPORTS_CXX11)
   set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -std=c++11")
   add_definitions(-DCOMPILEDWITHC11)
   message(STATUS "Using flag -std=c++11.")
elseif(COMPILER_SUPPORTS_CXX0X)
   set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -std=c++0x")
   add_definitions(-DCOMPILEDWITHC0X)
   message(STATUS "Using flag -std=c++0x.")
else()
   message(FATAL_ERROR "The compiler ${CMAKE_CXX_COMPILER} has no C++11 support. Please use a different C++ compiler.")
endif()
```

and add 
```
set(CMAKE_CXX_STANDARD 14)
set(CMAKE_CXX_EXTENSIONS OFF)
```

Kin-Zhang, 2022-05-13:
> In CMakeLists.txt, remove
>
> ```
> # Check C++11 or C++0x support
> include(CheckCXXCompilerFlag)
> CHECK_CXX_COMPILER_FLAG("-std=c++11" COMPILER_SUPPORTS_CXX11)
> CHECK_CXX_COMPILER_FLAG("-std=c++0x" [...]

It works for me. Thanks!

sameer196, 2022-09-20:
> This worked for me:
> 
> ```
> sed -i 's/++11/++14/g' CMakeLists.txt

> I was able to compile and excecute all the demos

baimo0001, 2023-11-18:
`sed -i 's/++11/++14/g' CMakeLists.txt`
This worked for me, but you should use "make" instead of "make -j4". I do not know the reason.

Layort, 2025-03-25:
> This worked for me:
> 
> ```
> sed -i 's/++11/++14/g' CMakeLists.txt
> ```
> 
> It's a bit hacky since the demos use `#ifdef COMPILEDWITHC11` but it works.

It works!