UZ-SLAMLab/ORB_SLAM3 #388: /usr/bin/ld: cannot find -lpybind11::embed

RG2806, 2021-09-20:
[ 39%] Linking CXX shared library ../lib/libORB_SLAM3.so
**/usr/bin/ld: cannot find -lpybind11::embed**
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/ORB_SLAM3.dir/build.make:572: ../lib/libORB_SLAM3.so] Error 1
make[1]: *** [CMakeFiles/Makefile2:358: CMakeFiles/ORB_SLAM3.dir/all] Error 2
make: *** [Makefile:91: all] Error 2

RG2806, 2021-09-20:
Fixed by adding
find_package(pybind11 REQUIRED)
to CMakeLists.txt

closed 2021-09-20 by RG2806

Matthew-Bradley, 2021-11-03:
@RG2806 Where did you add it to the file? Did you add anything else? Install anything else?

I'm having this issue too, but when I simply add "find_package(pybind11 REQUIRED)" to "CMakeLists.txt" I get:

-- Compiling on Unix
CMake Error at CMakeLists.txt:197 (find_package):
  By not providing "Findpybind11.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "pybind11",
  but CMake did not find one.

  Could not find a package configuration file provided by "pybind11" with any
  of the following names:

    pybind11Config.cmake
    pybind11-config.cmake

  Add the installation prefix of "pybind11" to CMAKE_PREFIX_PATH or set
  "pybind11_DIR" to a directory containing one of the above files.  If
  "pybind11" provides a separate development package or SDK, be sure it has
  been installed.

-- Configuring incomplete, errors occurred!
See also "/home/matt/Documents/waterloo/slam_on_vpr_dataset/orbslam3/ORB_SLAM3/build/CMakeFiles/CMakeOutput.log".
See also "/home/matt/Documents/waterloo/slam_on_vpr_dataset/orbslam3/ORB_SLAM3/build/CMakeFiles/CMakeError.log".
make: *** [Makefile:1468: cmake_check_build_system] Error 1

Matthew-Bradley, 2021-11-03:
I did "conda install pybind11" and that provided the needed cmake files. After that I just had to add "find_package(pybind11 REQUIRED)" to CMakeLists.txt which I did under the entries for pangolin and other orbslam dependencies.