UZ-SLAMLab/ORB_SLAM3 #390: Export point cloud

UcefMountacer, 2021-09-22:
Hi,

Has anyone tried to export the point cloud after processing data? there are some functions to export the trajectory but nothing for point cloud.

It would be great to export it like the trajectory that already have a function for that: 

`SLAM.SaveTrajectoryTUM("CameraTrajectory.txt");`

NickPerezCarletonUniversity, 2021-09-30:
I am also interested in this.

Mateo-tech, 2021-11-13:
Also interested.

NJNischal, 2023-03-08:
Hello everyone, have any of you figured out how to save the point cloud points from ORBSLAM3?

WFram, 2023-03-14:
For example, you can use ROS and PCL library for this purpose.

Just use functions `GetCurrentMap()` and `GetMapPoints()`, it will give a set of landmarks. Which can be then converted to `PointCloudXYZ` (data structure in PCL library to store a point cloud). Once the mapping has finished, the point cloud can be saved to `.pcd` file using PCL I/O interface.

There is a raw example of how to do it: https://github.com/WFram/vslam2/blob/98ddf36524b162664ec71cc9269eaaf83a560769/ORB_SLAM3_ROS/src/viewer.cpp#L216

altaykacan, 2024-01-02:
Hey all,

a bit late to the discussion but I implemented a function inspired by @WFram 's example that works similarly to the `SaveTrajectory` functions.

Here's my code:
```cpp
void System::SavePointCloud(const string &filename){
    // Code is based on MapDrawer::DrawMapPoints()
    cout << endl << "Saving map point coordinates to " << filename << " ..." << endl;
    cout << endl << "Number of maps is: " << mpAtlas->CountMaps() << endl;

    // TODO Get all maps or is the current active map is enough?
    // vector<Map*> vpAllMaps = mpAtlas->GetAllMaps()

    Map* pActiveMap = mpAtlas->GetCurrentMap();
    if(!pActiveMap) {
        cout << endl << "There is no active map (pActiveMap is null)" << endl;
        return;
    }

    // Vectors containing pointers to MapPoint objects contained in the maps
    // Vector of pointers for Map Points -- vpMPs
    // Vector of pointers for Reference Map Points -- vpRefMPs
    // TODO figure out the difference between Reference Map Points and normal Map Points
    const vector<MapPoint*> &vpMPs = pActiveMap->GetAllMapPoints();
    const vector<MapPoint*> &vpRefMPs = pActiveMap->GetReferenceMapPoints();

    if(vpMPs.empty()){
        cout << endl << "Vector of map points vpMPs is empty!" << endl;
        return;
    }

    // Use a set for fast lookup of reference frames
    set<MapPoint*> spRefMPs(vpRefMPs.begin(), vpRefMPs.end());

    // Get the output file stream in fixed-point format for map points
    ofstream f;
    f << "pos_x, pos_y, pos_z";
    f.open(filename.c_str());
    f << fixed;

    // TODO figure out if we need to consider whether the presence of IMU
    // requires some transforms/exceptions

    // Iterate over map points, skip "bad" ones and reference map points
    for (size_t i=0, iend=vpMPs.size(); i<iend;i++)
    {
        if (vpMPs[i]->isBad() || spRefMPs.count(vpMPs[i])){
            continue;
        }
        Eigen::Matrix<float,3,1> pos = vpMPs[i]->GetWorldPos();
        f << pos(0) << ", " << pos(1) << ", " << pos(2) << endl;
    }

    // Close the output stream
    f.close();

    // Get the output file stream in fixed-point format for reference map points
    f.open(("ref_" + filename).c_str());
    f << "pos_x, pos_y, pos_z" << endl;
    f << fixed;

    // Iterate over reference map points, skip if bad
    for (set<MapPoint*>::iterator sit=spRefMPs.begin(), send=spRefMPs.end(); sit!=send; sit++)
    {
        if((*sit)->isBad()){
            continue;
        }
        Eigen::Matrix<float,3,1> pos = (*sit)->GetWorldPos();
        f << pos(0) << ", " << pos(1) << ", " << pos(2) << endl;
[... 88 lines cut ...]
                'property float y\n' \
                'property float z\n' \
                'property uchar red\n' \
                'property uchar green\n' \
                'property uchar blue\n' \
                'end_header' % x.shape

    np.savetxt(output_file, np.column_stack((x, y, z, r, g, b)), fmt='%f %f %f %d %d %d', header=ply_header, comments='')

if __name__ == "__main__":
    input_file = "/path/to/point_cloud.txt" # ReferenceMapPoints seem to work better, use that file
    output_file = "./output.ply"

    input_trajectory = "/path/to/trajectory_file.txt"
    output_trajectory = "./output_trajectory.ply"

    save_pointcloud_from_ORB_SLAM(input_file, output_file)
    save_trajectory_from_ORB_SLAM(input_trajectory, output_trajectory)
```

After plotting the two `.ply` files on CloudCompare I got results like this on custom data, which shows a camera moving straight. So I believe the code works:

[image]

I know the answer a bit long and I'm no C++ expert but I really hope this helps! Please let me know if you notice any mistakes, we can maybe create a pull request later on. It seems like a very popular issue.

jonaug88, 2024-03-11:
> Hey all,
> 
> a bit late to the discussion but I implemented a function inspired by @WFram 's example that works similarly to the `SaveTrajectory` functions.
> 
> Here's my code:
> 
> ```c++
> void System::SavePointCloud(const string &filename){
>     // Code is based on MapDrawer::DrawMapPoints()
>     cout << endl << "Saving map point coordinates to " << filename << " ..." << endl;
>     cout << endl << "Number of maps is: " << mpAtlas->CountMaps() << endl;
> 
>     // TODO Get all maps or is the current active map is enough?
>     // vector<Map*> vpAllMaps = mpAtlas->GetAllMaps()
> 
>     Map* pActiveMap = mpAtlas->GetCurrentMap();
>     if(!pActiveMap) {
>         cout << endl << "There is no active map (pActiveMap is null)" << endl;
>         return;
>     }
> 
>     // Vectors containing pointers to MapPoint objects contained in the maps
>     // Vector of pointers for Map Points -- vpMPs
>     // Vector of pointers for Reference Map Points -- vpRefMPs
>     // TODO figure out the difference between Reference Map Points and normal Map Points
>     const vector<MapPoint*> &vpMPs = pActiveMap->GetAllMapPoints();
>     const vector<MapPoint*> &vpRefMPs = pActiveMap->GetReferenceMapPoints();
> 
>     if(vpMPs.empty()){
>         cout << endl << "Vector of map points vpMPs is empty!" << endl;
>         return;
>     }
> 
>     // Use a set for fast lookup of reference frames
>     set<MapPoint*> spRefMPs(vpRefMPs.begin(), vpRefMPs.end());
> 
>     // Get the output file stream in fixed-point format for map points
>     ofstream f;
>     f << "pos_x, pos_y, pos_z";
>     f.open(filename.c_str());
>     f << fixed;
> 
>     // TODO figure out if we need to consider whether the presence of IMU
>     // requires some transforms/exceptions
> 
>     // Iterate over map points, skip "bad" ones and reference map points
>     for (size_t i=0, iend=vpMPs.size(); i<iend;i++)
>     {
>         if (vpMPs[i]->isBad() || spRefMPs.count(vpMPs[i])){
>             continue;
>         }
>         Eigen::Matrix<float,3,1> pos = vpMPs[i]->GetWorldPos();
>         f << pos(0) << ", " << pos(1) << ", " << pos(2) << endl;
>     }
> 
>     // Close the output stream
>     f.close();
> 
>     // Get the output file stream in fixed-point format for reference map points
>     f.open(("ref_" + filename).c_str());
>     f << "pos_x, pos_y, pos_z" << endl;
>     f << fixed;
> 
>     // Iterate over reference map points, skip if bad
>     for (set<MapPoint*>::iterator sit=spRefMPs.begin(), send=spRefMPs.end(); sit!=send; sit++)
>     {
>         if((*sit)->isBad()){
>             continue;
>         }
>         Eigen::Matrix<float,3,1> pos = (*sit)->GetWorldPos();
[... 100 lines cut ...]
>     input_file = "/path/to/point_cloud.txt" # ReferenceMapPoints seem to work better, use that file
>     output_file = "./output.ply"
> 
>     input_trajectory = "/path/to/trajectory_file.txt"
>     output_trajectory = "./output_trajectory.ply"
> 
>     save_pointcloud_from_ORB_SLAM(input_file, output_file)
>     save_trajectory_from_ORB_SLAM(input_trajectory, output_trajectory)
> ```
> 
> After plotting the two `.ply` files on CloudCompare I got results like this on custom data, which shows a camera moving straight. So I believe the code works:
> 
> ![image](https://private-user-images.githubusercontent.com/33330747/293769450-57a535ee-c47e-4f2e-89a3-31cc03ef4703.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTAxNDc1MjAsIm5iZiI6MTcxMDE0NzIyMCwicGF0aCI6Ii8zMzMzMDc0Ny8yOTM3Njk0NTAtNTdhNTM1ZWUtYzQ3ZS00ZjJlLTg5YTMtMzFjYzAzZWY0NzAzLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAzMTElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMzExVDA4NTM0MFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWZmY2I0ZTE0NDE2NDM4ZmYzNjIwOGUyMjQ5OTk3M2RlZDIxN2FjYWM5ZDlmZTk1NTE3YzU5YzlkZjdkODZjYjkmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.FTMHh8QISRfys4B4mE2MEhNqCUxm5beB4E5qAquCxYU)
> 
> I know the answer a bit long and I'm no C++ expert but I really hope this helps! Please let me know if you notice any mistakes, we can maybe create a pull request later on. It seems like a very popular issue.

I am terribly sorry for ressurecting this topic, but i have a question regarding your solution. 
I have added your code and function signature but the segment: "and finally calling the function in whatever program you are using after SLAM.Shutdown() should hopefully work." confuse me abit. This is how i run it:
term 1: roscore
term 2: rosbag play --clock blabla.bag
term 3: roslaunch orb_slam3_ros tello.launch

Should there be something added in a terminal 4 to call SLAM.shutdown and the function you implemented? 

I also tried saving the map using rosservice call, but the .osa file extension gave me more questions than i could answer.

altaykacan, 2024-03-18:
> > Hey all,
> > a bit late to the discussion but I implemented a function inspired by @WFram 's example that works similarly to the `SaveTrajectory` functions.
> > Here's my code:
> > After plotting the two `.ply` files on CloudCompare I got results like this on custom data, which shows a camera moving straight. So I believe the code works:
> 
> > I know the answer a bit long and I'm no C++ expert but I really hope this helps! Please let me know if you notice any mistakes, we can maybe create a pull request later on. It seems like a very popular issue.
> 
> I am terribly sorry for ressurecting this topic, but i have a question regarding your solution. I implemented your changes to both System.cc and System.hm but when i stop the Map Viewer it saves the trajectory to CameraTrajectory.txt aswell as the keyframe trajectory. The point cloud itself does not save. Is there a way to add saving the point cloud in the same manner, or is there another way you'd rather enforce the SLAM.Shutdown() ?
> 
> Here is the output when stopping: " Shutdown
> 
> Saving trajectory to CameraTrajectory.txt ... There are 2 maps in the atlas Map 0 has 42 KFs Map 1 has 86 KFs
> 
> End of saving trajectory to CameraTrajectory.txt ...
> 
> Saving keyframe trajectory to KeyFrameTrajectory.txt ... "
> 
> I've made sure that the placement of "SavePointCloud" function is between the "SaveTrajectoryEuRoC" and SaveKeyFrameTrajectoryEuRoC functions as it seems that the output is originating from there, it seems like it doesnt initiate the Savepointcloud, as i dont even get the "couts". Do you have any ideas?
> 
> Thanks again for your hard work.

Hey! Thank you for your question, I'd be very happy if I can help. It seems weird that even the first `cout` prints don't show up. Can you try adding your own print statements before and after you call `SavePointCloud()`? If those don't show up that means some other code is being executed and you should have a look there.

I think even if there is some bug in my hacky solution the first `cout << endl << "Saving map point coordinates to " << filename << " ..." << endl;` should definitely show up on the terminal.

Also, I know this is very simple but did you build the project again after making your adjustments? Otherwise your executable wouldn't get updated as far as I know. I had to learn that the hard way when I switched to C++ from coding in Python for years :D

jonaug88, 2024-03-19:
> Hey! Thank you for your question, I'd be very happy if I can help. It seems weird that even the first `cout` prints don't show up. Can you try adding your own print statements before and after you [...]

Hello and thank you for getting back to me! 
First of, i would like to say, I almost kicked myself in anger from the possibility of it being such an easy fix as rebuilding the workspace... 
That being said, it seemed to bring me closer to my goal =)
When i stop the orb slam, i still get the same output, however, when i killed the process using CTRL + C, it followed up by actually involving the SavePointCloud call. 

"^C[orb_slam3_ros/trajectory_server_orb_slam3-3] killing on exit
[rviz-2] killing on exit
[orb_slam3-1] killing on exit

Saving point cloud #My own cout before the SavePointCloud() call.

Saving map point coordinates to /home/jonaug/orb_master/maps ...

Number of maps is: 1

Point cloud saved #My own added cout after SavePointCloud() call
Shutdown
"
The map didnt get saved to that path, but i am under the impression that it cant "track" the pointcloud data after the process is killed as it does not have any "memory" regarding the cloud, and it has to be called before the process termination ? 

Is it the placement of the call that is the issue or what am i missing? 

Thanks again

altaykacan, 2024-03-19:
Hmmm it seems like you're using ROS, I have never used that before so can't really help you there. From what I can tell by looking at [WFram's comment](https://github.com/UZ-SLAMLab/ORB_SLAM3/issues/390#issuecomment-1468825465) you have to publish the cloud somehow but I am not sure?

Maybe you can look for a loop like the `viewer::run()` like WFram has in his example and figure out where to put the pointcloud saving code. His implementation is probably a better reference if you're using ROS.

You might be right with the process being killed and the function not being able to find the memory, maybe somewhere in the code related to ROS (I have no idea) the map or the pointers get cleared. I guess you can also try putting more print statements in `SavePointCloud()` to see what exactly is missing.

Good luck! Hopefully it works out

jonaug88, 2024-03-19:
> Hmmm it seems like you're using ROS, I have never used that before so can't really help you there. From what I can tell by looking at [WFram's [...]

Thanks for taking your time to reply man, much appreciated, ill see if i can hook WFram for some more specifics as i am not that well versed in this myself =)

jonaug88, 2024-03-19:
> For example, you can use ROS and PCL library for this purpose.
>
> Just use functions `GetCurrentMap()` and `GetMapPoints()`, it will give a set of landmarks. Which can be then converted to [...]

I am sorry for pulling you back into this again a year after your contribution. Are you able to expand abit on this please? I have not been able to replicate your example...

RIPGuy, 2024-03-20:
> Hey all,
>
> a bit late to the discussion but I implemented a function inspired by @WFram 's example that works similarly to the `SaveTrajectory` functions.
>
> Here's my code:
>
> ```c++
> void [...]

Hello! I am new to ORBSLAM and planning to export the mappoints after processing. You said that first code export a .csv file for the mappoints, what are the values I should expect from its output are they like xyz values that correspond to each point in the map?

altaykacan, 2024-03-20:
> Hello! I am new to ORBSLAM and planning to export the mappoints after processing. You said that first code export a .csv file for the mappoints, what are the values I should expect from its output are they like xyz values that correspond to each point in the map?

Yup exactly, this code writes the x, y, and z positions of each point to the output file you specify line-by-line: 

```cpp
  Eigen::Matrix<float,3,1> pos = vpMPs[i]->GetWorldPos();
  f << pos(0) << ", " << pos(1) << ", " << pos(2) << endl;
```

Each line of the output (except the first line which has the name of the columns) should contain the coordinates for a single point. It worked for me, I hope it works for you too! :)

RIPGuy, 2024-03-30:
> > Hello! I am new to ORBSLAM and planning to export the mappoints after processing. You said that first code export a .csv file for the mappoints, what are the values I should expect from its output are they like xyz values that correspond to each point in the map?
> 
> Yup exactly, this code writes the x, y, and z positions of each point to the output file you specify line-by-line:
> 
> ```c++
>   Eigen::Matrix<float,3,1> pos = vpMPs[i]->GetWorldPos();
>   f << pos(0) << ", " << pos(1) << ", " << pos(2) << endl;
> ```
> 
> Each line of the output (except the first line which has the name of the columns) should contain the coordinates for a single point. It worked for me, I hope it works for you too! :)

Thank you for replying. I haven't tested it yet as I am having a hard time trying to test ORBSLAM3 on my custom data set. Can you please help me on this?

So far here is what I have done but I do not see it running at all.

I downloaded a dataset I got online. It is just a video which I downloaded.

**1. Generate KITTI sequence**
Once downloaded, generate the sequences. This can be done using the video2sequences.py script.

python video2sequences.py

**Place in a folder**
This script also generates the timestamp file, times.txt. This is also a Kitti requirement. Put the sequences in datasets/urban_dataset/sequences/XX/image_0" folder. Here XX is the sequence number. Now, paste the generated timestamp in the datasets/urban_dataset/sequences/XX/ folder.

**Run the script**
KITTI Format:
[mono_kitti] [vocabulary.txt] [yaml configuration] [path to sequence]
		
./Examples/Monocular/mono_kitti Vocabulary/ORBvoc.txt Examples/Monocular/KITTI00-02.yaml datasets/urban_dataset/sequences/00

I have performed them but there is no viewer popping up or output at all. Might you suggest another way to do it?

migsdigs, 2024-10-14:
Hi @altaykacan! Thank you so much for your code to save the point cloud. It looks as though it works (I am able to save txt with x,y,z positions of points). I was wondering if you made an attempt at also saving the RGB values of the points, or if you'd know how to approach this?

I was originally using a docker image that had Python wrapper of Orbslam. They were able to save the pointcloud along with the RGB values of the points to create a nice sparse pointcloud. Unfortunately that module doesnt seem to support Inertial assisted slam. 

If you're not sure, no worries, thank you very much for your code in any case :)

AndreasGZ, 2024-10-17:
Thank you @altaykacan.
I rewrote the Code, so that every Map gets saved and  I changed the saving-format of the pointcloud to a ply-format based on your python code.

```
void System::SavePointCloud(const string &filename){
    cout << endl << "Saving pointclouds to " << filename << " ..." << endl;
    int size = mpAtlas->CountMaps();
    cout << endl << "Number of maps is: " << size << endl;
    vector<Map*> vpAllMaps = mpAtlas->GetAllMaps();
    for(size_t i=0; i<size; i++){
        Map* vpMap = vpAllMaps[i];
        if(!vpMap) continue;
        const vector<MapPoint*> &vpMPs = vpMap->GetAllMapPoints();
        const vector<MapPoint*> &vpRefMPs = vpMap->GetReferenceMapPoints();
        set<MapPoint*> spRefMPs(vpRefMPs.begin(), vpRefMPs.end());
        if(vpMPs.empty()) continue;
        string nr = to_string(i) + "_"; 
        int count = 0;
        stringstream mapss("");
        mapss << fixed;
        for (size_t i=0, iend=vpMPs.size(); i<iend;i++)
        {
            if (vpMPs[i]->isBad() || spRefMPs.count(vpMPs[i])){
                continue;
            }
            count++;
            Eigen::Matrix<float,3,1> pos = vpMPs[i]->GetWorldPos();
            mapss << pos(0) << " " << pos(1) << " " << pos(2) << "\n";
        }
        // Farbe für Punkte -> "property uchar red\nproperty uchar green\nproperty uchar blue\n";
        ofstream f;
        if(count > 0) {
            f.open((nr + filename).c_str());
            f << "ply\nformat ascii 1.0\nelement vertex " << count << "\n";
            f << "property float x\nproperty float y\nproperty float z\n";
            f << "end_header\n";
            f << fixed;
            f << mapss.str();
            f.close();
            mapss.str("");
            mapss.clear();
        }

        count = 0;
        mapss << fixed;
        for (set<MapPoint*>::iterator sit=spRefMPs.begin(), send=spRefMPs.end(); sit!=send; sit++)
        {
            if((*sit)->isBad()){
                continue;
            }
            count++;
            Eigen::Matrix<float,3,1> pos = (*sit)->GetWorldPos();
            mapss << pos(0) << " " << pos(1) << " " << pos(2)  << "\n";
        }
        if(count > 0){
            f.open((nr + "ref_" + filename).c_str());
            f << "ply\nformat ascii 1.0\nelement vertex " << count << "\n";
            f << "property float x\nproperty float y\nproperty float z\n";
            f << "end_header\n";
            f << fixed;
            f << mapss.str();
            f.close();
            mapss.str("");
            mapss.clear();
        }
        
    }
}
```

altaykacan, 2024-10-20:
> Hi @altaykacan! Thank you so much for your code to save the point cloud. It looks as though it works (I am able to save txt with x,y,z positions of points). I was wondering if you made an attempt at [...]

Hey @migsdigs 

I'm glad my code helped you out! I didn't really work on getting the RGB values but it should be pretty straightforward if the `MapPoint` objects include the color as attributes. I guess ORB-SLAM uses gray-scale images though, so that might be an issue. If the RGB images are saved somewhere, you can probably take the pixel coordinates of the detected feature points and assign it to the 3D point and just add it to the final output when saving the map. What I recommend you to do would be to check the code from the docker image you were using before.

Another thing you could try would be to colorize the pointcloud after you get the sparse map and the poses, as a post-processing step. With a simple python script, I believe you can basically project your pointcloud to the RGB images, compare the projected points with your RGB images, and assign colors to each point. Open3D has lots of functions that would help you do this. This is probably not be best solution though and won't give you a real-time colored map.

 I'm not working on this project anymore so I unfortunately can't check it or implement it myself. I hope this still helps you out though :)

LaFeuilleMorte, 2024-12-03:
The saved point cloud seemed to be too sparse to view the basic shape of the reconstructed scene. Here's what I got from tum-vi datasets:
[image]

Any clues on getting denser points?

LaFeuilleMorte, 2024-12-03:
> The saved point cloud seemed to be too sparse to view the basic shape of the reconstructed scene. Here's what I got from tum-vi datasets: [...]

Alright, It seemed that I've exported the reference map points.