UZ-SLAMLab/ORB_SLAM3 #393: Initialising map everytime it losts track

tejalbarnwal, 2021-09-24:
The feature that makes a new pose trajectory graph every time the algorithm loses track. Can this feature be disabled somehow? and it continues to stitch the upcoming estimation after it lost track to the pose graph that it made before losing the track?

UcefMountacer, 2021-09-24:
totally agree with you on this. In fact, it not only initialize point cloud, but also trajectory. This one can become really messy.

I would like to have a solution too.

By the way @tejalbarnwal, were you able to export point cloud? if so, I'd like to know how. Thanks.

tejalbarnwal, 2021-09-25:
I havent done it, but I saw some code snippet which could help you maybe ` std::vector<MapPoint*> GetTrackedMapPoints()`
    and `std::vector<cv::KeyPoint> GetTrackedKeyPointsUn()` inside System.h
They have used in examples inside `ros_mono_ar.cc`

lktidaohuoxing, 2022-03-16:
@tejalbarnwal  i want to keep a global Tcw (when it lost), but i fall.  i only change the Tcw before  CreateInitialMapMonocular() , and orbslam3 cannot create a new map, what should i do ? what else should i modify ? Sorry, im not good at English.

tejalbarnwal, 2022-03-19:
@lktidaohuoxing Sorry, I didn't really look into the source code extensively to rectify the problem.  
But in case you just want to store the `Tcw` of the current frame, you can write your function to save it in a text file and then later you can make manipulations over that data to obtain all the points in one frame.
Further, it might be possible that the code isn't able to create a keyframe with the new `tcw` inside the `CreateInitialMapMonocular()`. I urge you to add `cout` statements  before and inside `CreateInitialMapMonocular()` and `CreateMapInAtlas()`
We would be very grateful if contributors to the repository @richard-elvira @jdtardos could help us with this.

chetanpandey1266, 2023-09-03:
Was someone able to work this out? I am not able to figure out how to not reset the map over a video, so that I can get the whole map in one file only

chetanpandey1266, 2023-09-03:
Have anyone worked on this and made it work ?

tejalbarnwal, 2023-09-03:
Hey @chetanpandey1266 ,
I used the repo for some student competition 2 years back and havent looked at it very much since then.