UZ-SLAMLab/ORB_SLAM3 #397: CMake error at the line find_package(Pangolin REQUIRED)

lafith, 2021-09-29:
There were no errors while installing the prerequisites. But when I ran ./build.sh following error is coming:

```
Build type: Release
-- Performing Test COMPILER_SUPPORTS_CXX11
-- Performing Test COMPILER_SUPPORTS_CXX11 - Success
-- Performing Test COMPILER_SUPPORTS_CXX0X
-- Performing Test COMPILER_SUPPORTS_CXX0X - Success
-- Using flag -std=c++11.
-- Found OpenCV: /usr/local (found suitable version "4.5.3", minimum required is "4.0")
OPENCV VERSION:
4.5.3
CMake Error at CMakeLists.txt:49 (find_package):
  By not providing "FindPangolin.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "Pangolin",
  but CMake did not find one.

  Could not find a package configuration file provided by "Pangolin" with any
  of the following names:

    PangolinConfig.cmake
    pangolin-config.cmake

  Add the installation prefix of "Pangolin" to CMAKE_PREFIX_PATH or set
  "Pangolin_DIR" to a directory containing one of the above files.  If
  "Pangolin" provides a separate development package or SDK, be sure it has
  been installed.
```

Maximvdw, 2021-10-02:
Have you installed Pangolin. If yes- did you set Pangolin_DIR to the install dir?

lafith, 2021-10-04:
A small change in Pangolin build instructions solved the issue.
Instead of this:
```
# Configure and build
mkdir build && cd build
cmake ..
cmake --build .
```
following fixed the issue:
```
# Configure and build
mkdir build && cd build
cmake ..
make
sudo make install
```

closed 2021-10-04 by lafith