UZ-SLAMLab/ORB_SLAM3 #398: Build crash ubuntu totally！

bluebird1999, 2021-09-30:
Try to build the SLAM3 on ubuntu 20.04, I build the opencv, eigen and Pangolin and then run the build.sh, during the process, it give me the error of missing boost, and I manually installed the boost, and then always hung during building, I have to forcefully reboot.

could anyone give me some clues? thanks!

Sherlock-hh, 2021-09-30:
I have the same problem, it's start crash when compiled optimizer.cc, I saw some solution which use "make" only , without "-j", which can help. maybe you can try it.
But I tried that way, it's not help.

makintuo, 2021-10-02:
Try following the instructions here https://github.com/Mauhing/ORB_SLAM3/blob/master/README.md

See also this video https://www.youtube.com/watch?v=DxqzwBQVCNw

(Not sure if it's the same problem but this helped me on 20.04)

nassim12, 2021-10-06:
same issue with ubuntu 18.04 when I build version 04. but not with version 03