UZ-SLAMLab/ORB_SLAM3 #399: libpango_windowing.so

NhlalalaDylan, 2021-09-30:
when I run :  ./Examples/Monocular/mono_euroc ./Vocabulary/ORBvoc.txt ./Examples/Monocular/EuRoC.yaml ~/Datasets/EuRoc/MH01 ./Examples/Monocular/EuRoC_TimeStamps/MH01.txt dataset-MH01_mono

I get:
./Examples/Monocular/mono_euroc: error while loading shared libraries: libpango_windowing.so: cannot open shared object file: No such file or directory

maghauke, 2021-09-30:
sudo ldconfig

NhlalalaDylan, 2021-09-30:
Thanks a lot

closed 2021-09-30 by NhlalalaDylan

Mechazo11, 2022-03-06:
@maghauke Thank you very much for providing the solution. I am a novice when it comes to using C++ projects and was wondering what is ldconfig doing in context of ORB-SLAM3?

AllenIsaacJose, 2024-02-17:
> @maghauke Thank you very much for providing the solution. I am a novice when it comes to using C++ projects and was wondering what is ldconfig doing in context of ORB-SLAM3?

After installing Pangolin libraries, it's a good idea to update the library cache using this command. Pangolin is the visualizing library used in ORB-SLAM 1, 2, and 3.

jiyooonp, 2024-04-15:
Hello, I tried to run the command, but it is still giving me the error. Could someone help me? Thank you.

GLtangzero, 2024-07-05:
@jiyooonp 
Hey, first you need check which *.so document you can't find.
Then try to find the path of it.
For example I got the error like:
`error while loading shared libraries: libORB_SLAM3.so: cannot open shared object file: No such file or directory`
Then I found this .os document in /home/user/ORBSLAM3_ROS2_2/src/ORB_SLAM3/lib,
next input `export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/home/user/ORBSLAM3_ROS2_2/src/ORB_SLAM3/lib`
in your terminal or just write it in the `.bashrc` and `source ~/.bashrc`.
It maybe will work, please check it.

Strikerfury22, 2024-07-09:
@jiyooonp 
Hello, I was having the same problem. Once you have made what @GLtangzero said to declare your $LD_LIBRARY_PATH, execute "sudo ldconfig $LD_LIBRARY_PATH". In that way ldconfig will create the necessary links. In my case, my $LD_LIBRARY_PATH only contained the directory, so no ":" were in it. This is what it contained: "/path/to/my/lib". I hope this helps everyone in our same situation (I just managed to make it right now).

Adithya1911, 2025-03-11:
Hello, I was facing the problem, 
./test_webcam_2: error while loading shared libraries: libOpenNI2.so.0: cannot open shared object file: No such file or directory
Launching ORB_SLAM3 with Monocular sensor
when i try to run the ORBSLAM3 this is first time i am facing this issue