UZ-SLAMLab/ORB_SLAM3 #400: install on jetson nx

43998213, 2021-09-30:
g2o error
cmake ..
make
-- The C compiler identification is GNU 7.5.0
-- The CXX compiler identification is GNU 7.5.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Compiling on Unix
-- Found Cholmod: /usr/include/suitesparse
-- A cache variable, namely BLAS_DIR, has been set to specify the install directory of BLAS
-- Looking for BLAS -- mkl.h not found
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE
-- Looking for MKL BLAS: not found
-- Looking for Goto BLAS: not found
-- Looking for Open BLAS: not found
-- Looking for Eigen BLAS: not found
-- Looking for Eigen BLAS: not found
-- Looking for Atlas BLAS: not found
-- Looking for PhiPACK BLAS: not found
-- Looking for CXML BLAS: not found
-- Looking for DXML BLAS: not found
-- Looking for SunPerf BLAS: not found
-- Looking for SCSL BLAS: not found
-- Looking for SGIMATH BLAS: not found
-- Looking for IBM ESSL BLAS: not found
-- Looking for IBM ESSL MT BLAS: not found
-- Looking for ACML BLAS: not found
-- Looking for ACML BLAS: not found
-- Looking for ACML BLAS: not found
-- Looking for Apple BLAS: not found
-- Looking for NAS BLAS: not found
-- Looking for sgemm_
-- Looking for sgemm_ - found
-- Looking for Generic BLAS: found
-- A library with BLAS API found.
-- BLAS_LIBRARIES /usr/lib/aarch64-linux-gnu/libblas.so
-- Looking for BLAS -- mkl.h not found
-- Looking for MKL BLAS: not found
-- Looking for Goto BLAS: not found
-- Looking for Open BLAS: not found
-- Looking for Eigen BLAS: not found
-- Looking for Eigen BLAS: not found
-- Looking for Atlas BLAS: not found
-- Looking for PhiPACK BLAS: not found
-- Looking for CXML BLAS: not found
-- Looking for DXML BLAS: not found
-- Looking for SunPerf BLAS: not found
-- Looking for SCSL BLAS: not found
-- Looking for SGIMATH BLAS: not found
-- Looking for IBM ESSL BLAS: not found
-- Looking for IBM ESSL MT BLAS: not found
-- Looking for ACML BLAS: not found
[... 126 lines cut ...]
[ 28%] Building CXX object g2o/types/slam3d_addons/CMakeFiles/types_slam3d_addons.dir/line3d.cpp.o
[ 29%] Building CXX object g2o/types/slam3d_addons/CMakeFiles/types_slam3d_addons.dir/vertex_line3d.cpp.o
[ 29%] Building CXX object g2o/types/slam3d_addons/CMakeFiles/types_slam3d_addons.dir/edge_se3_line.cpp.o
[ 29%] Building CXX object g2o/types/slam3d_addons/CMakeFiles/types_slam3d_addons.dir/edge_plane.cpp.o
[ 30%] Building CXX object g2o/types/slam3d_addons/CMakeFiles/types_slam3d_addons.dir/edge_se3_calib.cpp.o
[ 30%] Building CXX object g2o/types/slam3d_addons/CMakeFiles/types_slam3d_addons.dir/types_slam3d_addons.cpp.o
c++: internal compiler error: 段错误 (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See file:///usr/share/doc/gcc-7/README.Bugs for instructions.
g2o/types/slam3d_addons/CMakeFiles/types_slam3d_addons.dir/build.make:278: recipe for target 'g2o/types/slam3d_addons/CMakeFiles/types_slam3d_addons.dir/types_slam3d_addons.cpp.o' failed
make[2]: *** [g2o/types/slam3d_addons/CMakeFiles/types_slam3d_addons.dir/types_slam3d_addons.cpp.o] Error 4
CMakeFiles/Makefile2:1417: recipe for target 'g2o/types/slam3d_addons/CMakeFiles/types_slam3d_addons.dir/all' failed
make[1]: *** [g2o/types/slam3d_addons/CMakeFiles/types_slam3d_addons.dir/all] Error 2
Makefile:151: recipe for target 'all' failed
make: *** [all] Error 2

I tried
sudo dd if=/dev/zero of=/swapfile bs=64M count=32
sudo mkswap /swapfile
chmod 0600 /swapfile
sudo swapon /swapfile
I tried gcc7. gcc6. gcc5
but all failed
help please

Chenliezhen259, 2021-11-07:
i also have similar questions. But i am installing orbslam3 on jetson nx, that is enough to compile source code. why i always meet a problem：segmentation fault (code dumped).

greymaner, 2021-11-30:
I also meet the same problem,and then I modified the build.sh ->make -j to make -j2,solved!