UZ-SLAMLab/ORB_SLAM3 #403: "Aborted (core dumped)" when running Ros example

maghauke, 2021-10-04:
Hi, following the ROS example, when I use the command:
`rosrun ORB_SLAM3 Stereo_Inertial Vocabulary/ORBvoc.txt Examples/Stereo-Inertial/EuRoC.yaml true`
I get the following output:
`ORB-SLAM3 Copyright (C) 2017-2020 Carlos Campos, Richard Elvira, Juan J. Gómez, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
ORB-SLAM2 Copyright (C) 2014-2016 Raúl Mur-Artal, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
This program comes with ABSOLUTELY NO WARRANTY;
This is free software, and you are welcome to redistribute it
under certain conditions. See LICENSE.txt.

Input sensor was set to: Stereo-Inertial

Loading ORB Vocabulary. This could take a while...
Vocabulary loaded!

Creation of new map with id: 0
Creation of new map with last KF id: 0
Seq. Name: 

Camera Parameters: 
- Camera: Pinhole
- fx: 435.205
- fy: 435.205
- cx: 367.452
- cy: 252.201
- k1: 0
- k2: 0
- p1: 0
- p2: 0
- fps: 20
- color order: RGB (ignored if grayscale)

Depth Threshold (Close/Far Points): 3.85272

ORB Extractor Parameters: 
- Number of Features: 1200
- Scale Levels: 8
- Scale Factor: 1.2
- Initial Fast Threshold: 20
- Minimum Fast Threshold: 7
double free or corruption (out)
Aborted (core dumped)
`
Can anyone give tips on how to fix/debug this?

KristianSem, 2021-10-10:
I have the same problem. Getting "Segmentation fault (core dumped)" error when running any ROS example.

FeiXie8, 2021-11-03:
> > @Albertcqq do you have quick guide for installin opencv 3.2.0?
> > yes,I have ,here it is :https://zhuanlan.zhihu.com/p/52513112

no,it is not the problem of opencv,it is the type of the data when refresh the map we get is not 32f and 64f,opencv is just detect the error

maghauke, 2021-11-03:
@FeiXie8 do you have fix for this?

FeiXie8, 2021-11-03:
> @FeiXie8 do you have fix for this?

try to use another responsitory which named "ORB_SLAM3_FIXD"

Chenliezhen259, 2021-11-08:
> @FeiXie8 do you have fix for this?

excuse me, do you figure out it?

rm-azou, 2022-01-27:
我用这个里面的方法改了CmakelList.txt（把项目里的所有CmakelList.txt中关于opencv版本都检查一遍，都换成opencv3(ros自带opencv是opencv3)），删除所有编译生成的build文件和lib文件，重新编译（重新运行build.sh build_ros.sh）就好了
https://github.com/UZ-SLAMLab/ORB_SLAM3/issues/333

lifei8956097, 2022-04-02:
I have fixed the problem. Maybe my solution can help you ! 
it's the problem of opencv version,  i found that the bug is caused by reading the data of Matrix in yaml file.
So, I change the Matrix into the vector  and read it by Node Iterator.such like that:
First,change the Matrix to vector :
Tbc: [-9.9857676677198759e-01, -4.7901561421323555e-02,
       2.3449547471171506e-02, 5.9999999999999998e-02,
       4.7595588902835467e-02, -9.9877630271531237e-01,
       -1.3437151905179608e-02, 0., 2.4064512880917498e-02,
       -1.2301932682702903e-02, 9.9963471411914984e-01,
       -1.6400000000000000e-03, 0., 0., 0., 1.]
Second,change the reading code:
 cv::Mat cvTbc = cv::Mat::zeros(4, 4, CV_32F);
    cv::FileNode node = fSettings["Tbc"];
    cv::FileNodeIterator it = node.begin(), it_end = node.end();
    if (!node.empty())
    {
        for (int i = 0; it != it_end; ++it, i++)
        {
            cvTbc.at<float>(i / 4, i % 4) = *it;
        }
//        cvTbc = node.mat();

lifei8956097, 2022-04-02:
Do not use node.mat()

Water1775, 2022-04-19:
> 不要使用 node.mat（）

which file is "cvTbc = node.mat（）;" in?

lifei8956097, 2022-04-20:
> > 不要使用 node.mat（）
> 
> which file is "cvTbc = node.mat（）;" in?

tracking.cc parseImuParamters

BlueAkoasm, 2022-06-21:
> I have fixed the problem. Maybe my solution can help you ! it's the problem of opencv version, i found that the bug is caused by reading the data of Matrix in yaml file. So, I change the Matrix into the vector and read it by Node Iterator.such like that: First,change the Matrix to vector : Tbc: [-9.9857676677198759e-01, -4.7901561421323555e-02, 2.3449547471171506e-02, 5.9999999999999998e-02, 4.7595588902835467e-02, -9.9877630271531237e-01, -1.3437151905179608e-02, 0., 2.4064512880917498e-02, -1.2301932682702903e-02, 9.9963471411914984e-01, -1.6400000000000000e-03, 0., 0., 0., 1.] Second,change the reading code: cv::Mat cvTbc = cv::Mat::zeros(4, 4, CV_32F); cv::FileNode node = fSettings["Tbc"]; cv::FileNodeIterator it = node.begin(), it_end = node.end(); if (!node.empty()) { for (int i = 0; it != it_end; ++it, i++) { cvTbc.at(i / 4, i % 4) = *it; } // cvTbc = node.mat();

Hello i change the code according to your method , but something wrong happened. 
[error: no matching function for call to ‘cv::Mat::at(int, int)’]
How to solve it?

ricardo11235, 2022-06-21:
> Hello i change the code according to your method , but something wrong happened. [error: no matching function for call to ‘cv::Mat::at(int, int)’] How to solve it?

cvTbc.at<float>(i/4,i%4)=*it;