UZ-SLAMLab/ORB_SLAM3 #406: Stereo-Inertial mode error with [RESETING FRAME!!!]

GeYapeng, 2021-10-09:
Could anyone can help with this problem ?
It seems that slam is set to reset and initialize  repeatedly for the first 10 KFs unless I move the camera-imu body(same to Euroc dataset location) for  enough motion for initializing . But it failed with an error [RESETING FRAME!!!] after the motion is given to the body and details are below.
```
Camera Parameters: 
- Camera: Pinhole
- fx: 480.599
- fy: 480.599
- cx: 267.82
- cy: 264.612
- k1: 0
- k2: 0
- p1: 0
- p2: 0
- fps: 60
- color order: RGB (ignored if grayscale)

Depth Threshold (Close/Far Points): 4.2599

ORB Extractor Parameters: 
- Number of Features: 1200
- Scale Levels: 8
- Scale Factor: 1.2
- Initial Fast Threshold: 20
- Minimum Fast Threshold: 7

Left camera to Imu Transform (Tbc): 
[0, -1, 0, 0;
 1, 0, 0, -0.059999999;
 0, 0, 1, 0.0099999998;
 0, 0, 0, 1]

IMU frequency: 200 Hz
IMU gyro noise: 1.7 rad/s/sqrt(Hz)
IMU gyro walk: 1.9 rad/s^2/sqrt(Hz)
IMU accelerometer noise: 2 m/s^2/sqrt(Hz)
IMU accelerometer walk: 3 m/s^3/sqrt(Hz)

not IMU meas
Framebuffer with requested attributes not available. Using available framebuffer. You may see visual artifacts.First KF:0; Map init KF:0

New Map created with 697 points
build optimization graph
start optimization
end optimization
IMU in Map 0 is initialized
start VIBA 1
build optimization graph
start optimization
end optimization
end VIBA 1

RESETING FRAME!!!
```
I use Tbc and imu noise/walk parameters  with manual setting . The noise/walk parameters are increased for many times refering to issues before . I wonder that how could I fix it as long as it can be initialized normally and I'll also appreciate it if there are more details for adjusting or testing imu parameters .

Chenliezhen259, 2021-12-06:
i also met this problem， I'll also appreciate it if there are some advices to me.

closed 2021-12-24 by GeYapeng

Maithilishetty, 2022-03-21:
Hi @GeYapeng, could you fix this issue?

akkoyunlu, 2024-06-05:
HOW DİD YOU FİX THİS PROBLEM

SrOdranoel, 2025-03-19:
Amy one fix?