UZ-SLAMLab/ORB_SLAM3 #410: IMU align with camera's Coordinates

Abner0907, 2021-10-19:
I have met a issue when run realsense D455.
When I run the ros Mono node, it can detetced the ORB feature and local the map.
How ever, when I run ros Mono-inertial node, I found that it can not local the map with the D455 module.

I differ the offical Euor data bag, and monitor the imu topic, I found the imu has different coordinate.

so I guess it must be my bad input imu coordinates. I am very appreciated that someone can give me some references or anything related. many thanks again.

Sherlock-hh, 2021-10-20:
hi, I met the same problem, I refer to the Euor datasets, turn the coordinate like this,
`cv::Point3f acc = {-accel_data.y, accel_data.x, accel_data.z};`
but my code still have some problems

Abner0907, 2021-10-21:
@Sherlock-hh  hi, There is one question about use the D455.
I found that D455 only publish one image topic `/camera/color/image_raw`
if you use Stereo-Inertial, how can you subscribe the another image topic.

I use the different test, It works for me when use mono thread. 
run command:
> rosrun ORB_SLAM3 Mono Vocabulary/ORBvoc.txt Examples/Monocular-Inertial/D455i.yaml

It can not work for me when use below which get in the IMU data.
> rosrun ORB_SLAM3 Mono_Inertial Vocabulary/ORBvoc.txt Examples/Monocular-Inertial/D455i.yaml

Abner0907, 2021-10-21:
> I use [librealsense](https://github.com/Shaxpy/Intel_Realsense_D455) to get two infrared images, have you see this issue ?

I will also review your reference. thank you for your share.

Sherlock-hh, 2021-10-22:
> @Sherlock-hh Thank you for your kindly share, i have one T265, I will test it and let you know the result tomorrow. Today I have some flight controller work to finish.
> 
> Can you localtion with the stereo demo in the first 30 seconds?

Thanks for your share! no, it can't, usually it's tracked success at first several frames, and then It's shows **faile to track local map**,and reset the map, restart the Track, and loop above. As I check the code, when it's start to Intialize, **StereoInitialization()** need to have enough motion, I don't know is mono same? but when I try to move the camera fast, It's keep report **faile to track local map**, I'll keep find out the reason.

```
IMU frequency: 250 Hz
IMU gyro noise: 0.00201078 rad/s/sqrt(Hz)
IMU gyro walk: 1.33364e-05 rad/s^2/sqrt(Hz)
IMU accelerometer noise: 0.0170676 m/s^2/sqrt(Hz)
IMU accelerometer walk: 0.000377325 m/s^3/sqrt(Hz)
not IMU meas
dist:7798639747.243557
First KF:0; Map init KF:0
New Map created with 147 points
Tracked! 
Tracked! 
Tracked! 
Tracked! 
Tracked! 
Tracked! 
Tracked! 
Tracked! 
Tracked! 
Tracked! 
Tracked! 
Tracked! 
Tracked! 
Tracked! 
Tracked! 
Tracked! 
Tracked! 
Tracked! 
Tracked! 
Tracked! 
Tracked! 
Tracked! 
Tracked! 
Tracked! 
Tracked! 
Tracked! 
Tracked! 
Tracked! 
Tracked! 
Tracked! 
Tracked! 
Tracked! 
Tracked! 
Tracked! 
Tracked! 
Tracked! 
Tracked! 
Tracked! 
Tracked! 
Tracked! 
Tracked! 
Tracked! 
Tracked! 
Tracked! 
Tracked! 
Tracked! 
Fail to track local map!
IMU is not or recently initialized. Reseting active map...
Tracked! 
LM: Active map reset recieved
LM: Active map reset, waiting...
LM: Reseting current map in Local Mapping...
LM: End reseting Local Mapping...
LM: Reset free the mutex
LM: Active map reset, Done!!!
mnFirstFrameId = 1
mnInitialFrameId = 0
47 Frames set to lost
not IMU meas
dist:0.65457307804665832
First KF:9; Map init KF:0
New Map created with 199 points
Tracked! 
Tracked! 
Tracked! 
Tracked! 
Tracked! 
Tracked! 
Fail to track local map!
IMU is not or recently initialized. Reseting active map...
Tracked! 
LM: Active map reset recieved
LM: Active map reset, waiting...
LM: Reseting current map in Local Mapping...
LM: End reseting Local Mapping...
LM: Reset free the mutex
LM: Active map reset, Done!!!
mnFirstFrameId = 49
mnInitialFrameId = 47
54 Frames set to lost
not IMU meas
dist:0.53143759665025792
First KF:10; Map init KF:9
New Map created with 124 points
Tracked! 
Fail to track local map!
IMU is not or recently initialized. Reseting active map...
Tracked! 
LM: Active map reset recieved
LM: Active map reset, waiting...
LM: Reseting current map in Local Mapping...
LM: End reseting Local Mapping...
LM: Reset free the mutex
LM: Active map reset, Done!!!
mnFirstFrameId = 57
mnInitialFrameId = 55
'''
```

0y8w1x, 2022-03-18:
@Sherlock-hh @Abner0907 did you resolve the problem ? Because I'm having the same troubles.

Abner0907, 2022-04-01:
@0y8w1x nope, I can not test it at last as my camera is used by others.

Sherlock-hh, 2022-04-13:
> 

the imu initialization need motion, you can try to move the imu at first till it print the "end VBA2".

Abner0907, 2022-07-16:
no issues any more. I close it. 
If anyone is interested in ORB_SLAM. we can chat in wechat.

closed 2022-07-16 by Abner0907

zgxsin, 2022-09-05:
> no issues any more. I close it. If anyone is interested in ORB_SLAM. we can chat in wechat.

Could you tell how you solved the issue? thanks a lot!

Styazoua, 2024-02-08:
> no issues any more. I close it. If anyone is interested in ORB_SLAM. we can chat in wechat.

How did you solve this problem?