UZ-SLAMLab/ORB_SLAM3 #422: core dump when create a new map

FeiXie8, 2021-11-03:
mnFirstFrameId = 370
mnInitialFrameId = 353
228 Frames set to lost
First KF:17; Map init KF:12
New Map created with 148 points
Fail to track local map!
Fail to track local map!
Fail to track local map!
Fail to track local map!
Fail to track local map!
Fail to track local map!
Fail to track local map!
Fail to track local map!
Fail to track local map!
Fail to track local map!
Fail to track local map!
Fail to track local map!
Fail to track local map!
Fail to track local map!
Fail to track local map!
Fail to track local map!
Fail to track local map!
Fail to track local map!
Fail to track local map!
Fail to track local map!
Fail to track local map!
Fail to track local map!
Fail to track local map!
Fail to track local map!
Fail to track local map!
Fail to track local map!
Fail to track local map!
Fail to track local map!
Fail to track local map!
Fail to track local map!
Fail to track local map!
Fail to track local map!
Fail to track local map!
Fail to track local map!
Fail to track local map!
Fail to track local map!
Fail to track local map!
Fail to track local map!
Fail to track local map!
**Relocalized!!
TRACK_REF_KF: Less than 15 matches!!
Fail to track local map!
Creation of new map with id: 1
Exits current map 
Saved map with ID: 0
Creation of new map with last KF id: 23
[1]    42306 segmentation fault (core dumped)  rosrun orbslam3 mynteye_orbslam   false**

closed 2021-11-04 by FeiXie8

davidlai0522, 2022-09-01:
Hi, do you have any solution for this? I faced the same issue. It only happens occasionally. I noticed that there is a high chance that it will happen when I am building a big map. When I tried to printed out the MapPoint pointer pMPi in Map.cc file Map::PreSave() function, I noticed that it will point to some weird address when the segmentation fault happens.
[image]
[image]
Notice the last pointer address before it fails. Do you have any idea about this?

Suhao07, 2024-09-12:
bro how did you solve this problem?