UZ-SLAMLab/ORB_SLAM3 #424: Unable to get good results in my own dataset for monocular

Shantnu12345, 2021-11-11:
Hi,

When I run on Kitty sequences I am getting really good results. However, when I am trying my own dateset collected on a highway (please find attached pictures), I am facing sever issues. My dateset is similar to Kitty sequence 21. I have tried to modify the YAML parameters (nFeatures, scaleFactor, level, iniThFast, minThFast, etc) but no luck. Here are some of the issues I am facing:

- Taking long time to initialize (which varies from 5 secs to 30 secs randomly) for the same image stream.
- Map gets reset occasionally and I get the error  "Fail to track local map!"

Please let me know what can be done to alleviate these problems. Thank you a lot in advance :)

[image]
[image]
ms

tejalbarnwal, 2021-11-11:
Did you check for the camera model, lapping area, size of the image?
Also in what mode are you running?

Shantnu12345, 2021-11-18:
Hi,

Thank you for your response.

I am running the monocular only mode. By camera model do you mean camera type (Ex pinhole). My image size is 1936 by 1220. Could you please enlighten me what do you mean by lapping area?

tejalbarnwal, 2021-12-03:
@Shantnu12345 I am not really very sure about this but it represents the overlapping between images of two different cameras.
Refer to #88 for more details on the lapping area.
Yes by camera type I mean the camera model - it could be fisheye( Kannala Brandt) or pinhole
Also, by looking at the YAML files for monocular only in the examples given in the repository, it seems we don't delve into the factors-like lapping area for monocular only!
I suggest trying to figure out your camera model ( if you find great fov and distortions, it is better to calibrate your camera using the OpenCV fisheye model) and then calibrate it to find the extrinsic and intrinsic parameters of the camera.

Also, if you could try orb-slam2 too.

Maithilishetty, 2022-03-21:
@Shantnu12345 were you able to fix this?

yyanzzw, 2023-12-22:
hi！
I am also trying my own dateset but facing the same problem: Map gets reset occasionally and I get the error "Fail to track local map!"
Have you solved that problem already?