UZ-SLAMLab/ORB_SLAM3 #425: .pnged to load image at

yriuss, 2021-11-12:
Running `./euroc_examples` the output stops at ".pnged to load image at" and nothing happens. What should I do now?

The output:

```
Launching MH01 with Stereo sensor
num_seq = 1
file name: dataset-MH01_stereo
Loading images for sequence 0...LOADED!

-------

ORB-SLAM3 Copyright (C) 2017-2020 Carlos Campos, Richard Elvira, Juan J. Gómez, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
ORB-SLAM2 Copyright (C) 2014-2016 Raúl Mur-Artal, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
This program comes with ABSOLUTELY NO WARRANTY;
This is free software, and you are welcome to redistribute it
under certain conditions. See LICENSE.txt.

Input sensor was set to: Stereo

Loading ORB Vocabulary. This could take a while...
Vocabulary loaded!

Creation of new map with id: 0
Creation of new map with last KF id: 0
Seq. Name: 

Camera Parameters: 
- Camera: Pinhole
- fx: 435.20468139648438
- fy: 435.20468139648438
- cx: 367.45172119140625
- cy: 252.20085144042969
- k1: 0
- k2: 0
- p1: 0
- p2: 0
- fps: 20
- color order: RGB (ignored if grayscale)

Depth Threshold (Close/Far Points): 3.8527247905731201

ORB Extractor Parameters: 
- Number of Features: 1200
- Scale Levels: 8
- Scale Factor: 1.2000000476837158
- Initial Fast Threshold: 20
- Minimum Fast Threshold: 7

.pnged to load image at: /Datasets/EuRoC/MH01/mav0/cam0/data/1403636579763555584
```

mskhrk, 2021-11-12:
Check whether the file in the record image file path contains CR newline characters
Try to clear CR newline characters
ex ./Monocular/EuRoC_TimeStamps/MH01.txt

yriuss, 2021-11-12:
@mskhrk Before ".pnged to load image at" the output was "Failed to load image at". Then I tried this and the output changed to what it is now. One other thing that I can mention is that now the path changed to "~/data/EuRoC/MH01/mav0/cam0/data/1403636579763555584", which is the path that I was expecting.

yriuss, 2021-11-12:
I am still analyzing, but there is clearly something wrong with the line "strPathLeft + "/" + ss.str() + ".png"" in the code...

yriuss, 2021-11-12:
update: I changed to ""/home/adriel/data/EuRoC" + strPathLeft + "/" + ss.str().substr(0, ss.str().size()-1) + ".png"" and it worked!

closed 2021-11-12 by yriuss