UZ-SLAMLab/ORB_SLAM3 #431: Question: How to use multi-session on KITTI dataset.

haradatm, 2021-12-01:
When I input multiple overlapping sequences in a row and process them in multi-session, only the first sequence outputs a large number of KFs. I want to get a certain number of accurate camera poses for the second and subsequent sequences. Does anyone know how to do that?

ProVival, 2022-03-19:
I am confused about how orbslam3 multi session. Were there multiple online orbslam's for your approach? I am trying to run multiple orbslam's and i want them to use each others map and provide a global map.  Can you explain how did you do it, i want to try too :D

haradatm, 2022-03-19:
My approach is offline. My motivation is to collect multiple in-vehicle camera images like KITTI (showing the same scene)  and improve the accuracy of self-location estimation by multi-map.

However, my current understanding is that ORB-SLAM3 is not concerned with connecting multiple disparate maps, but rather with making any one map larger.

ProVival, 2022-03-19:
@haradatm  Ah okay i understand it now. I guess i 'll have to try to implement it somehow :D Thanks a lot!

closed 2022-03-19 by haradatm

hankge, 2023-08-16:
Hi @ProVival @haradatm . For now, I want using orb_slam3 to do multi-camera mapping. Have you guys done that successfully through this repo?