UZ-SLAMLab/ORB_SLAM3 #432: ZED2 run monocular inertial navigation mode

RoZhong, 2021-12-01:
I keep reinitializing the IMU when I run mono+imu.
[image]
I used zed2. After Kalibr was used to calibrate the camera, I used imu_utils to calibrate the imu, and finally Kalibr was used again to calibrate the camera and IMU jointly, but In the end, this problem still occurs. I wonder if you know where the problem is and how to solve it?

Blithe98, 2021-12-02:
Hi! When your calibration is right, you need to move your device enough because of  IMU needing to initialize.

RoZhong, 2021-12-02:
> Hi! When your calibration is right, you need to move your device enough because of IMU needing to initialize.

emmm,是可以初始化的，但是每次初始化后，总是会尝试重新初始化，我看跟多人都是这个问题，如果您实现过orb3，mono+imu，并且您愿意的话，能否加个微信帮我查看一下问题所在

Blithe98, 2021-12-03:
如果您能确定内外参标定均是正确的情况下，至少要保证旋转矩阵偏差不会太大，就能正常运行单目imu模式。您说的不停的初始化是单目imu下的正常现象，因为单目imu模式下流程是首先会初始化imu，然后是两次VIBA，再就是尺度优化和优化重力方向。重点在于单目imu模式下，需要您不停运动您的设备，等上述流程打印完成之后，才算完成了初始化。这个坑我也淌过，其他的暂无可奉告。

祝好！

发自个板马的iPhone

RoZhong, 2021-12-03:
I solved this problem. It feels that the IMU is not calibrated enough. I have been calibrating these days, and then I can calibrate it today, especially the rotation matrix from the left camera to the imu. This is very important.
This is my rotation matrix, for your reference.
 [ 0.00609392 , 0.00366033 , 0.99997473, 0.02461291,
           -0.99991961 , 0.01114129 , 0.00605281, 0.02212419,
          -0.01111886 ,-0.99993123 , 0.00372793 ,0.00025937,
           0.0, 0.0, 0.0, 1.0]

closed 2021-12-03 by RoZhong

dingshuoshuo, 2022-08-11:
> I solved this problem. It feels that the IMU is not calibrated enough. I have been calibrating these days, and then I can calibrate it today, especially the rotation matrix from the left camera to [...]
你好，可以加个QQ嘛，43205504,我也遇到了这个问题，一直在重新初始化，请问如何成功的