UZ-SLAMLab/ORB_SLAM3 #433: ./build.sh stuck on build.make examples Ubunut 20.04

chamou-f, 2021-12-01:
Hi. After changing the compiler version in CMakeLists.txt from c++11 to c+14 Cmake stuck on making the examples. Does anyone know how to fix this? 

---------------------------------------------------------------------------------------------------------------------------------
ma@MA:~/Dev/ORB_SLAM3$ ./build.sh
Configuring and building Thirdparty/DBoW2 ...
mkdir: cannot create directory ‘build’: File exists
-- Configuring done
-- Generating done
-- Build files have been written to: /home/ma/Dev/ORB_SLAM3/Thirdparty/DBoW2/build
[100%] Built target DBoW2
Configuring and building Thirdparty/g2o ...
mkdir: cannot create directory ‘build’: File exists
-- BUILD TYPE:Release
-- Compiling on Unix
-- Configuring done
-- Generating done
-- Build files have been written to: /home/ma/Dev/ORB_SLAM3/Thirdparty/g2o/build
[100%] Built target g2o
Uncompress vocabulary ...
Configuring and building ORB_SLAM3 ...
mkdir: cannot create directory ‘build’: File exists
Build type: Release
-- Using flag -std=c++14.
OPENCV VERSION: 4.4.0
-- BUILD TYPE:Release
-- Compiling on Unix
-- Configuring done
-- Generating done
-- Build files have been written to: /home/ma/Dev/ORB_SLAM3/build
[ 37%] Built target g2o
[ 70%] Built target ORB_SLAM3
[ 71%] Building CXX object CMakeFiles/mono_tum.dir/Examples/Monocular/mono_tum.cc.o
[ 74%] Building CXX object CMakeFiles/stereo_kitti.dir/Examples/Stereo/stereo_kitti.cc.o
[ 74%] Building CXX object CMakeFiles/rgbd_tum.dir/Examples/RGB-D/rgbd_tum.cc.o
[ 75%] Building CXX object CMakeFiles/mono_kitti.dir/Examples/Monocular/mono_kitti.cc.o
[ 76%] Building CXX object CMakeFiles/stereo_euroc.dir/Examples/Stereo/stereo_euroc.cc.o
[ 78%] Building CXX object CMakeFiles/stereo_inertial_euroc.dir/Examples/Stereo-Inertial/stereo_inertial_euroc.cc.o
In file included from /home/ma/Dev/ORB_SLAM3/include/Atlas.h:27,
                 from /home/ma/Dev/ORB_SLAM3/include/FrameDrawer.h:25,
                 from /home/ma/Dev/ORB_SLAM3/include/Viewer.h:23,
                 
           .....
           .....
           .....

/home/ma/Dev/ORB_SLAM3/Examples/Monocular/mono_tum.cc:89:88: error: ‘t1’ was not declared in this scope; did you mean ‘y1’?
   89 |         double ttrack= std::chrono::duration_cast<std::chrono::duration<double> >(t2 - t1).count();
      |                                                                                        ^~
      |                                                                                        y1
/home/ma/Dev/ORB_SLAM3/Examples/Stereo-Inertial/stereo_inertial_euroc.cc: In function ‘int main(int, char**)’:
/home/ma/Dev/ORB_SLAM3/Examples/Stereo-Inertial/stereo_inertial_euroc.cc:230:26: error: ‘std::chrono::monotonic_clock’ has not been declared
  230 |             std::chrono::monotonic_clock::time_point t1 = std::chrono::monotonic_clock::now();
      |                          ^~~~~~~~~~~~~~~
/home/ma/Dev/ORB_SLAM3/Examples/Stereo-Inertial/stereo_inertial_euroc.cc:239:26: error: ‘std::chrono::monotonic_clock’ has not been declared
  239 |             std::chrono::monotonic_clock::time_point t2 = std::chrono::monotonic_clock::now();
      |                          ^~~~~~~~~~~~~~~
/home/ma/Dev/ORB_SLAM3/Examples/Stereo-Inertial/stereo_inertial_euroc.cc:247:87: error: ‘t2’ was not declared in this scope; did you mean ‘tm’?
  247 |             double ttrack= std::chrono::duration_cast<std::chrono::duration<double> >(t2 - t1).count();
      |                                                                                       ^~
      |                                                                                       tm
/home/ma/Dev/ORB_SLAM3/Examples/Stereo-Inertial/stereo_inertial_euroc.cc:247:92: error: ‘t1’ was not declared in this scope; did you mean ‘y1’?
  247 |             double ttrack= std::chrono::duration_cast<std::chrono::duration<double> >(t2 - t1).count();
      |                                                                                            ^~
      |                                                                                            y1
/home/ma/Dev/ORB_SLAM3/Examples/Stereo-Inertial/stereo_inertial_euroc.cc:169:16: warning: unused variable ‘t_rect’ [-Wunused-variable]
  169 |         double t_rect = 0;
      |                ^~~~~~
/home/ma/Dev/ORB_SLAM3/Examples/Stereo-Inertial/stereo_inertial_euroc.cc:170:16: warning: unused variable ‘t_track’ [-Wunused-variable]
  170 |         double t_track = 0;
      |                ^~~~~~~
/home/ma/Dev/ORB_SLAM3/Examples/Stereo-Inertial/stereo_inertial_euroc.cc:171:13: warning: unused variable ‘num_rect’ [-Wunused-variable]
  171 |         int num_rect = 0;
      |             ^~~~~~~~
make[2]: *** [CMakeFiles/stereo_euroc.dir/build.make:82: CMakeFiles/stereo_euroc.dir/Examples/Stereo/stereo_euroc.cc.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:247: CMakeFiles/stereo_euroc.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make[2]: *** [CMakeFiles/stereo_kitti.dir/build.make:82: CMakeFiles/stereo_kitti.dir/Examples/Stereo/stereo_kitti.cc.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:220: CMakeFiles/stereo_kitti.dir/all] Error 2
make[2]: *** [CMakeFiles/mono_kitti.dir/build.make:82: CMakeFiles/mono_kitti.dir/Examples/Monocular/mono_kitti.cc.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:274: CMakeFiles/mono_kitti.dir/all] Error 2
make[2]: *** [CMakeFiles/stereo_inertial_euroc.dir/build.make:82: CMakeFiles/stereo_inertial_euroc.dir/Examples/Stereo-Inertial/stereo_inertial_euroc.cc.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:166: CMakeFiles/stereo_inertial_euroc.dir/all] Error 2
make[2]: *** [CMakeFiles/mono_tum.dir/build.make:82: CMakeFiles/mono_tum.dir/Examples/Monocular/mono_tum.cc.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:193: CMakeFiles/mono_tum.dir/all] Error 2
make[2]: *** [CMakeFiles/rgbd_tum.dir/build.make:82: CMakeFiles/rgbd_tum.dir/Examples/RGB-D/rgbd_tum.cc.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:139: CMakeFiles/rgbd_tum.dir/all] Error 2
make: *** [Makefile:103: all] Error 2
----------------------------------------------------------------------------------------------------------------------------------------------------

arjunskumar, 2021-12-13:
```         
std::chrono::steady_clock::time_point t1 = std::chrono::steady_clock::now();

// Pass the image to the SLAM system

SLAM.TrackMonocular(im,tframe);

std::chrono::steady_clock::time_point t2 = std::chrono::steady_clock::now();
```
make these changes in the files

nopnop2002, 2023-02-14:
@chamou-f 

Did you change the following part of CMakeLists.txt?
```
#CHECK_CXX_COMPILER_FLAG("-std=c++11" COMPILER_SUPPORTS_CXX11)
CHECK_CXX_COMPILER_FLAG("-std=c++14" COMPILER_SUPPORTS_CXX14)
```

Here you need to change:
```
#CHECK_CXX_COMPILER_FLAG("-std=c++11" COMPILER_SUPPORTS_CXX11)
CHECK_CXX_COMPILER_FLAG("-std=c++14" COMPILER_SUPPORTS_CXX11)
```