UZ-SLAMLab/ORB_SLAM3 #436: run mono_tum_vi use dataset-room4_512_16 cann't creat map

git3210, 2021-12-14:
command:
./Examples/Monocular/mono_tum_vi Vocabulary/ORBvoc.txt Examples/Monocular/TUM1.yaml ../ORB_SLAM3_detailed_comments/data/dataset-room4_512_16/mav0/cam0/data Examples/Monocular/TUM_TimeStamps/dataset-room4_512.txt dataset-room4_512_mono

when i use this command to run mono_tum_vi, it cann't build map. The datasets is dataset-room4_512_16, I found matcher.SearchForInitialization return nmatches less than 10, why? does the dataset need to undistortion ?

Thanks!

git3210, 2021-12-14:
pictures
[image]

sadsasuke, 2022-03-17:
Hi any updates to this issue?