UZ-SLAMLab/ORB_SLAM3 #439: Sophus exp failed!

knazneen123, 2021-12-24:
Thank alot guys for the updated version amazing work! I am recently facing this issue while running whereas in previous version I had no such issue and it use to work flawlessly

When I try to run mono inertial mode with ROS using updated V1.0 yaml file it runs for a while and then I get this error whereas in monocular mode it runs perfectly-

____________________________________________________________________________________________________-
ORB-SLAM3 Copyright (C) 2017-2020 Carlos Campos, Richard Elvira, Juan J. Gómez, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
ORB-SLAM2 Copyright (C) 2014-2016 Raúl Mur-Artal, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
This program comes with ABSOLUTELY NO WARRANTY;
This is free software, and you are welcome to redistribute it
under certain conditions. See LICENSE.txt.

Input sensor was set to: Monocular-Inertial
Loading settings from /home/../slam/mi6.yaml
	-Loaded camera 1
	-Loaded image info
IMU.InsertKFsWhenLost optional parameter does not exist...
	-Loaded IMU calibration
	-Loaded ORB settings
Viewer.imageViewScale optional parameter does not exist...
	-Loaded viewer settings
System.LoadAtlasFromFile optional parameter does not exist...
System.SaveAtlasToFile optional parameter does not exist...
	-Loaded Atlas settings
System.thFarPoints optional parameter does not exist...
	-Loaded misc parameters
----------------------------------
SLAM settings: 
	-Camera 1 parameters (Pinhole): [ 572.513 596.433 321.78 244.678 ]
	-Camera 1 distortion parameters: [  0.444542 -4.39982 -0.0050516 -0.00312799 11.5163 ]
	-Original image size: [ 640 , 480 ]
	-Current image size: [ 640 , 480 ]
	-Camera 1 parameters after resize: [  572.513 596.433 321.78 244.678 ]
	-Sequence FPS: 30
	-Gyro noise: 0.00150402
	-Accelerometer noise: 0.00177764
	-Gyro walk: 3.72336e-05
	-Accelerometer walk: 0.000560036
	-IMU frequency: 200
	-Features per image: 1000
	-ORB scale factor: 1.2
	-ORB number of scales: 8
	-Initial FAST threshold: 20
	-Min FAST threshold: 7

Loading ORB Vocabulary. This could take a while...
Vocabulary loaded!

Initialization of Atlas from scratch 
Creation of new map with id: 0
Creation of new map with last KF id: 0
Seq. Name: 
There are 1 cameras in the atlas
Camera 0 is pinhole
Starting the Viewer
First KF:0; Map init KF:0
New Map created with 90 points
Fail to track local map!
IMU is not or recently initialized. Reseting active map...
SYSTEM-> Reseting active map in monocular case
LM: Active map reset recieved
LM: Active map reset, waiting...
LM: Reseting current map in Local Mapping...
LM: End reseting Local Mapping...
LM: Reset free the mutex
LM: Active map reset, Done!!!
mnFirstFrameId = 44
mnInitialFrameId = 0
7 Frames set to lost
First KF:2; Map init KF:0
New Map created with 89 points
TRACK_REF_KF: Less than 15 matches!!
Fail to track local map!
SYSTEM-> Reseting active map in monocular case
LM: Active map reset recieved
LM: Active map reset, waiting...
LM: Reseting current map in Local Mapping...
LM: End reseting Local Mapping...
LM: Reset free the mutex
LM: Active map reset, Done!!!
mnFirstFrameId = 55
mnInitialFrameId = 50
8 Frames set to lost
First KF:4; Map init KF:2
New Map created with 134 points
Fail to track local map!
IMU is not or recently initialized. Reseting active map...
SYSTEM-> Reseting active map in monocular case
LM: Active map reset recieved
LM: Active map reset, waiting...
LM: Reseting current map in Local Mapping...
LM: End reseting Local Mapping...
LM: Reset free the mutex
LM: Active map reset, Done!!!
mnFirstFrameId = 74
mnInitialFrameId = 56
10 Frames set to lost
First KF:6; Map init KF:4
New Map created with 139 points
Fail to track local map!
IMU is not or recently initialized. Reseting active map...
SYSTEM-> Reseting active map in monocular case
LM: Active map reset recieved
LM: Active map reset, waiting...
LM: Reseting current map in Local Mapping...
LM: End reseting Local Mapping...
LM: Reset free the mutex
LM: Active map reset, Done!!!
mnFirstFrameId = 97
mnInitialFrameId = 75
12 Frames set to lost
First KF:8; Map init KF:6
New Map created with 186 points
Sophus ensure failed in function 'static Sophus::SO3<Scalar_> Sophus::SO3<Scalar_, Options>::expAndTheta(const Tangent&, Sophus::SO3<Scalar_, Options>::Scalar*) [with Scalar_ = float; int Options = 0; Sophus::SO3<Scalar_, Options>::Tangent = Eigen::Matrix<float, 3, 1>; Sophus::SO3<Scalar_, Options>::Scalar = float]', file '/home/../slam/projects/ORB_SLAM3_V1.0/Thirdparty/Sophus/sophus/so3.hpp', line 614.
SO3::exp failed! omega: -nan -nan -nan, real: -nan, img: -nan
Aborted (core dumped)

arcanon, 2022-02-06:
seeing this issue as well.

not always but on the current test all the time.

Maithilishetty, 2022-03-21:
@knazneen123 did you manage to fix this?

knazneen123, 2022-04-12:
you might have to debug and get the line causing the error and use isNan to add an assert or a condition check to avoid seg fault or other error

macTracyHuang, 2022-05-26:
me facing the same issue, did anyone fix it? thx

cpymaple, 2022-05-30:
I face this problem too, but I do not know how to solve it.