UZ-SLAMLab/ORB_SLAM3 #442: v1.0 realse ros版本编译,提示找不到sophus库，及 Sophus::SE3f, cv::MAT,Eigen::Vector3f类型转换报错

oscar1780, 2021-12-25:
系统环境  ubuntu 18.04 , melodic, opencv4.0(本地编译 仅make 未make install,考虑兼容性问题未安装)
解决办法：
1、编译ros版本，需要将Examples_old文件夹下ROS文件夹复制到Examples文件夹下。

closed 2021-12-26 by oscar1780

knazneen123, 2021-12-27:
Are you facing this sophus error while running monocular inertial using ROS

Sophus ensure failed in function 'static Sophus::SO3<Scalar_> Sophus::SO3<Scalar_, Options>::expAndTheta(const Tangent&, Sophus::SO3<Scalar_, Options>::Scalar*) [with Scalar_ = float; int Options = 0; Sophus::SO3<Scalar_, Options>::Tangent = Eigen::Matrix<float, 3, 1>; Sophus::SO3<Scalar_, Options>::Scalar = float]', file '/home/../slam/projects/ORB_SLAM3_V1.0/Thirdparty/Sophus/sophus/so3.hpp', line 614.
SO3::exp failed! omega: -nan -nan -nan, real: -nan, img: -nan
??

nindanaoto, 2021-12-27:
@knazneen123 You can use my code for the PR if you want.

knazneen123, 2021-12-28:
Sure, are you not facing sophus error issue I mentioned earlier when you try to run Monocular-Inertial slam using ROS?

nindanaoto, 2021-12-28:
I've just tested Stereo SLAM. If the error is Monocular specific one, I don't have any knowledge to add......

BSSNBSSN, 2022-02-17:
That helps a lot, thank you very much!

KunSong-L, 2022-03-01:
按照上述方法解决了  **cv::Mat Tcw = mpSLAM->TrackMonocular(cv_ptr->image,cv_ptr->header.stamp.toSec());**   报错以后，又出现了新的问题。

error: ‘eigen2cv’ is not a member of ‘cv’

所用的代码片段如下：
```cpp
void ImageGrabber::GrabImage(const sensor_msgs::ImageConstPtr& msg)
{
    // Copy the ros image message to cv::Mat.
    cv_bridge::CvImageConstPtr cv_ptr;
    try
    {
        cv_ptr = cv_bridge::toCvShare(msg);
    }
    catch (cv_bridge::Exception& e)
    {
        ROS_ERROR("cv_bridge exception: %s", e.what());
        return;
    }
    cv::Mat im = cv_ptr->image.clone();
    cv::Mat imu;

    cv::Mat Tcw;
    Sophus::SE3f Tcw_SE3f = mpSLAM->TrackMonocular(cv_ptr->image,cv_ptr->header.stamp.toSec());
    Eigen::Matrix4f Tcw_Matrix = Tcw_SE3f.matrix();
    cv::eigen2cv(Tcw_Matrix, Tcw);

    int state = mpSLAM->GetTrackingState();
    vector<ORB_SLAM3::MapPoint*> vMPs = mpSLAM->GetTrackedMapPoints();
    vector<cv::KeyPoint> vKeys = mpSLAM->GetTrackedKeyPointsUn();

    cv::undistort(im,imu,K,DistCoef);

    if(bRGB)
        viewerAR.SetImagePose(imu,Tcw,state,vKeys,vMPs);
    else
    {
        cv::cvtColor(imu,imu,CV_RGB2BGR);
        viewerAR.SetImagePose(imu,Tcw,state,vKeys,vMPs);
    }    
}
```

KunSong-L, 2022-03-01:
> 
如果遇到这个问题（`error: ‘eigen2cv’ is not a member of ‘cv’`），加入三个头文件就可以了
```cpp
#include <Eigen/Dense>
#include <opencv2/core/eigen.hpp>
#include <opencv2/opencv.hpp>
```

h904798869, 2022-04-15:
hi，compeled successfully，runros ORB_SLAM3 Mono 
Input sensor was set to: Monocular
段错误 (核心已转储)
How to deal with it？

KunSong-L, 2022-04-21:
For me, this is due to that when building ROS part, OpenCV4 is used, which is different from no ROS part (use Opencv3).
You can just change the CMakeLists.txt in ROS part to force it linked to Opencv3.

lhyakn, 2022-04-21:
> hi，compeled successfully，runros ORB_SLAM3 Mono Input sensor was set to: Monocular 段错误 (核心已转储) How to deal with it？

Did you fix this problem, I have the same problem as you...

lhyakn, 2022-04-23:
> no solve。are u a stduents？
> […](#)

对,我现在还是学生. 我解决这个问题了,是我这里ROS-neotic-cv_bridge的OpenCV版本的问题,我卸载了cv_bridge后,从github重新下载并编译安装cv_bridge问题就解决了,希望可以帮到你

cpymaple, 2022-05-30:
> 

hi, I face this problem, but I do not know how to solve it. Could you please help me?

meng152634, 2025-12-18:
> hi，compeled successfully，runros ORB_SLAM3 Mono Input sensor was set to: Monocular 段错误 (核心已转储) How to deal with it？

@linClubs @lhyakn this error can be fixed by delete the `-march=native`  option in the CMakeLists.txt file located in the ROS/ORB_SLAM3, Thirdarty/DBoW2, and Thirdarty/g2o.