UZ-SLAMLab/ORB_SLAM3 #443: Load/Save Map

sumitsarkar1, 2021-12-25:
Thank you guys for the updated version ! On the GUI I could not figure out how to Save/Load a map. Do we have a separate flag for that or we need to use the GUI

dz306271098, 2021-12-26:
@sumitsarkar1  Please see Calibration_Tutorial.pdf in this project. It has introduced each parameter. And you can just Save/Load a map through some parameters.

sumitsarkar1, 2021-12-27:
@dz306271098 are you able to save the map? I applied the flag `System.SaveAtlasToFile: "home/map"` but I get segmentation fault

dz306271098, 2021-12-27:
@sumitsarkar1 NO, sometimes can save the map successfully. I also met the same problem. And I think there are two bugs in the SAVE/LOAD function. 

1. `void Map::PreSave(std::set<GeometricCamera*> &spCams) ` in `Map.cc`
```
       if(!pMPi || pMPi->isBad())
            continue;
```
I find two parts used to determine whether MapPoint is valid. And in the function `bool MapPoint::isBad()`, the lock just get a NULL mutex, so their lock operate ( `lock(lock1, lock2);` ) leads to error. And I have tried to validate my guess， access to `mbBad` or ' mnId' before lock. After some successful access, also appear error in the newly added codes.
I also don't know why the condition `!pMPi ` can't prevent invalid data.

2. `void MapPoint::PreSave(set<KeyFrame*>& spKF,set<MapPoint*>& spMP)` in `MapPoint.cc`
```
    for(std::map<KeyFrame*,std::tuple<int,int> >::const_iterator it = mObservations.begin(), end = mObservations.end(); it != end; ++it)
    {
        KeyFrame* pKFi = it->first;
        if(spKF.find(pKFi) != spKF.end())
        {
            mBackupObservationsId1[it->first->mnId] = get<0>(it->second);
            mBackupObservationsId2[it->first->mnId] = get<1>(it->second);
        }
        else
        {
            EraseObservation(pKFi);
        }
    }
```
I think it is not a reliable operation when a struct map needs to erase its element. This may lead to the iterator being invalid.

sumitsarkar1, 2021-12-27:
When the stop button is pressed `Shutdown` is printed twice

dz306271098, 2021-12-27:
@sumitsarkar1 which dataset you test.  I haven't met this situation when test in V101, MH01, MH05 EuRoC datasets.

sumitsarkar1, 2021-12-27:
I am testing with Intel Realsense T265

dz306271098, 2021-12-27:
@sumitsarkar1  I have checked. When you save map successfully, the cout gets printed. Like this
[image]
some of them are added by me. The end cout if from `backward.cc` which is used to show stack trace.

sumitsarkar1, 2021-12-27:
@dz306271098 let me know if you have any success for the seg fault

dz306271098, 2021-12-27:
@sumitsarkar1 ok~ But unfortunately, I don't have a reliable way, and usually meet this error, like this 

```
start VIBA 1
end VIBA 1
start VIBA 2
end VIBA 2
Shutdown
start save map!!!!!!!!!11
Stack trace (most recent call last):
#10   Object "", at 0xffffffffffffffff, in 
#9    Object "/home/c2214-hs/Documents/test/ORB_SLAM3/Examples/Stereo-Inertial/stereo_inertial_euroc", at 0x55c83d209469, in _start
#8    Source "../csu/libc-start.c", line 310, in __libc_start_main [0x7f53fe5c5bf6]
#7    Object "/home/c2214-hs/Documents/test/ORB_SLAM3/Examples/Stereo-Inertial/stereo_inertial_euroc", at 0x55c83d208adc, in main
#6    Object "/home/c2214-hs/Documents/test/ORB_SLAM3/lib/libORB_SLAM3.so", at 0x7f5401244458, in ORB_SLAM3::System::Shutdown()
#5    Object "/home/c2214-hs/Documents/test/ORB_SLAM3/lib/libORB_SLAM3.so", at 0x7f5401243c96, in ORB_SLAM3::System::SaveAtlas(int)
#4    Object "/home/c2214-hs/Documents/test/ORB_SLAM3/lib/libORB_SLAM3.so", at 0x7f54012de274, in ORB_SLAM3::Atlas::PreSave()
#3    Object "/home/c2214-hs/Documents/test/ORB_SLAM3/lib/libORB_SLAM3.so", at 0x7f54012e0ffc, in ORB_SLAM3::Map::PreSave(std::set<ORB_SLAM3::GeometricCamera*, std::less<ORB_SLAM3::GeometricCamera*>, std::allocator<ORB_SLAM3::GeometricCamera*> >&)
#2    Object "/home/c2214-hs/Documents/test/ORB_SLAM3/lib/libORB_SLAM3.so", at 0x7f54012cad8e, in ORB_SLAM3::MapPoint::isBad()
#1    Object "/home/c2214-hs/Documents/test/ORB_SLAM3/lib/libORB_SLAM3.so", at 0x7f54012ce88c, in void std::lock<std::unique_lock<std::mutex>, std::unique_lock<std::mutex>>(std::unique_lock<std::mutex>&, std::unique_lock<std::mutex>&)
#0    Source "../nptl/pthread_mutex_lock.c", line 67, in __pthread_mutex_lock [0x7f53ff64efd0]
Segmentation fault (Signal sent by the kernel [(nil)])
Segmentation fault (core dumped)
```
I will continue to try.

knazneen123, 2021-12-27:
I have tried save load map feature and it works for euroc dataset MH_01 series, you need to change the .yaml file and add these params 
#--------------------------------------------------------------------------------------------
# System config
#--------------------------------------------------------------------------------------------

# When the variables are commented, the system doesn't load a previous session or not store the current one

# If the LoadFile doesn't exist, the system give a message and create a new Atlas from scratch
System.LoadAtlasFromFile: "mapeuro"

# The store file is created from the current session, if a file with the same name exists it is deleted
System.SaveAtlasToFile: "mapeuro"

Above camera params
By default it saves binary type for maps for this System::SaveAtlas(int type)

sumitsarkar1, 2021-12-27:
@knazneen123 already done those but I face seg fault for T265 and because of the fault may be I cant save the map

knazneen123, 2021-12-28:
I also faced this seg fault issue when the program terminates or while I try to shut it down using ctrl+c, it mainly happens due to some pointer being corrupted or null, you can try to check the functions called in saveAtlas() like presave or dont add .osa extension in the params they have already added it in the code, do verify the file name as well once it using to save. Also, do check the shutdown function of previous version and this in System.cc they have made some changes for viewer try removing those comments as well if in case later seg fault happens after saving function

dz306271098, 2021-12-29:
@sumitsarkar1 I think I have found the solution. There are maybe two unsuitable operations.

1.  `void Map::PreSave(std::set<GeometricCamera*> &spCams)` in Map.cc. When iterate mspMapPoints, mspMapPoints' size reduces. And I find `void MapPoint::EraseObservation(KeyFrame* pKF)`  can change the 'mspMapPoints' during iteration.
2. The similar situation also occur in `void PreSave(set<KeyFrame*>& spKF,set<MapPoint*>& spMP)` in MapPoint.cc. When iterate `mObservations`, the `mObservations` exists the possibility of change because of `void MapPoint::EraseObservation(KeyFrame* pKF)` during the iteration. 

I have done some rough work on both of them， and it seems successful.
Well, the above is just my guess, I can't guarantee that it will work. So it's just a reference.

sumitsarkar1, 2021-12-29:
@dz306271098 thnx for the suggestions...It will be helpful if you have a forked version of the recent ORBSLAM3 with the rough changes so that I can build that...

dz306271098, 2021-12-29:
@sumitsarkar1 I have updated the rough changes without using mutex in my forked version. And there are also some other problems. You can have a try. Maybe utilizing mutex is more reliable. I am fighting with my course examination recently _(:з」∠)_

Sherlock-hh, 2022-01-11:
Hello, I have save and load the map successfully, but every time I loaded the map, the Tracker didn't use the saved map but create a new map, even though I turn the OnlyTracking-mode on, it's still tracking in the new map, am I use it in a wrong way? How could I use the loaded map to track?
and I check the code I found
```
`bool isRead = LoadAtlas(FileType::BINARY_FILE);

        if(!isRead)
        {
            cout << "Error to load the file, please try with other session file or vocabulary file" << endl;
            exit(-1);
        }
        //mpKeyFrameDatabase = new KeyFrameDatabase(*mpVocabulary);

        //cout << "KF in DB: " << mpKeyFrameDatabase->mnNumKFs << "; words: " << mpKeyFrameDatabase->mnNumWords << endl;

        loadedAtlas = true;

        mpAtlas->CreateNewMap();`
```
and when the first frame came, StereoIntialization will set the first frame as the initial Frame and set it's pose to a identity matrix，how could relocalize in and turn the the Coordinate system to the loaded map

sumitsarkar1, 2022-01-11:
@Sherlock-hh on what are you able to save and load map successfully? Benchmark data sets or Sensors (Realsense)

Sherlock-hh, 2022-01-12:
> @Sherlock-hh on what are you able to save and load map successfully? Benchmark data sets or Sensors (Realsense)

realsense,  stereo, sometimes it's failed to save map with different reasons,  but most are accessing a null pointer, maybe it's because while presaving the map, loopclosing thread or localmapping thread are delete mappoint or keyFrame.

fengepsilon, 2022-03-10:
delete `mpMap->EraseMapPoint(this);` in function   `      void MapPoint::SetBadFlag()`

JACKLiuDay, 2022-04-13:
> delete `mpMap->EraseMapPoint(this);` in function ` void MapPoint::SetBadFlag()`

Hi guy, I tried your method ,while when I try to save map still got seg fault.
[image]

Sherlock-hh, 2022-04-13:
> > delete `mpMap->EraseMapPoint(this);` in function ` void MapPoint::SetBadFlag()`
> 
> Hi guy, I tried your method ,while when I try to save map still got seg fault. [image]

Are u use imu+stereo mode? sorry, I didn't test it at imu+stereo.

JACKLiuDay, 2022-04-13:
> > > delete `mpMap->EraseMapPoint(this);` in function ` void MapPoint::SetBadFlag()`
> > 
> > 
> > Hi guy, I tried your method ,while when I try to save map still got seg fault. [image]
> 
> Are u use imu+stereo mode? sorry, I didn't test it at imu+stereo.

Hi guy, I used it in mono+imu. And I am trying to read the code for digging out the reason.

[image]
I believe this guy show the reason. But I do not know how to deal with it.

Sherlock-hh, 2022-04-13:
yes, most of time it's caused by null pointer , you can add judge if it null when you prepare to save, or add mutex.

JACKLiuDay, 2022-04-13:
> yes, most of time it's caused by null pointer , you can add judge if it null when you prepare to save, or add mutex.

OK, thank you for your remind. I will try to find it. Maybe before saving the map, add some mutex to the map or map points will work.

JACKLiuDay, 2022-04-15:
> > delete `mpMap->EraseMapPoint(this);` in function ` void MapPoint::SetBadFlag()`
> 
> Hi guy, I tried your method ,while when I try to save map still got seg fault. [image]

After some test, I found the mpMap->EraseMapPoint(this) do have bugs when saving map. If we delete this function,  there would be no error when saving a small map. When saving big map , about 100-200M, there still some errors after delete this function. It looks like the program stopped too quick while the map is still under saving. So some delay time should be added in the main function while loop. 
I still have some question. If we deleted the function mpMap->EraseMapPoint(this), the map we built would be bad quality or not？It seems like this function is running when the map is constructing. Or maybe we should implement this function in another form for avoiding the potential segmentation fault.

heiwang1997, 2022-06-14:
Applying the changes from [this fork](https://github.com/dz306271098/ORB_SLAM3) seems to solve my problem. 
Specifically, replace the `Map::PreSave` function and `MapPoint::PreSave` function with the following:

```c++
void Map::PreSave(std::set<GeometricCamera*> &spCams)
{
    int nMPWithoutObs = 0;

    std::set<MapPoint*> tmp_mspMapPoints1;
    tmp_mspMapPoints1.insert(mspMapPoints.begin(), mspMapPoints.end());

    for(MapPoint* pMPi : tmp_mspMapPoints1)
    {
        if(!pMPi || pMPi->isBad())
            continue;

        if(pMPi->GetObservations().size() == 0)
        {
            nMPWithoutObs++;
        }
        map<KeyFrame*, std::tuple<int,int>> mpObs = pMPi->GetObservations();
        for(map<KeyFrame*, std::tuple<int,int>>::iterator it= mpObs.begin(), end=mpObs.end(); it!=end; ++it)
        {
            if(it->first->GetMap() != this || it->first->isBad())
            {
                pMPi->EraseObservation(it->first);
            }

        }
    }

    // Saves the id of KF origins
    mvBackupKeyFrameOriginsId.clear();
    mvBackupKeyFrameOriginsId.reserve(mvpKeyFrameOrigins.size());
    for(int i = 0, numEl = mvpKeyFrameOrigins.size(); i < numEl; ++i)
    {
        mvBackupKeyFrameOriginsId.push_back(mvpKeyFrameOrigins[i]->mnId);
    }

    // Backup of MapPoints
    mvpBackupMapPoints.clear();

    std::set<MapPoint*> tmp_mspMapPoints2;
    tmp_mspMapPoints2.insert(mspMapPoints.begin(), mspMapPoints.end());

    for(MapPoint* pMPi : tmp_mspMapPoints2)
    {
        if(!pMPi || pMPi->isBad())
            continue;

        mvpBackupMapPoints.push_back(pMPi);
        pMPi->PreSave(mspKeyFrames,mspMapPoints);
    }

    // Backup of KeyFrames
    mvpBackupKeyFrames.clear();
    for(KeyFrame* pKFi : mspKeyFrames)
    {
        if(!pKFi || pKFi->isBad())
            continue;

        mvpBackupKeyFrames.push_back(pKFi);
        pKFi->PreSave(mspKeyFrames,mspMapPoints, spCams);
    }

    mnBackupKFinitialID = -1;
    if(mpKFinitial)
    {
        mnBackupKFinitialID = mpKFinitial->mnId;
    }

    mnBackupKFlowerID = -1;
    if(mpKFlowerID)
    {
        mnBackupKFlowerID = mpKFlowerID->mnId;
    }

}
```
and
```c++
void MapPoint::PreSave(set<KeyFrame*>& spKF,set<MapPoint*>& spMP)
{
    mBackupReplacedId = -1;
    if(mpReplaced && spMP.find(mpReplaced) != spMP.end())
        mBackupReplacedId = mpReplaced->mnId;

    mBackupObservationsId1.clear();
    mBackupObservationsId2.clear();
    // Save the id and position in each KF who view it
    std::map<KeyFrame*,std::tuple<int,int> > tmp_mObservations;
    tmp_mObservations.insert(mObservations.begin(), mObservations.end());

    for(std::map<KeyFrame*,std::tuple<int,int> >::const_iterator it = tmp_mObservations.begin(), end = tmp_mObservations.end(); it != end; ++it)
    {
        KeyFrame* pKFi = it->first;
        if(spKF.find(pKFi) != spKF.end())
        {
            mBackupObservationsId1[it->first->mnId] = get<0>(it->second);
            mBackupObservationsId2[it->first->mnId] = get<1>(it->second);
        }
        else
        {
            EraseObservation(pKFi);
        }
    }

    // Save the id of the reference KF
    if(spKF.find(mpRefKF) != spKF.end())
    {
        mBackupRefKFId = mpRefKF->mnId;
    }
}
```

tungtungyan, 2022-06-14:
> 您好，我已經保存並加載地圖成功了，但是每次加載地圖時，Tracker並沒有使用保存的地圖而是創建了一個新地圖，即使我打開了OnlyTracking-mode，它仍然在新的跟踪地圖，我用錯了嗎？如何使用加載的地圖進行跟踪？ 我檢查了我找到的代碼
> 
> ```
> `bool isRead = LoadAtlas(FileType::BINARY_FILE);
> 
>         if(!isRead)
>         {
>             cout << "Error to load the file, please try with other session file or vocabulary file" << endl;
>             exit(-1);
>         }
>         //mpKeyFrameDatabase = new KeyFrameDatabase(*mpVocabulary);
> 
>         //cout << "KF in DB: " << mpKeyFrameDatabase->mnNumKFs << "; words: " << mpKeyFrameDatabase->mnNumWords << endl;
> 
>         loadedAtlas = true;
> 
>         mpAtlas->CreateNewMap();`
> ```
> 
> 而當第一幀到來時，StereoIntialization會將第一幀設置為初始幀並將其位姿設置為單位矩陣，如何重新定位並將坐標系轉為加載的地圖

Did you solve this problem?

heiwang1997, 2022-06-14:
地图应该是加载成功了的，不过得等重定位成功了才能把加载好的地图显示进来吧

tungtungyan, 2022-06-14:
> Hello, I have save and load the map successfully, but every time I loaded the map, the Tracker didn't use the saved map but create a new map, even though I turn the OnlyTracking-mode on, it's still [...]

When I Load the bin File, it shows the problem of std::bad_alloc. How can I solve it?
Initialization of Atlas from file: map_all
Starting to read the save binary file: ./map_all.bin
terminate called after throwing an instance of 'std::bad_alloc'
  what():  std::bad_alloc

Sherlock-hh, 2022-06-15:
> 

maybe the bin file is genarated before? Before u reloaded it，u changed some class Member variables, it's may causing the error. Or u didn't load the map file in the same order of saving, like u save it in order of atlas, map, and keyframedatebase, so when u load it like atlas, map, and keyframedatebase.

lygods, 2022-06-29:
@tungtungyan I can save the map, but there is a segmentation error when loading the 

[image]

aganal, 2022-07-27:
I met similar problem and tried all the method above, but it stil not work. Then  I try to call System::Shutdown() manually, it works though i don't konw why ... 
here are the rough code:
`thread waitShut([&](){
        while (getchar()!='\n');
        flagShutdown = true;
    });
    
    while (!flagShutdown)
    {
       ...
    }
    SLAM.Shutdown();
    waitShut.join();`

maalouf, 2022-07-28:
Hello, saving the map works well (after very slight changes), however loading the map freezes the program, (pangolin freezes on startup and nothing works) no error is printed out. what do you recommend or suggest ?

827346462, 2022-10-31:
> 

helle.I can load osa file . I ture  Onlytracking modls . 
[image]

I change system.cc "mbActivateLocalizationMode(true)" to OnlyTracking.
please you met?

build-error, 2024-03-06:
A lot of people are having problems with saving and loading the map, I was too getting segmentation faults while saving/loading map. 

I believe that the issue with `string strVocabularyChecksum = CalculateCheckSum(mStrVocabularyFilePath,TEXT_FILE);` .
There may be some issue with `CalculateCheckSum()` function.

I ended up replacing this line with string `strVocabularyChecksum = "someSampleText";`.
Make sure to change this line too `string strInputVocabularyChecksum = "someSampleText";`

These changes reduced the number of seg faults I was getting but, it didn't remove completely.

I also observed that when I was using Pangolin to see the GUI, while exiting I am seeing seg faults while the GUI is closing.

Running orbslam3 without GUI also reduced the number of seg faults.

doing all this hasn't removed all the seg faults i am facing but reduced it by a lot.

kameel2311, 2024-04-30:
> Applying the changes from [this fork](https://github.com/dz306271098/ORB_SLAM3) seems to solve my problem. Specifically, replace the `Map::PreSave` function and `MapPoint::PreSave` function with the [...]

Have applied the changes and still yet a segmentation fault occurs. Even forking the full repo and running euroc monocular gives this issue:

```
Loading ORB Vocabulary. This could take a while...
terminate called after throwing an instance of 'std::bad_alloc'
  what():  std::bad_alloc
Stack trace (most recent call last):
#13   Object "[0xffffffffffffffff]", at 0xffffffffffffffff, in 
#12   Object "./Examples/Monocular/mono_euroc", at 0x561607c0a859, in _start
#11   Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f93fd551c86, in __libc_start_main
#10   Object "./Examples/Monocular/mono_euroc", at 0x561607c09226, in main
#9    Object "/root/Dev/ORB_SLAM3/lib/libORB_SLAM3.so", at 0x7f9401810bc1, in ORB_SLAM3::System::System(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, ORB_SLAM3::System::eSensor, bool, int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)
#8    Object "/root/Dev/ORB_SLAM3/lib/libORB_SLAM3.so", at 0x7f940182216c, in DBoW2::TemplatedVocabulary<cv::Mat, DBoW2::FORB>::loadFromBinaryFile(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)
#7    Object "/root/Dev/ORB_SLAM3/lib/libORB_SLAM3.so", at 0x7f9401820f9f, in std::vector<DBoW2::TemplatedVocabulary<cv::Mat, DBoW2::FORB>::Node, std::allocator<DBoW2::TemplatedVocabulary<cv::Mat, DBoW2::FORB>::Node> >::_M_default_append(unsigned long)
#6    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7f93fdbcc2db, in operator new(unsigned long)
#5    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7f93fdbcbd53, in __cxa_throw
#4    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7f93fdbcbb20, in std::terminate()
#3    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7f93fdbcbae5, in 
#2    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6", at 0x7f93fdbc5956, in 
#1    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f93fd5707f0, in abort
#0    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f93fd56ee87, in gsignal
Aborted (Signal sent by tkill() 8807 0)
Aborted (core dumped)
```

the-lgx1, 2024-10-23:
[image]
i dont konw why

hsutzu, 2024-12-11:
Hello evreyone,
i currently met some problem, when i loading map
it could only showing keyframe, and wiothout showing map points and the graph
is anyone having the same issue and have any solution?

Ukaegbu-Uchechi, 2024-12-17:
Hello @hsutzu, any luck with successfully loading the map? Please, can you let me know how you loaded the map? I am quite new in this space.

mirpouya, 2025-01-16:
> [@sumitsarkar1](https://github.com/sumitsarkar1) I have checked. When you save map successfully, the cout gets printed. Like this [image] some of them are added by me. The end cout is from `backward.cc` which is used to show stack trace.

for me:
there's good guideline in #5 
I solved the issue with this docker file:

FROM nvidia/cuda:11.8.0-cudnn8-devel-ubuntu22.04

ENV DEBIAN_FRONTEND=noninteractive

RUN apt-get update && apt-get install -y \
    software-properties-common \
    wget

# gcc
RUN add-apt-repository ppa:ubuntu-toolchain-r/test -y

RUN apt-get install -y gcc-11 g++-11

RUN update-alternatives --install /usr/bin/gcc gcc /usr/bin/gcc-11 100 && \
    update-alternatives --install /usr/bin/g++ g++ /usr/bin/g++-11 100

# dependency
RUN apt-get install -y \
    git \
    build-essential \
    sudo \
    libeigen3-dev \
    libboost-all-dev \
    libjsoncpp-dev \
    libopengl-dev \
    mesa-utils \
    libglfw3-dev \
    libglm-dev \
    python3-pip \
    python3-dev \
    libjpeg-dev \
    libpng-dev \
    libtiff-dev \
    curl \
    zip \
    libavcodec-dev \
    libavformat-dev \
    libswscale-dev \
    libswresample-dev \
    libssl-dev \
    && rm -rf /var/lib/apt/lists/*

# cmake
RUN wget https://github.com/Kitware/CMake/releases/download/v3.22.1/cmake-3.22.1-Linux-x86_64.sh -O /cmake-3.22.1.sh && \
    chmod +x /cmake-3.22.1.sh && \
    /cmake-3.22.1.sh --skip-license --prefix=/usr/local

# opencv
RUN mkdir /opencv && cd /opencv && \
	wget https://github.com/opencv/opencv/archive/refs/tags/4.8.0.zip -O opencv-4.8.0.zip && \
    wget https://github.com/opencv/opencv_contrib/archive/refs/tags/4.8.0.zip -O opencv_contrib-4.8.0.zip && \
    unzip opencv-4.8.0.zip && \
    unzip opencv_contrib-4.8.0.zip && \
    rm opencv-4.8.0.zip && rm opencv_contrib-4.8.0.zip
    
RUN mkdir /opencv/opencv-4.8.0/build && cd /opencv/opencv-4.8.0/build && \
	cmake -DCMAKE_BUILD_TYPE=RELEASE \
          -DWITH_CUDA=ON \
          -DWITH_CUDNN=ON \
          -DOPENCV_DNN_CUDA=ON \
          -DWITH_NVCUVID=ON \
          -DCUDA_TOOLKIT_ROOT_DIR=/usr/local/cuda-11.8 \
          -DOPENCV_EXTRA_MODULES_PATH=/opencv/opencv_contrib-4.8.0/modules \
          -DBUILD_TIFF=ON \
          -DBUILD_ZLIB=ON \
          -DBUILD_JASPER=ON \
          -DBUILD_JPEG=ON \
          -DWITH_FFMPEG=ON \
          .. && \
    make -j$(nproc) && \
    make install && \
    ldconfig

# libtorch
RUN cd / && \
    wget https://download.pytorch.org/libtorch/nightly/cu118/libtorch-cxx11-abi-shared-with-deps-2.4.0.dev20240425%2Bcu118.zip -O libtorch-cu118.zip && \
    unzip libtorch-cu118.zip && rm libtorch-cu118.zip

ENV Torch_DIR /libtorch/share/cmake/Torch

# repo
RUN git clone https://github.com/HuajianUP/Photo-SLAM.git /Photo-SLAM

WORKDIR /Photo-SLAM

everything is the same as https://github.com/HuajianUP/Photo-SLAM/issues/2#issuecomment-2089933703
but the libtorch version is 2.4.0

then you must modify all of:
     c10::guts::to_string(param.unsafeGetTensorImpl());
to:
     param.unsafeGetTensorImpl();
which is mentioned in #5 

and you will see the visualization