UZ-SLAMLab/ORB_SLAM3 #450: example script files are missing!

SeonghoNam, 2021-12-28:
I cannot find example script files(euroc_examples.sh, euroc_eval_examples.sh, tum_vi_examples.sh, tum_vi_eval_examples.sh).
Why these files were deleted in recent commit?

canglangzhige, 2022-04-29:
Me too. What is the problems?

jatinnimawat, 2022-08-22:
Same problem here.

jatinnimawat, 2022-08-22:
I downloaded the previous commit of ORBSLAM3 (made on Sep 29, 2021). It does contain tum_vi_examples.sh in the "Examples" folder of the root directory. It is working for me.

EnriqueSolarte, 2023-03-03:
I found the following information. 
In Dic 22-2021 commit 0df83dde1c85c7ab91a0d47de7a29685d046f637. Several files were added to the .gitignore and some files were deleted. I tracked a stable commit to 578500b836ea05be40d6188b815bb0ddbcdc75f1 on April 21, 2021