UZ-SLAMLab/ORB_SLAM3 #451: SO3::exp failed

DOGEwbx, 2021-12-29:
Thanks for the V1.0 version and the updated functions.
I am running the V1.0 version  with stereo-inertial on my own sensor and the SO3 exp failed to pass the SO3 check in the `GetDeltaRotation` method called by function `EdgeInertial::computeError`.
The message goes like:
```
 SO3::exp failed! omega: nan nan nan, real: nan, img: nan.
```
I wonder if there is anyone who met the same problem, thanks~

knazneen123, 2021-12-29:
Hey, I am also facing the same issue while using Monocular Inertial with data from ros topics

Dcasadoherraez, 2021-12-29:
Thanks for the v1.0 update!
And yep! I'm also facing the issue for stereo in the TUM corridor1 dataset. It does not happen in every run at the same point though. 

`Sophus ensure failed in function 'static Sophus::SO3<Scalar_> Sophus::SO3<Scalar_, Options>::expAndTheta(const Tangent&, Sophus::SO3<Scalar_, Options>::Scalar*) [with Scalar_ = float; int Options = 0; Sophus::SO3<Scalar_, Options>::Tangent = Eigen::Matrix<float, 3, 1>; Sophus::SO3<Scalar_, Options>::Scalar = float]', file '/ssd1_lin/Thesis/ORB-SLAM/ORB_SLAM3/Thirdparty/Sophus/sophus/so3.hpp', line 617.
SO3::exp failed! omega: -nan -nan -nan, real: -nan, img: -nan
Aborted (core dumped)
`

makolele12, 2022-01-04:
Hi, I am facing the exact same issue with Monocular-Inertial while reading data from my own recorded rosbag file.

@Dcasadoherraez in my case the error happens always at the same point in every run, interestingly, when turning around... In the previous beta version I was always meeting  segmentation fault problems when turning around.

DOGEwbx, 2022-01-07:
> Thanks for the V1.0 version and the updated functions. I am running the V1.0 version with stereo-inertial on my own sensor and the SO3 exp failed to pass the SO3 check in the `GetDeltaRotation` [...]
I found that this assertion failed when ORB_SLAM3 tried to merge two maps, which seems to be a longstanding  problem  #416 #319

JACKLiuDay, 2022-01-25:
I meet the same issue while using stereo-imu
[image]

Thiszzz, 2022-05-06:
hello ,i have the same problem  , has anyone solved it？

VINO6520, 2022-06-17:
I face the same issue :
Sophus ensure failed in function 'static SO3<Sophus::SO3::Scalar> Sophus::SO3<float, 0>::expAndTheta(const Sophus::SO3::Tangent &, Sophus::SO3::Scalar *) [Scalar_ = float, Options = 0]', file '../../../../libs/include/sophus/so3.hpp', line 617.
SO3::exp failed! omega: nan nan nan, real: nan, img: nan

AhmadDurrani579, 2025-06-11:
Hey Guys has anyone fixed this issue ? i am also facing the same issue