UZ-SLAMLab/ORB_SLAM3 #452: segmentation fault: examples: rgbd_tum?

jcyhcs, 2021-12-29:
hi,professor:
i test examples/rgbd: command line:
./rgbd_tum ../../Vocabulary/ORBvoc.txt TUM1.yaml rgbd_dataset_freiburg1_xyz associations/fr1_xyz.txt
then run, everything seems ok, snapshot:
[image]
but, when the program ending, it print some error:
First KF:0; Map init KF:0
New Map created with 837 points
Starting the Viewer
Shutdown
-------

median tracking time: 0.0636379
mean tracking time: 0.064307

Saving camera trajectory to CameraTrajectory.txt ...

Saving keyframe trajectory to KeyFrameTrajectory.txt ...
**Segmentation fault (core dumped)**

some weird , is there any bug? oh,please help

cuixuyang615, 2021-12-30:
I have also met the same problem, may I ask have you solved it?

sandstorm666, 2022-03-23:
I encounter this problem，have anyone solved?

sandstorm666, 2022-03-23:
I solved by uncommented some code. LocalMapper is not finished.

sandstorm666, 2022-03-23:
if(mpViewer)
    {
        mpViewer->RequestFinish();
        while(!mpViewer->isFinished())
            usleep(5000);
    }

    //Wait until all thread have effectively stopped
    while(!mpLocalMapper->isFinished() || !mpLoopCloser->isFinished() || mpLoopCloser->isRunningGBA())
    {
        if(!mpLocalMapper->isFinished())
            cout << "mpLocalMapper is not finished" << endl;
        if(!mpLoopCloser->isFinished())
            cout << "mpLoopCloser is not finished" << endl;
        if(mpLoopCloser->isRunningGBA()){
            cout << "mpLoopCloser is running GBA" << endl;
            cout << "break anyway..." << endl;
            break;
        }
        usleep(5000);
    }
these code is needed in System::Shutdown().

nickxiang0306, 2022-07-15:
> ```
> if(mpViewer)
> {
>     mpViewer->RequestFinish();
>     while(!mpViewer->isFinished())
>         usleep(5000);
> }
>
> //Wait until all thread have effectively stopped
> while(!mpLocalMapper->isFinished() || [...]
Hello @sandstorm666 , I appreciate your answer, and it helps me to fix the segfault bug.

JonathanLehner, 2023-02-19:
does not seem to work. I uncommented it but there is still a segmentation fault

SJTUlibochen, 2023-03-02:
> does not seem to work. I uncommented it but there is still a segmentation fault

maybe u need to rebuild it

IndShiv, 2023-05-02:
Hi I am facing the same issue. I have tried uncommenting the above section of the code and rebuilding but I still end up with a Segmentation Fault. Any other suggestions?

yutongwangBIT, 2023-09-14:
Well I found that at the end of System::Shutdown() there is 
```if(mpViewer)  pangolin::BindToContext("ORB-SLAM2: Map Viewer");```
I have solved the segmentation fault by changing ORB-SLAM2 to ORB-SLAM3. Hope this could help.

StanleyOf427, 2024-03-12:
> Well I found that at the end of System::Shutdown() there is `if(mpViewer) pangolin::BindToContext("ORB-SLAM2: Map Viewer");` I have solved the segmentation fault by changing ORB-SLAM2 to ORB-SLAM3. Hope this could help.

That‘s worked, after uncommenting the above section of the code, and modifing 
"    if(mpViewer)
        pangolin::BindToContext("ORB-SLAM2: Map Viewer");
"
 to
"    if(mpViewer)
        pangolin::BindToContext("ORB-SLAM3: Map Viewer");
" 
, you can get the result:
"There are 1 maps in the atlas
  Map 0 has 256 KFs
End of saving trajectory to f_dataset-MH01_mono.txt ...
Saving keyframe trajectory to kf_dataset-MH01_mono.txt ...
(base) guo@M6700:~/ORBSLAM/tmp/ORB_SLAM3-master/Examples$"
afer rebuilding it.

Yangyang221, 2024-04-19:
I did so and still got the same error.
[image]

StanleyOf427, 2024-04-21:
> I did so and still got the same error. ![2024-04-19 16-03-54 的屏幕截图](https://private-user-images.githubusercontent.com/61937281/323881479-986e9f35-5b74-4a65-b7ef-481595fcd71b.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTM2Nzc3MjEsIm5iZiI6MTcxMzY3NzQyMSwicGF0aCI6Ii82MTkzNzI4MS8zMjM4ODE0NzktOTg2ZTlmMzUtNWI3NC00YTY1LWI3ZWYtNDgxNTk1ZmNkNzFiLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA0MjElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNDIxVDA1MzAyMVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWIyOTc3ZjZkMDkxNTdmYTE3OTQzNmI1Mzk3MDYyY2RkZTViMGQ5YTI5MjVkZDg1ZDdhOWFmMTNlNGIxYTNhZmEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.Np0Sj6-sV6wQcu9Nc0fepTHqvyxtPGx-xPo4xlbQxTQ)

Would you like to provide more information about your environment? What's your Ubuntu , compiler and pangolin version ?

Yangyang221, 2024-04-21:
hi, My environment  as follows:
pangolin 0.6,ubuntu20.04

StanleyOf427, 2024-06-07:
maybe you should update pangolin to v0.8, which works well in the same OS.

Lhyshining, 2024-08-09:
> That‘s worked, after uncommenting the above section of the code, and modifing " if(mpViewer) pangolin::BindToContext("ORB-SLAM2: Map Viewer"); " to " if(mpViewer) pangolin::BindToContext("ORB-SLAM3: [...]

Thank you, my pangolin version is 0.6, and i also run tum rgbd dataset. It works for me!

musnows, 2025-02-04:
> Thank you, my pangolin version is 0.6, and i also run tum rgbd dataset. It works for me!

In newest code this path has already been comment, but I still get coredump error. Does anyone know why?

```
-------
Start processing sequence ...
Images in the sequence: 573

First KF:0; Map init KF:0
New Map created with 939 points
Starting the Viewer
Shutdown
-------

median tracking time: 0.0136386
mean tracking time: 0.0140401

Saving camera trajectory to CameraTrajectory.txt ...

Saving keyframe trajectory to KeyFrameTrajectory.txt ...
Segmentation fault (core dumped)

```

[image]