UZ-SLAMLab/ORB_SLAM3 #454: Examples/ROS/ORB_SLAM3 Path

HVKHVK, 2021-12-29:
Examples/ROS/ORB_SLAM3 path changed to Examples_old/ROS/ORB_SLAM3. Update build_ros.sh

DavideLuc, 2021-12-29:
Is ROS working with the new YAML configuration? For me didn't work with stereo configuration, it's asking for old parameters on the YAML file.

HVKHVK, 2021-12-30:
Not really. I was not able to compile the ROS part completely while trying to overcome the issue. I am tracking several other issues and I will wait for more stable library. Till then ORB_SLAM2.

nindanaoto, 2021-12-30:
@HVKHVK You can use the code for this PR for a compilable ROS example.

robin-shaun, 2022-01-03:
I use the PR #447, for pinhole cameras, it's OK. But for fisheye cameras such as T265.  ERROR: Calibration parameters to rectify stereo are missing