UZ-SLAMLab/ORB_SLAM3 #456: CMake Error at CMakeLists.txt:35 (message):   OpenCV > 4.4 not found.

Raman95431, 2022-01-04:
hi 
I install the program through the following link (https://github.com/Mauhing/ORB_SLAM3) and when I get to install ORB-SLAM 3
and run 
./build.sh
i have this problem

./build.sh
Configuring and building Thirdparty/DBoW2 ...
mkdir: cannot create directory ‘build’: File exists
-- Configuring done
-- Generating done
-- Build files have been written to: /home/dark/Dev/ORB_SLAM3/Thirdparty/DBoW2/build
[100%] Built target DBoW2
Configuring and building Thirdparty/g2o ...
mkdir: cannot create directory ‘build’: File exists
-- BUILD TYPE:Release
-- Compiling on Unix
-- Found OpenMP_C: -fopenmp  
-- Found OpenMP_CXX: -fopenmp  
-- Found OpenMP: TRUE   
-- Configuring done
-- Generating done
-- Build files have been written to: /home/dark/Dev/ORB_SLAM3/Thirdparty/g2o/build
[100%] Built target g2o
Configuring and building Thirdparty/Sophus ...
mkdir: cannot create directory ‘build’: File exists
-- Configuring done
-- Generating done
-- Build files have been written to: /home/dark/Dev/ORB_SLAM3/Thirdparty/Sophus/build
[ 16%] Built target test_common
[ 16%] Built target test_geometry
[ 25%] Built target test_so2
[ 33%] Built target test_rxso2
[ 58%] Built target test_velocities
[ 58%] Built target test_rxso3
[ 66%] Built target test_sim3
[ 58%] Built target test_se2
[ 75%] Built target test_se3
[ 91%] Built target test_so3
[ 91%] Built target test_sim2
[100%] Built target HelloSO3
Uncompress vocabulary ...
Configuring and building ORB_SLAM3 ...
mkdir: cannot create directory ‘build’: File exists
Build type: Release
-- Using flag -std=c++11.
CMake Warning at CMakeLists.txt:33 (find_package):
  Could not find a configuration file for package "OpenCV" that is compatible
  with requested version "4.4".

  The following configuration files were considered but not accepted:

    /usr/local/share/OpenCV/OpenCVConfig.cmake, version: 3.2.0
    /usr/lib/x86_64-linux-gnu/cmake/opencv4/OpenCVConfig.cmake, version: 4.2.0
    /lib/x86_64-linux-gnu/cmake/opencv4/OpenCVConfig.cmake, version: 4.2.0

CMake Error at CMakeLists.txt:35 (message):
  OpenCV > 4.4 not found.

-- Configuring incomplete, errors occurred!
See also "/home/dark/Dev/ORB_SLAM3/build/CMakeFiles/CMakeOutput.log".
make: *** No targets specified and no makefile found.  Stop.

makolele12, 2022-01-04:
You dont have OpenCV 4.4.0 installed. Just go to the CMakeFile and change the line where 4.4 is specified and write 4.2.0 instead. I also have 4.2.0 installed and I can confirm it compiles.

Alternatively, install OpenCV 4.4

LMWafer, 2022-01-04:
I can suggest 2 solutions: one with docker, one without.

### Docker one
See my image on [Dockerhub](https://hub.docker.com/r/lmwafer/orb-slam-3-ready), everything is explained. 
This is an image with everything built, installed, and ready to use. It is based on a layer that includes Relasense SDK 2.0 to work with Intel Realsense cameras (like D435i and T265) but I'm working on an RS-free version. Although it has realsense installed you can still use it for ORB-SLAM 3 only. 

### No docker one
***Automatic installation***
"Automatic" installation with `apt-get install libopencv-dev` or similar command does not exceed 4.2.0 version (for now, 04/01/2022). To know your opencv version if installed "automaticaly" simply run 
```bash
locate opencv | grep libopencv # requires locate command : https://www.cyberithub.com/install-locate-command-in-linux/
```
You should get a list with a lot of */var/lib/dpkg/info/libopencv* stuff. Find a line with *libopencvX.Y* in it (probably among the last ones). *X.Y* gives you the version (now I have *libopencv4.2*). You can try to use the path listed by the command or manually install opencv.

***Manual installation***
This has the advantage to gives a lot more control as it yields several files that you can move and whose path can be extracted ! I recommend you to purge your OpenCV previous installation. 
1. Run the following inside an empty directory of your choice. This should install OpenCV with build parameters that work for ORB-SLAM 3. 
```bash
git clone https://github.com/Itseez/opencv.git opencv && \
git clone https://github.com/Itseez/opencv_contrib.git opencv_contrib && \
cd opencv/ && \
mkdir build && \
cd build/ && \
cmake -D CMAKE_BUILD_TYPE=RELEASE \
-D BUILD_TIFF=ON \
-D WITH_CUDA=OFF \
-D ENABLE_AVX=OFF \
-D WITH_OPENGL=OFF \
-D WITH_OPENCL=OFF \
-D WITH_IPP=OFF \
-D WITH_TBB=ON \
-D BUILD_TBB=ON \
-D WITH_EIGEN=ON \
-D WITH_V4L=OFF \
-D WITH_VTK=OFF \
-D BUILD_TESTS=OFF \
-D BUILD_PERF_TESTS=OFF \
-D OPENCV_GENERATE_PKGCONFIG=ON \
-D OPENCV_EXTRA_MODULES_PATH=../opencv_contrib/modules \
.. && \
make && \
make install && \
ldconfig
```
2. Enter the directory with ORB-SLAM3 repo. Inside its *build.sh* add `-D CMAKE_PREFIX_PATH=/usr/lib/x86_64-linux-gnu/` parameter to both DBow2 and ORB `cmake` paragraphs. You should have something like this : 
```bash
cmake .. -DCMAKE_BUILD_TYPE=Release -D CMAKE_PREFIX_PATH=/usr/lib/x86_64-linux-gnu/
```
This parameter tells CMake to look for a file named *opencv4.pc*. It contains information about the version, available modules, flags, etc. If ORB installation keeps failing consider changing the location of this file or the search path in ORB CMake system. Or simply run my Docker image :wink: .

Raman95431, 2022-01-05:
> You dont have OpenCV 4.4.0 installed. Just go to the CMakeFile and change the line where 4.4 is specified and write 4.2.0 instead. I also have 4.2.0 installed and I can confirm it compiles.
> 
> Alternatively, install OpenCV 4.4

Thank you it work