UZ-SLAMLab/ORB_SLAM3 #458: ./build.sh Error >> make: *** [Makefile:84: all] Error 2

mark-91, 2022-01-05:
ubuntu 20.4
install opencv + Pangolin
clone orb-slam3
./build.sh>>>>

[ 20%] Building CXX object Thirdparty/g2o/CMakeFiles/g2o.dir/g2o/stuff/property.cpp.o
[ 21%] Linking CXX shared library ../../../Thirdparty/g2o/lib/libg2o.so
[ 21%] Built target g2o
In file included from /home/ai/work/Ram/Dev/ORB_SLAM3/include/Frame.h:30,
                 from /home/ai/work/Ram/Dev/ORB_SLAM3/include/KeyFrame.h:28,
                 from /home/ai/work/Ram/Dev/ORB_SLAM3/include/MapPoint.h:23,
                 from /home/ai/work/Ram/Dev/ORB_SLAM3/include/FrameDrawer.h:24,
                 from /home/ai/work/Ram/Dev/ORB_SLAM3/include/Viewer.h:23,
                 from /home/ai/work/Ram/Dev/ORB_SLAM3/include/Tracking.h:26,
                 from /home/ai/work/Ram/Dev/ORB_SLAM3/src/Tracking.cc:20:
/home/ai/work/Ram/Dev/ORB_SLAM3/include/ImuTypes.h: In member function ‘void ORB_SLAM3::IMU::Preintegrated::printMeasurements() const’:
/home/ai/work/Ram/Dev/ORB_SLAM3/include/ImuTypes.h:204:23: warning: comparison of integer expressions of different signedness: ‘int’ and ‘std::vector<ORB_SLAM3::IMU::Preintegrated::integrable>::size_type’ {aka ‘long unsigned int’} [-Wsign-compare]
  204 |         for(int i=0; i<mvMeasurements.size(); i++){
      |                      ~^~~~~~~~~~~~~~~~~~~~~~
In file included from /home/ai/work/Ram/Dev/ORB_SLAM3/Thirdparty/g2o/g2o/types/types_six_dof_expmap.h:37,
                 from /home/ai/work/Ram/Dev/ORB_SLAM3/include/Converter.h:26,
                 from /home/ai/work/Ram/Dev/ORB_SLAM3/include/Frame.h:33,
                 from /home/ai/work/Ram/Dev/ORB_SLAM3/include/KeyFrame.h:28,
                 from /home/ai/work/Ram/Dev/ORB_SLAM3/include/MapPoint.h:23,
                 from /home/ai/work/Ram/Dev/ORB_SLAM3/include/FrameDrawer.h:24,
                 from /home/ai/work/Ram/Dev/ORB_SLAM3/include/Viewer.h:23,
                 from /home/ai/work/Ram/Dev/ORB_SLAM3/include/Tracking.h:26,
                 from /home/ai/work/Ram/Dev/ORB_SLAM3/src/Tracking.cc:20:
/home/ai/work/Ram/Dev/ORB_SLAM3/Thirdparty/g2o/g2o/types/../core/base_vertex.h: At global scope:
/home/ai/work/Ram/Dev/ORB_SLAM3/Thirdparty/g2o/g2o/types/../core/base_vertex.h:62:74: warning: ‘Eigen::AlignedBit’ is deprecated [-Wdeprecated-declarations]
   62 |     typedef Eigen::Map<Matrix<double, D, D>, Matrix<double,D,D>::Flags & AlignedBit ? Aligned : Unaligned >  HessianBlockType;
      |                                                                          ^~~~~~~~~~
In file included from /usr/include/eigen3/Eigen/Core:363,
                 from /usr/include/eigen3/Eigen/LU:11,
                 from /home/ai/work/Ram/Dev/ORB_SLAM3/Thirdparty/Sophus/sophus/so2.hpp:12,
                 from /home/ai/work/Ram/Dev/ORB_SLAM3/Thirdparty/Sophus/sophus/se2.hpp:7,
                 from /home/ai/work/Ram/Dev/ORB_SLAM3/Thirdparty/Sophus/sophus/geometry.hpp:7,
                 from /home/ai/work/Ram/Dev/ORB_SLAM3/include/Frame.h:28,
                 from /home/ai/work/Ram/Dev/ORB_SLAM3/include/KeyFrame.h:28,
                 from /home/ai/work/Ram/Dev/ORB_SLAM3/include/MapPoint.h:23,
                 from /home/ai/work/Ram/Dev/ORB_SLAM3/include/FrameDrawer.h:24,
                 from /home/ai/work/Ram/Dev/ORB_SLAM3/include/Viewer.h:23,
                 from /home/ai/work/Ram/Dev/ORB_SLAM3/include/Tracking.h:26,
                 from /home/ai/work/Ram/Dev/ORB_SLAM3/src/Tracking.cc:20:
/usr/include/eigen3/Eigen/src/Core/util/Constants.h:162:37: note: declared here
  162 | EIGEN_DEPRECATED const unsigned int AlignedBit = 0x80;
      |                                     ^~~~~~~~~~
In file included from /home/ai/work/Ram/Dev/ORB_SLAM3/Thirdparty/g2o/g2o/types/types_six_dof_expmap.h:38,
                 from /home/ai/work/Ram/Dev/ORB_SLAM3/include/Converter.h:26,
                 from /home/ai/work/Ram/Dev/ORB_SLAM3/include/Frame.h:33,
                 from /home/ai/work/Ram/Dev/ORB_SLAM3/include/KeyFrame.h:28,
                 from /home/ai/work/Ram/Dev/ORB_SLAM3/include/MapPoint.h:23,
                 from /home/ai/work/Ram/Dev/ORB_SLAM3/include/FrameDrawer.h:24,
                 from /home/ai/work/Ram/Dev/ORB_SLAM3/include/Viewer.h:23,
                 from /home/ai/work/Ram/Dev/ORB_SLAM3/include/Tracking.h:26,
                 from /home/ai/work/Ram/Dev/ORB_SLAM3/src/Tracking.cc:20:
/home/ai/work/Ram/Dev/ORB_SLAM3/Thirdparty/g2o/g2o/types/../core/base_binary_edge.h:59:82: warning: ‘Eigen::AlignedBit’ is deprecated [-Wdeprecated-declarations]
   59 |       typedef Eigen::Map<Matrix<double, Di, Dj>, Matrix<double, Di, Dj>::Flags & AlignedBit ? Aligned : Unaligned > HessianBlockType;
      |                                                                                  ^~~~~~~~~~
In file included from /usr/include/eigen3/Eigen/Core:363,
                 from /usr/include/eigen3/Eigen/LU:11,
                 from /home/ai/work/Ram/Dev/ORB_SLAM3/Thirdparty/Sophus/sophus/so2.hpp:12,
                 from /home/ai/work/Ram/Dev/ORB_SLAM3/Thirdparty/Sophus/sophus/se2.hpp:7,
                 from /home/ai/work/Ram/Dev/ORB_SLAM3/Thirdparty/Sophus/sophus/geometry.hpp:7,
                 from /home/ai/work/Ram/Dev/ORB_SLAM3/include/Frame.h:28,
                 from /home/ai/work/Ram/Dev/ORB_SLAM3/include/KeyFrame.h:28,
                 from /home/ai/work/Ram/Dev/ORB_SLAM3/include/MapPoint.h:23,
                 from /home/ai/work/Ram/Dev/ORB_SLAM3/include/FrameDrawer.h:24,
                 from /home/ai/work/Ram/Dev/ORB_SLAM3/include/Viewer.h:23,
                 from /home/ai/work/Ram/Dev/ORB_SLAM3/include/Tracking.h:26,
[... 1467 lines cut ...]
      |                  ^~~~~
In file included from /usr/local/include/pangolin/utils/signal_slot.h:3,
                 from /usr/local/include/pangolin/windowing/window.h:35,
                 from /usr/local/include/pangolin/display/display.h:34,
                 from /usr/local/include/pangolin/pangolin.h:38,
                 from /home/ai/work/Ram/Dev/ORB_SLAM3/include/Map.h:27,
                 from /home/ai/work/Ram/Dev/ORB_SLAM3/include/KeyFrameDatabase.h:30,
                 from /home/ai/work/Ram/Dev/ORB_SLAM3/include/KeyFrame.h:29,
                 from /home/ai/work/Ram/Dev/ORB_SLAM3/include/MapPoint.h:23,
                 from /home/ai/work/Ram/Dev/ORB_SLAM3/include/FrameDrawer.h:24,
                 from /home/ai/work/Ram/Dev/ORB_SLAM3/include/Viewer.h:23,
                 from /home/ai/work/Ram/Dev/ORB_SLAM3/include/Tracking.h:26,
                 from /home/ai/work/Ram/Dev/ORB_SLAM3/src/Tracking.cc:20:
/usr/local/include/sigslot/signal.hpp: In instantiation of ‘void sigslot::signal_base< <template-parameter-1-1>, <template-parameter-1-2> >::operator()(U&& ...) [with U = {pangolin::VarState::Event}; Lockable = std::mutex; T = {pangolin::VarState::Event}]’:
/usr/local/include/pangolin/var/varstate.h:264:23:   required from ‘std::map<std::__cxx11::basic_string<char>, std::shared_ptr<pangolin::VarValueGeneric> >::iterator pangolin::VarState::AddUpgradedVar(const std::shared_ptr<pangolin::VarValue<T> >&, const iterator&, bool) [with T = bool; std::map<std::__cxx11::basic_string<char>, std::shared_ptr<pangolin::VarValueGeneric> >::iterator = std::_Rb_tree_iterator<std::pair<const std::__cxx11::basic_string<char>, std::shared_ptr<pangolin::VarValueGeneric> > >]’
/usr/local/include/pangolin/var/varstate.h:205:13:   required from ‘std::shared_ptr<pangolin::VarValueGeneric> pangolin::VarState::GetOrCreateVar(const T&, const pangolin::VarMeta&) [with T = bool]’
/usr/local/include/pangolin/var/var.h:83:50:   required from ‘pangolin::Var<T>::Var(const T&, const pangolin::VarMeta&) [with T = bool]’
/usr/local/include/pangolin/var/var.h:88:35:   required from ‘pangolin::Var<T>::Var(const string&, const T&) [with T = bool; std::string = std::__cxx11::basic_string<char>]’
/usr/local/include/pangolin/display/widgets.h:153:50:   required from here
/usr/local/include/sigslot/signal.hpp:1180:65: error: ‘slots_reference’ was not declared in this scope
 1180 |         cow_copy_type<list_type, Lockable> ref = slots_reference();
      |                                                  ~~~~~~~~~~~~~~~^~
make[2]: *** [CMakeFiles/ORB_SLAM3.dir/build.make:76: CMakeFiles/ORB_SLAM3.dir/src/Tracking.cc.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:971: CMakeFiles/ORB_SLAM3.dir/all] Error 2
make: *** [Makefile:84: all] Error 2

kastoria3, 2022-01-06:
What OS+version and compiler+version are you using?  It looks like your compiler does not have the same version of the C++ STDLIB that the code base is using.

brianm-sra, 2022-01-07:
I am also seeing build errors similar to those shown above, with

Linux Ubuntu 18.04.6 LTS

make version is GNU Make 4.1, cmake version 3.21.3, c++ / cc / gcc are version "(Ubuntu 7.5.0-3ubuntu1~18.04) 7.5.0"

Trying to build ORB_SLAM3 v1.0 release (December 2021)

The errors that I am seeing include:

could not find "realsense2"
‘decay_t’ is not a member of ‘std’
‘remove_pointer_t’ is not a member of ‘std’
‘enable_if_t’ is not a member of ‘std’
...
‘m_slots’ was not declared in this scope
‘slots_reference’ was not declared in this scope

brianm-sra, 2022-01-07:
The fix is to follow the suggestion from

https://github.com/UZ-SLAMLab/ORB_SLAM3/issues/387

`sed -i 's/++11/++14/g' CMakeLists.txt`

and then re-run

build.sh

@mark-91 : see if that helps you

Subjectivist, 2022-01-07:
install _boost_ and upgrade _openssl_ may help

ryangoh206, 2022-01-19:
This works, thanks @brianm-sra

mark-91, 2022-01-25:
thanks @brianm-sra

closed 2022-01-25 by mark-91

Ali-fe, 2022-04-28:
it doesn't work for me, how can i solve it?

Mechazo11, 2022-10-14:
> it doesn't work for me, how can i solve it?

@Ali-fe did you get the problem resolved?

qn757275373, 2023-05-12:
> The fix is to follow the suggestion from
> 
> #387
> 
> `sed -i 's/++11/++14/g' CMakeLists.txt`
> 
> and then re-run
> 
> build.sh
> 
> @mark-91 : see if that helps you

thanks so much ! I love you