UZ-SLAMLab/ORB_SLAM3 #464: running EuRoC Examples but nothing shows up

KaqiKaqi, 2022-01-08:
The output in terminal is as follows.
Initialization of Atlas from scratch 
Creation of new map with id: 0
Creation of new map with last KF id: 0
Seq. Name: 
There are 1 cameras in the atlas
Camera 0 is pinhole
[ INFO:0] global /home/ber/Documents/software/opencv-4.5.4/modules/core/src/parallel/registry_parallel.impl.hpp (96) ParallelBackendRegistry core(parallel): Enabled backends(3, sorted by priority): ONETBB(1000); TBB(990); OPENMP(980)
First KF:0; Map init KF:0
New Map created with 271 points

The program holds but no window shows up.
Have no idea what's going on. Can anyone help?

jiejie567, 2022-01-09:
I have the same problem.

NaleRaphael, 2022-01-10:
Hi @KaqiKaqi,

TL; DR, taking `Examples/Monocular/mono_euroc.cc` as example:
```bash
$ cd ORB_SLAM3

# Search the line `ORB_SLAM3::System SLAM(argv[1],argv[2],ORB_SLAM3::System::MONOCULAR, false);`
# the modify the 4th argument from `false` to `true`.
$ vim Examples/Monocular/mono_euroc.cc

# Rebuild example
$ cd build
$ make

# Run example again (assume that you have already prepared the dataset)
$ cd ../Examples
$ ./Monocular/mono_euroc ../Vocabulary/ORBvoc.txt ./Monocular/EuRoC.yaml PATH_TO_YOUR_MH01_DATASET ./Monocular/EuRoC_TimeStamps/MH01.txt dataset-MH01_mono
```

---
That might be the 4th argument `bUseViewer` in examples for initializing SLAM system has been changed recently in the latest version (`v1.0-release`). If you want to enable the viewer, you can simply modify the 4th argument from `false` to `true` in corresponding source file of example (e.g., `Examples/Monocular/mono_euroc.cc`), then rebuild it again.

And if you have already built the ORB SLAM3 library successfully, you simply go to the `build` directory (which should locate under the root directory of this repository) and run the command `make` again. This would avoid rebuilding the whole library, and which could save you lots more time.

For more details, please checkout the following code:
- Function signature of `ORB_SLAM3::System`: https://github.com/UZ-SLAMLab/ORB_SLAM3/blob/851db08347849a94c51d3ebc3a36df8a114b800f/src/System.cc#L41-L44
- Code for initialize SLAM system (v1.0-release) `mono_euroc` example, `bUseViewer` is `false`: https://github.com/UZ-SLAMLab/ORB_SLAM3/blob/0df83dde1c85c7ab91a0d47de7a29685d046f637/Examples/Monocular/mono_euroc.cc#L82-L83
- Code for initialize SLAM system (v0.4-beta) for `mono_euroc` example, `bUseViewer` is `true`: https://github.com/UZ-SLAMLab/ORB_SLAM3/blob/093173baccc70930c87c64bce5b54e5664d6535f/Examples/Monocular/mono_euroc.cc#L81-L82

KaqiKaqi, 2022-01-12:
> Hi @KaqiKaqi,
> 
> TL; DR, taking `Examples/Monocular/mono_euroc.cc` as example:
> 
> ```shell
> $ cd ORB_SLAM3
> 
> # Search the line `ORB_SLAM3::System SLAM(argv[1],argv[2],ORB_SLAM3::System::MONOCULAR, false);`
> # the modify the 4th argument from `false` to `true`.
> $ vim Examples/Monocular/mono_euroc.cc
> 
> # Rebuild example
> $ cd build
> $ make
> 
> # Run example again (assume that you have already prepared the dataset)
> $ cd ../Examples
> $ ./Monocular/mono_euroc ../Vocabulary/ORBvoc.txt ./Monocular/EuRoC.yaml PATH_TO_YOUR_MH01_DATASET ./Monocular/EuRoC_TimeStamps/MH01.txt dataset-MH01_mono
> ```
> 
> That might be the 4th argument `bUseViewer` in examples for initializing SLAM system has been changed recently in the latest version (`v1.0-release`). If you want to enable the viewer, you can simply modify the 4th argument from `false` to `true` in corresponding source file of example (e.g., `Examples/Monocular/mono_euroc.cc`), then rebuild it again.
> 
> And if you have already built the ORB SLAM3 library successfully, you simply go to the `build` directory (which should locate under the root directory of this repository) and run the command `make` again. This would avoid rebuilding the whole library, and which could save you lots more time.
> 
> For more details, please checkout the following code:
> 
>     * Function signature of `ORB_SLAM3::System`: https://github.com/UZ-SLAMLab/ORB_SLAM3/blob/851db08347849a94c51d3ebc3a36df8a114b800f/src/System.cc#L41-L44
> 
>     * Code for initialize SLAM system (v1.0-release) `mono_euroc` example, `bUseViewer` is `false`: https://github.com/UZ-SLAMLab/ORB_SLAM3/blob/0df83dde1c85c7ab91a0d47de7a29685d046f637/Examples/Monocular/mono_euroc.cc#L82-L83
> 
>     * Code for initialize SLAM system (v0.4-beta) for `mono_euroc` example, `bUseViewer` is `true`: https://github.com/UZ-SLAMLab/ORB_SLAM3/blob/093173baccc70930c87c64bce5b54e5664d6535f/Examples/Monocular/mono_euroc.cc#L81-L82

This works for me. Thanks a lot :)

But Segmentation fault appeared at the end of the program as follows.
`Fail to track local map!
Fail to track local map!
Fail to track local map!
Fail to track local map!
Relocalized!!
Shutdown

Saving trajectory to f_dataset-MH01_mono.txt ...
There are 1 maps in the atlas
  Map 0 has 324 KFs

End of saving trajectory to f_dataset-MH01_mono.txt ...

Saving keyframe trajectory to kf_dataset-MH01_mono.txt ...
[ INFO:0] global /home/ber/Documents/software/opencv-4.5.4/modules/highgui/src/window_gtk.cpp (1273) destroy OpenCV/UI: destroying GTK window: ORB-SLAM3: Current Frame
./run_euroc.sh: line 1: 11458 Segmentation fault      (core dumped) ./Examples/Monocular/mono_euroc ./Vocabulary/ORBvoc.txt ./Examples/Monocular/EuRoC.yaml /home/redbear/Documents/data/euroc/MH01 ./Examples/Monocular/EuRoC_TimeStamps/MH01.txt dataset-MH01_mono`

closed 2022-01-14 by KaqiKaqi

dholukeval, 2022-06-16:
> Hi @KaqiKaqi,
>
> TL; DR, taking `Examples/Monocular/mono_euroc.cc` as example:
>
> ```shell
> $ cd ORB_SLAM3
>
> # Search the line `ORB_SLAM3::System SLAM(argv[1],argv[2],ORB_SLAM3::System::MONOCULAR, [...]

Thank you very much. That worked!!

kenhuang1964, 2023-11-15:
> This works for me. Thanks a lot :)
>
> But Segmentation fault appeared at the end of the program as follows. `Fail to track local map! Fail to track local map! Fail to track local map! Fail to track [...]

Hey @KaqiKaqi, did you end up resolving the Segmentation fault appeared at the end of the program? I am facing the same issue ;(

Redalligator291, 2024-02-21:
> Hey @KaqiKaqi, did you end up resolving the Segmentation fault appeared at the end of the program? I am facing the same issue ;(

Hey guys, I am also running into this same issue, did anyone find a solution to this problem. I am using the EuRoc dataset as well, and I am on linux.

NaleRaphael, 2024-02-21:
@KaqiKaqi Sorry, I missed the notification, it might be too late for your situation, but still hope you get through it.

To All, currently I don't have resources to run that example, so I can only provide some tips to help you figure it out.

In the last few lines of your message, it shows "Saving keyframe trajectory to kf_dataset-MH01_mono.txt ...", that message should come from [`System::SaveKeyFrameTrajectoryEuRoC()`][f01]. And since it's the last operation in `main()@Examples/Monocular/mono_euroc.cc`, there might be something wrong while saving keyframe trajectory.

But since there is no sufficient information for further investigation, I would suggest running the program with gdb and check with the stack trace. For example:
```bash
# format: gdb -ex run --args <PROGRAM> [args...]
# let's take the example above:
$ gdb -ex run --args ./Monocular/mono_euroc ../Vocabulary/ORBvoc.txt ./Monocular/EuRoC.yaml PATH_TO_YOUR_MH01_DATASET ./Monocular/EuRoC_TimeStamps/MH01.txt dataset-MH01_mono

# when it stopped while segmentation fault occurs, enter "bt" under the interactive mode to print stack trace
```

If you found the information in stack trace is still insufficient to figure out the problem, you might want to rebuilt the library and ROS examples in debug mode to get more details.

For `libORB_SLAM3`, you can set `CMAKE_BUILD_TYPE` to `Debug` or `RelWithDebInfo` (release with debug info) the in last section of [`build.sh`][1].
For ROS examples, build type is set in [this line][2].
> [!NOTE]  
> 1. You might **not** need to change all `CMAKE_BUILD_TYPE`s in those scripts at once. Try to get more details from the stack trace of `libORB_SLAM3` first before applying further changes to other dependencies.
> 2. If the build type of any other dependencies is changed, remember to rebuild `libORB_SLAM3` as well. (you don't need to worry about this if you always run `build.sh`)
> 3. [Insert breakpoints][3] (even to specific thread) would help you investigate it.
> 4. If you are running the example with `roslaunch`, you can add `launch-prefix="gdb -ex run -ex bt --args"` to the node in launch file, it would run with gdb and show stack trace automatically right after segmentation fault. See also [this document][4].

[f01]: https://github.com/UZ-SLAMLab/ORB_SLAM3/blob/4452a3c/src/System.cc#L1057-L1059
[f02]: https://github.com/UZ-SLAMLab/ORB_SLAM3/blob/4452a3c/Examples/Monocular/mono_euroc.cc#L192-L195
[1]: https://github.com/UZ-SLAMLab/ORB_SLAM3/blob/4452a3c/build.sh#L39
[2]: https://github.com/UZ-SLAMLab/ORB_SLAM3/blob/4452a3c/build_ros.sh#L6
[3]: https://ftp.gnu.org/old-gnu/Manuals/gdb/html_node/gdb_28.html
[4]: https://wiki.ros.org/roslaunch/Tutorials/Roslaunch%20Nodes%20in%20Valgrind%20or%20GDB