UZ-SLAMLab/ORB_SLAM3 #466: Suspected bug encountered when running the EuRoC dataset with MonoInertial

RayShark0605, 2022-01-11:
I successfully compiled **the latest version of ORB-SLAM3 on Windows 10** by Visual Studio 2017 and tested Mono-EuRoC, Mono-Kitti, Mono-TUM without problems.
However, when running the EuRoC dataset in **MonoInertial**, there is no problem at the beginning, and an error will be reported in the half-way. The error point is located in the following position in the "**LocalInertialBA**" function in "**Optimizer.cc**"(around line 2835)
`
...
for(map<int,int>::iterator mit=mVisEdges.begin(),mend=mVisEdges.end(); mit!=mend; mit++)
{
          assert(mit->second>=3);    //Error reported here
}
optimizer.initializeOptimization();
...
`

I don't know what the  **assertion** here does. Could someone tell me what should I do?
[image]

DOGEwbx, 2022-01-11:
Maybe you complile the source code with DEBUG flag? #145 shows that the RELEASE flag will figure it out.

RayShark0605, 2022-01-11:
> Maybe you complile the source code with DEBUG flag? #145 shows that the RELEASE flag will figure it out.

Thanks Reply!
After I adjust to Debug mode, assert errors **in other locations** are always triggered, almost all of which are asserts in the function library of **VC itself**. That's why I use Release mode.

0y8w1x, 2022-03-09:
@RayShark0605 I tried to compile for windows aswell, but with no success so far. Could you share the windows version please ?