UZ-SLAMLab/ORB_SLAM3 #469: running mono_euroc with MH01 successfuly, but has segmentation fault(core dumped) at the end

KaqiKaqi, 2022-01-14:
I ran mono_euroc with MH01 successfuly, but has segmentation fault at the end. log is as follows:
```
num_seq = 1
file name: dataset-MH01_mono
Loading images for sequence 0...LOADED!

-------

ORB-SLAM3 Copyright (C) 2017-2020 Carlos Campos, Richard Elvira, Juan J. Gómez, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
ORB-SLAM2 Copyright (C) 2014-2016 Raúl Mur-Artal, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
This program comes with ABSOLUTELY NO WARRANTY;
This is free software, and you are welcome to redistribute it
under certain conditions. See LICENSE.txt.

Input sensor was set to: Monocular
Loading settings from ./Examples/Monocular/EuRoC.yaml
Camera1.k3 optional parameter does not exist...
	-Loaded camera 1
	-Loaded image info
	-Loaded ORB settings
Viewer.imageViewScale optional parameter does not exist...
	-Loaded viewer settings
System.LoadAtlasFromFile optional parameter does not exist...
System.SaveAtlasToFile optional parameter does not exist...
	-Loaded Atlas settings
System.thFarPoints optional parameter does not exist...
	-Loaded misc parameters
----------------------------------
SLAM settings: 
	-Camera 1 parameters (Pinhole): [ 458.65399169921875 457.29598999023438 367.21499633789062 248.375 ]
	-Camera 1 distortion parameters: [  -0.28340810537338257 0.073959067463874817 0.00019359000725671649 1.7618711353861727e-05 ]
	-Original image size: [ 752 , 480 ]
	-Current image size: [ 600 , 350 ]
	-Camera 1 parameters after resize: [  365.94735717773438 333.44500732421875 292.99069213867188 181.10678100585938 ]
	-Sequence FPS: 20
	-Features per image: 1000
	-ORB scale factor: 1.2000000476837158
	-ORB number of scales: 8
	-Initial FAST threshold: 20
	-Min FAST threshold: 7

Loading ORB Vocabulary. This could take a while...
Vocabulary loaded!

Initialization of Atlas from scratch 
Creation of new map with id: 0
Creation of new map with last KF id: 0
Seq. Name: 
There are 1 cameras in the atlas
Camera 0 is pinhole
[ INFO:0@4.408] global /home/ber/workspace/code/opencv-4.5.5/modules/core/src/parallel/registry_parallel.impl.hpp (96) ParallelBackendRegistry core(parallel): Enabled backends(3, sorted by priority): ONETBB(1000); TBB(990); OPENMP(980)
[ INFO:16@4.462] global /home/ber/workspace/code/opencv-4.5.5/modules/highgui/src/registry.impl.hpp (114) UIBackendRegistry UI: Enabled backends(2, sorted by priority): GTK(1000); GTK2(990) + BUILTIN(GTK2)
[ INFO:16@4.462] global /home/ber/workspace/code/opencv-4.5.5/modules/highgui/src/backend.cpp (90) createUIBackend UI: using backend: GTK (priority=1000)
[ INFO:16@4.462] global /home/ber/workspace/code/opencv-4.5.5/modules/highgui/src/window_gtk.cpp (2364) createWindow OpenCV/UI: Creating GTK window: ORB-SLAM3: Current Frame (1)
[ INFO:16@4.470] global /home/ber/workspace/code/opencv-4.5.5/modules/highgui/src/window_gtk.cpp (573) CvWindow OpenCV/UI: creating GTK window: ORB-SLAM3: Current Frame
Starting the Viewer
First KF:0; Map init KF:0
New Map created with 271 points
Shutdown

Saving trajectory to f_dataset-MH01_mono.txt ...
There are 1 maps in the atlas
  Map 0 has 342 KFs

End of saving trajectory to f_dataset-MH01_mono.txt ...

Saving keyframe trajectory to kf_dataset-MH01_mono.txt ...
[ INFO:0@196.009] global /home/ber/workspace/code/opencv-4.5.5/modules/highgui/src/window_gtk.cpp (1273) destroy OpenCV/UI: destroying GTK window: ORB-SLAM3: Current Frame
./run_mh01.sh: line 1: 21428 Segmentation fault      (core dumped) ./Examples/Monocular/mono_euroc ./Vocabulary/ORBvoc.txt ./Examples/Monocular/EuRoC.yaml /home/ber/data/euroc/MH01 ./Examples/Monocular/EuRoC_TimeStamps/MH01.txt dataset-MH01_mono
```

But no segmentation fault happed when running MH02.
Does anyone have some idea?

yhabib29, 2022-01-25:
Try to print the `pBiggerMap` Map pointer in System.cc `SaveKeyFrameTrajectoryEuRoC` function.
You might see that the pointer is `null` because your map does not have enough keyframes.

yhabib29, 2022-01-25:
You should initialize the `Map` pointer with a `nullptr`, this solve the problem:
Replace:

by:
`Map* pBiggerMap == nullptr;`

aichishiguocai, 2022-04-14:
hello @jdtardos @yhabib29 !
when I ran "./Examples/Stereo-Inertial/stereo_inertial_realsense_D435i Vocabulary/ORBvoc.txt ./Examples/Stereo-Inertial/RealSense_D435i.yaml",it has a segmentation fault at the end,log is as follows:
"SLAM settings:-Camera 1 parameters (Pinhole): [382.613 382.613 320.183 236.455]"
Could you please give some suggestions?
Thanks!

aichishiguocai, 2022-04-15:
Hello@yhabib29
Thanks for your reply,when  I change the line as you mentioned,it has another error,as fllows:
“Loading ORB Vocabulary.This could take a while ,Illegal instruction”.
I do not know what it occurs,what's your ideas?

aichishiguocai, 2022-04-15:
Hello @yhabib29 !
Thanks for your reply,when I change the line as you mentioned,it has another error,as fllows:
“Loading ORB Vocabulary.This could take a while ,Illegal instruction”.
I do not know what it occurs,what's your ideas?

yhabib29, 2022-04-26:
Sorry, no idea on this error, it is not supposed to be linked with the changes I suggested previously.
If you undo the previous change only, does it still happen ?
Maybe send a screenshot of the program output ...

nickxiang0306, 2022-07-15:
https://github.com/UZ-SLAMLab/ORB_SLAM3/issues/452 maybe this issue could help you fix this segfault bug. And I fixed the bug by following its solution.

MynhLoveBunCha, 2023-09-07:
> hello @jdtardos @yhabib29 ! when I ran "./Examples/Stereo-Inertial/stereo_inertial_realsense_D435i Vocabulary/ORBvoc.txt ./Examples/Stereo-Inertial/RealSense_D435i.yaml",it has a segmentation fault [...]

Hi @miscedence12, I am having the same issue. Did you fix the problem?

miscedence12, 2023-09-07:
@MynhLoveBunCha hello, I haven't used this algorithm for over a year, and I remember we solved this problem very well at that time, it seems to be a problem with the camera configuration file,best wishes!