UZ-SLAMLab/ORB_SLAM3 #471: Tracking::Reset reports an error

panping1314, 2022-01-17:
backtrace:
      #00 pc 00000000000e7340  /apex/com.android.runtime/lib64/bionic/libc.so (pthread_mutex_lock) (BuildId: 91cd22218589ba7b9ea39ce46bc68dff)
      #01 pc 0000000000093ec0  /data/app/com.sim.slamar3N-lZvYl5P2IqjfhAJ7YakCwQ==/lib/arm64/libc++_shared.so (std::__ndk1::mutex::lock()+8) (BuildId: 6de1c39a010a6596b5bb6546e6fe55445b037c1b)
      #02 pc 000000000018e4f4  /data/app/com.sim.slamar3N-lZvYl5P2IqjfhAJ7YakCwQ==/lib/arm64/libSlam.so (ORB_SLAM3::KeyFrame::GetPose()+36) (BuildId: d77a3e88b9b18f39f46d2f1f87d119b1f0920dce)
      #03 pc 00000000002a7648  /data/app/com.sim.slamar3N-lZvYl5P2IqjfhAJ7YakCwQ==/lib/arm64/libSlam.so (ORB_SLAM3::Tracking::UpdateLastFrame()+112) (BuildId: d77a3e88b9b18f39f46d2f1f87d119b1f0920dce)
      #04 pc 00000000002a4cf0  /data/app/com.sim.slamar3N-lZvYl5P2IqjfhAJ7YakCwQ==/lib/arm64/libSlam.so (ORB_SLAM3::Tracking::TrackWithMotionModel()+68) (BuildId: d77a3e88b9b18f39f46d2f1f87d119b1f0920dce)
      #05 pc 00000000002a0fa0  /data/app/com.sim.slamar3N-lZvYl5P2IqjfhAJ7YakCwQ==/lib/arm64/libSlam.so (ORB_SLAM3::Tracking::Track()+2144) (BuildId: d77a3e88b9b18f39f46d2f1f87d119b1f0920dce)
      #06 pc 00000000002a05a8  /data/app/com.sim.slamar3N-lZvYl5P2IqjfhAJ7YakCwQ==/lib/arm64/libSlam.so (ORB_SLAM3::Tracking::GrabImageStereo(cv::Mat const&, cv::Mat const&, double const&, std::__ndk1::basic_string<char, std::__ndk1::char_traits<char>, std::__ndk1::allocator<char>>)+2108) (BuildId: d77a3e88b9b18f39f46d2f1f87d119b1f0920dce)
      #07 pc 000000000025ff5c  /data/app/com.sim.slamar3N-lZvYl5P2IqjfhAJ7YakCwQ==/lib/arm64/libSlam.so (ORB_SLAM3::System::TrackStereo(cv::Mat const&, cv::Mat const&, double const&, std::__ndk1::vector<ORB_SLAM3::IMU::Point, std::__ndk1::allocator<ORB_SLAM3::IMU::Point>> const&, std::__ndk1::basic_string<char, std::__ndk1::char_traits<char>, std::__ndk1::allocator<char>>)+2728) (BuildId: d77a3e88b9b18f39f46d2f1f87d119b1f0920dce)
      #08 pc 000000000001e1d4  /data/app/com.sim.slamar3N-lZvYl5P2IqjfhAJ7YakCwQ==/lib/arm64/libSLAM_JNI.so (slam_thread_func+2660) (BuildId: bd3897530c4c805eb5d45d292b325eb143071a68)
      #09 pc 00000000000e6890  /apex/com.android.runtime/lib64/bionic/libc.so (__pthread_start(void*)+36) (BuildId: 91cd22218589ba7b9ea39ce46bc68dff)
      #10 pc 0000000000084b6c  /apex/com.android.runtime/lib64/bionic/libc.so (__start_thread+64) (BuildId: 91cd22218589ba7b9ea39ce46bc68dff)

panping1314, 2022-01-17:
I have fixed this problem

closed 2022-01-17 by panping1314

wast0n, 2024-11-13:
hello, I meet the same problem. There is something wrong with mutex in GetPose in UpdateLastFrame in TrackWithMotionModel. Could you tell me how did you solve it?

wast0n, 2024-11-13:
> I have fixed this problem

I thin I know how to deal with it. Just remove ''-march=native'' in CMakeLists.txt.