UZ-SLAMLab/ORB_SLAM3 #472: Errors occur  "parse error in template argument list"

hero37th, 2022-01-17:
Ubuntu:16.04
OPENCV VERSION: 4.5.3
EIGEN VERSION: 3.4.1
Pangolin: 0.6

What's the current version of these tthirdparty?
______________
[ 37%] Building CXX object CMakeFiles/ORB_SLAM3.dir/src/TwoViewReconstruction.cc.o
[ 37%] Building CXX object CMakeFiles/ORB_SLAM3.dir/src/Config.cc.o
[ 38%] Building CXX object CMakeFiles/ORB_SLAM3.dir/src/Settings.cc.o
In file included from /usr/local/include/eigen3/unsupported/Eigen/CXX11/Tensor:88:0,
                 from /usr/local/opencv453/include/opencv4/opencv2/core/eigen.hpp:64,
                 from /home/xx/Downloads/ORB_SLAM3/src/Settings.cc:29:
/usr/local/include/eigen3/unsupported/Eigen/CXX11/src/Tensor/TensorBlock.h: In static member function ‘static IndexType Eigen::internal::TensorBlockIO<Scalar, IndexType, NumDims, Layout>::Copy(const Eigen::internal::TensorBlockIO<Scalar, IndexType, NumDims, Layout>::Dst&, const Eigen::internal::TensorBlockIO<Scalar, IndexType, NumDims, Layout>::Src&, const DimensionsMap&)’:
/usr/local/include/eigen3/unsupported/Eigen/CXX11/src/Tensor/TensorBlock.h:1303:7: error: the value of ‘j’ is not usable in a constant expression
       COPY_INNER_DIM(LinCopy::Kind::Linear);
       ^
/usr/local/include/eigen3/unsupported/Eigen/CXX11/src/Tensor/TensorBlock.h:1303:7: note: ‘int j’ is not const
       COPY_INNER_DIM(LinCopy::Kind::Linear);
       ^
/usr/local/include/eigen3/unsupported/Eigen/CXX11/src/Tensor/TensorBlock.h:1303:7: error: parse error in template argument list
       COPY_INNER_DIM(LinCopy::Kind::Linear);
       ^
/usr/local/include/eigen3/unsupported/Eigen/CXX11/src/Tensor/TensorBlock.h:1305:7: error: the value of ‘j’ is not usable in a constant expression
       COPY_INNER_DIM(LinCopy::Kind::Scatter);
       ^
/usr/local/include/eigen3/unsupported/Eigen/CXX11/src/Tensor/TensorBlock.h:1305:7: note: ‘int j’ is not const
       COPY_INNER_DIM(LinCopy::Kind::Scatter);
       ^
/usr/local/include/eigen3/unsupported/Eigen/CXX11/src/Tensor/TensorBlock.h:1305:7: error: parse error in template argument list
       COPY_INNER_DIM(LinCopy::Kind::Scatter);
       ^
/usr/local/include/eigen3/unsupported/Eigen/CXX11/src/Tensor/TensorBlock.h:1307:7: error: the value of ‘j’ is not usable in a constant expression
       COPY_INNER_DIM(LinCopy::Kind::FillLinear);
       ^
/usr/local/include/eigen3/unsupported/Eigen/CXX11/src/Tensor/TensorBlock.h:1307:7: note: ‘int j’ is not const
       COPY_INNER_DIM(LinCopy::Kind::FillLinear);
       ^
/usr/local/include/eigen3/unsupported/Eigen/CXX11/src/Tensor/TensorBlock.h:1307:7: error: parse error in template argument list
       COPY_INNER_DIM(LinCopy::Kind::FillLinear);
       ^
/usr/local/include/eigen3/unsupported/Eigen/CXX11/src/Tensor/TensorBlock.h:1309:7: error: the value of ‘j’ is not usable in a constant expression
       COPY_INNER_DIM(LinCopy::Kind::FillScatter);
       ^
/usr/local/include/eigen3/unsupported/Eigen/CXX11/src/Tensor/TensorBlock.h:1309:7: note: ‘int j’ is not const
       COPY_INNER_DIM(LinCopy::Kind::FillScatter);
       ^
/usr/local/include/eigen3/unsupported/Eigen/CXX11/src/Tensor/TensorBlock.h:1309:7: error: parse error in template argument list
       COPY_INNER_DIM(LinCopy::Kind::FillScatter);
       ^
/usr/local/include/eigen3/unsupported/Eigen/CXX11/src/Tensor/TensorBlock.h:1311:7: error: the value of ‘j’ is not usable in a constant expression
       COPY_INNER_DIM(LinCopy::Kind::Gather);
       ^
/usr/local/include/eigen3/unsupported/Eigen/CXX11/src/Tensor/TensorBlock.h:1311:7: note: ‘int j’ is not const
       COPY_INNER_DIM(LinCopy::Kind::Gather);
       ^
/usr/local/include/eigen3/unsupported/Eigen/CXX11/src/Tensor/TensorBlock.h:1311:7: error: parse error in template argument list
       COPY_INNER_DIM(LinCopy::Kind::Gather);
       ^
/usr/local/include/eigen3/unsupported/Eigen/CXX11/src/Tensor/TensorBlock.h:1313:7: error: the value of ‘j’ is not usable in a constant expression
       COPY_INNER_DIM(LinCopy::Kind::Random);
       ^
/usr/local/include/eigen3/unsupported/Eigen/CXX11/src/Tensor/TensorBlock.h:1313:7: note: ‘int j’ is not const
       COPY_INNER_DIM(LinCopy::Kind::Random);
       ^
/usr/local/include/eigen3/unsupported/Eigen/CXX11/src/Tensor/TensorBlock.h:1313:7: error: parse error in template argument list
       COPY_INNER_DIM(LinCopy::Kind::Random);
       ^
/usr/local/include/eigen3/unsupported/Eigen/CXX11/src/Tensor/TensorBlock.h: In static member function ‘static void Eigen::internal::TensorBlockAssignment<Scalar, NumDims, TensorBlockExpr, IndexType>::Run(const Eigen::internal::TensorBlockAssignment<Scalar, NumDims, TensorBlockExpr, IndexType>::Target&, const TensorBlockExpr&)’:
/usr/local/include/eigen3/unsupported/Eigen/CXX11/src/Tensor/TensorBlock.h:1532:35: error: the value of ‘j’ is not usable in a constant expression
[... 36 lines cut ...]
                 from /home/xx/Downloads/ORB_SLAM3/src/Settings.cc:29:
/usr/local/include/eigen3/unsupported/Eigen/CXX11/src/Tensor/TensorReverse.h: In member function ‘Eigen::TensorEvaluator<const Eigen::TensorReverseOp<ReverseDimensions, XprType>, Device>::TensorBlock Eigen::TensorEvaluator<const Eigen::TensorReverseOp<ReverseDimensions, XprType>, Device>::block(Eigen::TensorEvaluator<const Eigen::TensorReverseOp<ReverseDimensions, XprType>, Device>::TensorBlockDesc&, Eigen::TensorEvaluator<const Eigen::TensorReverseOp<ReverseDimensions, XprType>, Device>::TensorBlockScratch&, bool) const’:
/usr/local/include/eigen3/unsupported/Eigen/CXX11/src/Tensor/TensorReverse.h:316:28: error: parse error in template argument list
     while (it[NumDims - 1].count < it[NumDims - 1].size) {
                            ^
/usr/local/include/eigen3/unsupported/Eigen/CXX11/src/Tensor/TensorReverse.h:342:21: error: parse error in template argument list
         if (++it[i].count < it[i].size) {
                     ^
In file included from /usr/local/include/eigen3/unsupported/Eigen/CXX11/Tensor:119:0,
                 from /usr/local/opencv453/include/opencv4/opencv2/core/eigen.hpp:64,
                 from /home/xx/Downloads/ORB_SLAM3/src/Settings.cc:29:
/usr/local/include/eigen3/unsupported/Eigen/CXX11/src/Tensor/TensorGenerator.h: In member function ‘Eigen::TensorEvaluator<const Eigen::TensorGeneratorOp<Generator, XprType>, Device>::TensorBlock Eigen::TensorEvaluator<const Eigen::TensorGeneratorOp<Generator, XprType>, Device>::block(Eigen::TensorEvaluator<const Eigen::TensorGeneratorOp<Generator, XprType>, Device>::TensorBlockDesc&, Eigen::TensorEvaluator<const Eigen::TensorGeneratorOp<Generator, XprType>, Device>::TensorBlockScratch&, bool) const’:
/usr/local/include/eigen3/unsupported/Eigen/CXX11/src/Tensor/TensorGenerator.h:220:28: error: parse error in template argument list
     while (it[NumDims - 1].count < it[NumDims - 1].size) {
                            ^
/usr/local/include/eigen3/unsupported/Eigen/CXX11/src/Tensor/TensorGenerator.h:243:21: error: parse error in template argument list
         if (++it[i].count < it[i].size) {
                     ^
CMakeFiles/ORB_SLAM3.dir/build.make:432: recipe for target 'CMakeFiles/ORB_SLAM3.dir/src/Settings.cc.o' failed
make[2]: *** [CMakeFiles/ORB_SLAM3.dir/src/Settings.cc.o] Error 1
make[2]: *** Waiting for unfinished jobs....
CMakeFiles/Makefile2:476: recipe for target 'CMakeFiles/ORB_SLAM3.dir/all' failed
make[1]: *** [CMakeFiles/ORB_SLAM3.dir/all] Error 2
Makefile:102: recipe for target 'all' failed
make: *** [all] Error 2

wubobupt, 2022-02-18:
I met the same error. do you have any solution now?

lihzh233, 2023-06-10:
change `if (++it[i].count < it[i].size) {` to `if (++(it[i].count) < it[i].size) {`,
and `while (it[NumDims - 1].count < it[NumDims - 1].size) {` to `while ((it[NumDims - 1].count) < it[NumDims - 1].size) {`.