UZ-SLAMLab/ORB_SLAM3 #473: [Test on our own data]Did not create new map when lose tracking

ckLibra, 2022-01-18:
Hi, thanks for your great work! I figure out the system is easy to fail when the fast rotation appears. At this time, the terminal outputs
"fail to track local map".
According to the paper, the system is supposed to relocalize. But it didn't. The viewer doesn't show any key points or update the map anymore.  Can anyone help me?

zhaodejian, 2022-03-08:
the same to me

zhaodejian, 2022-03-11:
> 

thanks bro

859557181QQ, 2023-06-09:
> > the same to me
> 
> I rename my own data with the same timestamp formulation as the public datasets and the problem is solved. Maybe it will be helpful to you.
> 
> > 我也一样
> 
> 我用与公共数据集相同的时间戳公式重命名我自己的数据，问题就解决了。也许对你有帮助。
hello,sorry to bother you，I have encountered the same problem as you, but I haven't solved it yet. I want to know how you changed the naming formula for the dataset? Or how do you get the naming formula for public datasets? Sorry, my language may be a bit confusing because my English is not good. This was translated by a translation software. If I could get your help, I would be very grateful!