UZ-SLAMLab/ORB_SLAM3 #475: T265 "Flys up" in Mono-Inertial ROS mode?

kastoria3, 2022-01-19:
I'm trying to run the "./Examples_old/ROS/ORB_SLAM3/Mono_Inertial" sample program with a T265.  I'm using the T265 because
- I have one
- There seemed to be calibration files already present

When I run the ROS in "Mono" only then everything seems to work correctly.
When I run in ROS "Mono_Inertial" then most of the time the system resets them map every 1-5 frames.  When it does work, the camera seems to 'fly up'.  Like the Gravity vector is pointed in the wrong direction.

I can tell that because the Z component in the 'CameraTrajectory.txt' is constantly increasing in the positive direction.  That matches with ROS's coordinate space being:
- +X forward
- +Y left
- +Z up

I did some calibrations with Kalibr because I thought something might be off with the IMU<->Camera matrix.  But my matrix is essentially the same as the one that is present in the repository.

Has anyone seen this work?  Is there any known issues around this?  Like the T265 calibration file is just incorrect or something?

kastoria3, 2022-01-20:
More information
The Example/Monocular-Inertial sample app for the T265 has the same behavior as the ROS version.
The Example/Stereo-Inertial T265 sample app works as expected.
The Example/Monocular T265 ROS app works correctly.

sumitsarkar1, 2022-01-21:
@kastoria3 are you able to perform map saving for T265 stereo inertial (no ROS)

kastoria3, 2022-01-21:
> @kastoria3 are you able to perform map saving for T265 stereo inertial (no ROS)

I'm not sure what you mean by 'map saving'.  I do get a valid, stable location for my device when I'm using Example/Stereo-Inertial (no ROS).  It is only the Monicular-Inertial code path that is not stable for me.

I currently think that this is an issue with my IMU.  The T265 says that it "comes calibrated" from the factory.  However, the linear acceleration sitting on my desk is between 9.4 and 9.6.  I think that ORB-SLAM3 wants that value to be 9.81 at rest (gravity only).

The other thing that I have noticed is that the axis for my IMU does not match the T265 documentation.  T265 says the IMU relative to the body of the device is:
- Right handed
- +x is left
- +y is up
- +z is forward

However measuring with the 'realsense-viewer' application shows
- Left handed
- +x is right
- +y is down
- +z is back / -z is forward

I'm still trying to sort out what all this means for the IMU.

sumitsarkar1, 2022-01-22:
@kastoria3 are you ablle to save the sparse map (which you build using ORB SLAM3 and T265) to your disk

kastoria3, 2022-01-24:
> @kastoria3 are you ablle to save the sparse map (which you build using ORB SLAM3 and T265) to your disk

I'm still not sure what you mean by that.  The program saves a CameraTrajectory and KeyFrameTrajectory file when it exits.  That does not sounds like the 'map' to me though.