UZ-SLAMLab/ORB_SLAM3 #476: Example script with dr35i crashes on launch

saunair, 2022-01-20:
`./Examples/Stereo-Inertial/stereo_inertial_realsense_D435i Vocabulary/ORBvoc.txt ./Examples/Stereo-Inertial/RealSense_D435i.yaml` crashes with the following error in the end: 

```
terminate called after throwing an instance of 'rs2::error'
  what():  Couldn't resolve requests
Aborted (core dumped)
```

clyde-7, 2022-02-18:
./Examples/Stereo-Inertial/stereo_inertial_realsense_D435i Vocabulary/ORBvoc.txt /home/max/ORB_SLAM3/Examples/Stereo-Inertial/RealSense_D435i.yaml
terminate called after throwing an instance of 'rs2::invalid_value_error'
  what():  object doesn't support option #85
[1]    10339 abort (core dumped)  ./Examples/Stereo-Inertial/stereo_inertial_realsense_D435i

Z-star1, 2022-04-22:
i have the same problem!

tan-may16, 2022-04-23:
Hi,

Even I am facing the same issue. @saunair did you find any solution to this?
Thanks

EmmCo, 2022-05-19:
> ./Examples/Stereo-Inertial/stereo_inertial_realsense_D435i Vocabulary/ORBvoc.txt /home/max/ORB_SLAM3/Examples/Stereo-Inertial/RealSense_D435i.yaml terminate called after throwing an instance of [...]

same problem, anyone can help us?

EmmCo, 2022-05-19:
> same problem, anyone can help us?

resolve the problem by annotating the line 'sensor.set_option(RS2_OPTION_AUTO_EXPOSURE_LIMIT,50000);'

saunair, 2022-06-08:
(Sorry for the delay)I tried the above mentioned solution by @EmmCo , still see the same issue.

saunair, 2022-06-09:
@aganal could you provide a concrete example? In my case, I haven't changed anything with the config

saunair, 2022-06-10:
> @aganal could you provide a concrete example? In my case, I haven't changed anything with the config

Weirdly if I try this for a couple of times, it starts working. This issue is intermittent in my case, will follow up after debugging for a bit

IAMHAADICOOL, 2023-11-08:
Hey, I am facing the same error, can you provide a permanent resolve?